slimbootloader/slimbootloader #787: [APL] SDHCI report error on LeafHill while booting

andreyv1978, 2020-07-30:
Hi

It is a strange error reported while SBL is booting
EmmcCheckSwitchStatus: Send status fails with CRC Error
EmmcSetBusMode: Switch to HS400 CRC Error
Error reported by SDHCI
Interrupt status = 8000
Error interrupt status = 1

However, the yocto boot has no critical errors:
sdhci: Secure Digital Host Controller Interface driver
sdhci: Copyright(c) Pierre Ossman
sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
broxton-pinctrl INT3452:03: pin 26 cannot be used as IRQ
genirq: Setting trigger mode 3 for irq 128 failed (intel_gpio_irq_type+0x0/0x140)
mmc0: SDHCI controller on PCI [0000:00:1b.0] using ADMA 64-bit
sdhci-pci 0000:00:1c.0: SDHCI controller found [8086:5acc] (rev b)
mmc1: SDHCI controller on PCI [0000:00:1c.0] using ADMA 64-bit
sdhci-pltfm: SDHCI platform and OF driver helper
brd: module loaded
mmc1: new HS400 MMC card at address 0001
mmcblk1: mmc1:0001 8GME4R 7.28 GiB 
mmcblk1boot0: mmc1:0001 8GME4R partition 1 4.00 MiB
mmcblk1boot1: mmc1:0001 8GME4R partition 2 4.00 MiB
mmcblk1rpmb: mmc1:0001 8GME4R partition 3 512 KiB, chardev (245:0)

I can't find in SBL code, where it comes from, also not sure if it has any impact on functionalities 
We have a customized LeafHill board design 

Andrey

andreyv1978, 2020-07-30:
Also, SD card disabled in .dlt, because we do not have this on the board
DEV_EN_CFG_DATA.DevEnControl1.SdcardEnable | 0x0

the booting error like:
SdCardIdentification: Executing Cmd8 fails with Device Error

The only reference to that I can find in edk2 code, but can't see where it called in SBL

mauricema, 2020-07-30:
You are right.   This message was not from SBL. It is from UEFI payload.  For that part,  you will need to get support from EDK2 community.
This message indicates HS400 mode might not be reliable since CRC error was seen.   Sometimes, it might be due to incorrect 
DLL tuning settings like register R_SCC_MEM_TX_CMD_DLL_CNTL/1/2,  R_SCC_MEM_RX_CMD_DATA_DLL_CNTL1,  R_SCC_MEM_RX_STROBE_DLL_CNTL, etc.  Or it could be some signal integrity issues on the board.

andreyv1978, 2020-07-31:
closing it, because I can access eMMC and format it

closed 2020-07-31 by andreyv1978