slimbootloader/slimbootloader #791: fwupdate is not recognised

andreyv1978, 2020-08-03:
Hi

I am following step by step info to generate capsula update and trigger it from UEFI shell
1. Original image is created and BIOS flashed and passed a full Yocto boot
2. made changes in configuration and generate capsula image as instructed https://slimbootloader.github.io/supported-hardware/apollo-lake-crb.html
3. Following Firmware update procedure:
https://slimbootloader.github.io/security/firmware-update.html
used generated image from section 2 to generate FwuImage.bin. I do not have CSME , CSMD, TMAC:IPFW not used
4. Copy FwuImage.bin into root directory on FAT partition of a USB key
5.  Type command fwupdate from shell

Output:
'fwupdate' is not recognized as an internal or external command, operable program, or script file.

I have UEFI shell from edk2 to run Yocto, not internal OsLoader

Andrey

rprangar, 2020-08-03:
Hi

FW Update is a OsShell payload command and should be tried from the OS Loader shell. Here is the code for reference - https://github.com/slimbootloader/slimbootloader/blob/master/Platform/ApollolakeBoardPkg/Library/ShellExtensionLib/CmdFwUpdate.c

Thanks
Ravi

rprangar, 2020-08-03:
Basically, you may be able to develop an UEFI APP that does the same if desired
https://slimbootloader.github.io/supported-hardware/apollo-lake-crb.html#triggering-firmware-update

andreyv1978, 2020-08-04:
would I be able to use multiple payloads:
1. UEFI for the general use
2. OsLoader for fwupgrade

rprangar, 2020-08-04:
UEFI for general use and booting UEFI OS is fine.

You dont need OsLoader for FW Upgrade. Actually, FW Update is implemented in another payload dedicated for this purpose. What you are looking for is a mechanism to trigger FW update.

SBL uses a sticky scratch pad register to signal FW update. For example, SBL for Apollo Lake platform uses BIT16 of PMC I/O register (Scratchpad for sharing data between BIOS and PMC Firmware (BIOS_SCRATCHPAD) - Offset 1090h) to trigger firmware update. When BIT16 is set, SBL will set the boot mode to FLASH_UPDATE.You can signal FW Update even from OS. 

Now, to trigger FW update, **all you need to do to signal FW update is set this register**. This can be done from the OsLoader Shell (built in command) or even from the OS. We are in the process of submitting a Linux script to signal FW update for SBL (https://github.com/slimbootloader/slimbootloader/pull/765). This relies on a kernel driver as documented in the script and provides a platform independent way to signal FW update from Linux.

mauricema, 2020-08-04:
BTW, this driver has been merged into kernel 5.8 release.  And to enable this , you will need to set INTEL_WMI_SBL_FW_UPDATE kernel config option.
https://github.com/torvalds/linux/blob/master/drivers/platform/x86/intel-wmi-sbl-fw-update.c

For SBL FWU trigger, please also refer below on how to trigger FWU.
https://slimbootloader.github.io/security/firmware-update.html?highlight=wmi

andreyv1978, 2020-08-05:
So, I can try to test the Fwupdate #765 ?
1. boot Yocto to the login screen from eMMC
2. Place SblFwUpdate.py and FwuImage.bin on USB
3. execute SblFwUpdate.py from USB
4. whait the system restarting and updating
 
not sure I have python on current Yocto, but can you confirm the steps to test ?

rprangar, 2020-08-05:
For the above PR to work, the SBL WMI kernel driver is required. Otherwise this method will not work. 

To trigger/signal FW update to SBL, we need to set a platform specific register (after copying the capsule to a specific location). This can be accomplished in different ways. When you do it from the OS, register accesses can be done only through ring 0 code and so a driver support is needed. 

For APL, based on the the [reference ](https://github.com/slimbootloader/slimbootloader/blob/master/Platform/ApollolakeBoardPkg/Library/ShellExtensionLib/CmdFwUpdate.c) I provided earlier, 
MmioAndThenOr32 (PMC_BASE_ADDRESS + R_PMC_BIOS_SCRATCHPAD, 0xFF00FFFF, BIT16);

If you are looking for a quick way to test FW Update and from UEFI shell, you can try the mm command. 
mm address [value] [-w 1|2|4|8] [-MEM | -PMEM | -MMIO | -IO | -PCI | -PCIE] [-n]

Here , the address would be PMC_BASE_ADDRESS + R_PMC_BIOS_SCRATCHPAD and the value would be BIT16

If you are using Windows, you can use a tool like RWEverything to set the same register.

Hope this helps.

andreyv1978, 2020-08-06:
closing it for now, the lack of drivers in Yocto image for LeafHill platform makes the process over complicated to go forward
There is no plan to update the main kernel to 5.8

closed 2020-08-06 by andreyv1978

andreyv1978, 2020-08-07:
just in case you would want to update LeafHill Linux kernel 
We are using Linux version 4.19.80-intel-pk-standard