slimbootloader/slimbootloader #805: [APL] Invalid IPC1 settings in LeafHill

andreyv1978, 2020-08-07:
Hi

Our board is customized based LeafHill Apollo Lake
The IPC1 settings is set as enabled by default in the SIlicon Settings->Device enable->Control 1
Unfortunately the Yocto boot log has a following errors:

[    7.079855] intel_punit_ipc intel_punit_ipc: Invalid IRQ, using polling mode
[   10.148663] pmc-ipc-plat INT34D2:00: IPC timed out, TS=0x24, CMD=0x20eb
[   10.277159] Invalid pltconfig, ensure IPC1 device is enabled in BIOS
[   10.294537] Invalid pltconfig, ensure IPC1 device is enabled in BIOS

see atached log file
[sbl-logging-07-08-2020.log](https://github.com/slimbootloader/slimbootloader/files/5041816/sbl-logging-07-08-2020.log)

What else should I set to make it work properly ?

myguitar, 2020-08-10:
Hi @andreyv1978 

I tried to replicate your issue, but the IPC works as expected. My test Yocto image is from http://downloads.yoctoproject.org/releases/yocto/yocto-2.0/machines/leafhill/.
Not sure but your issue might be caused by Yocto IPC driver regression in the later release.

You can try sysfs for your testing. These test commands will return immediately if there is no errors.
```
root@intel-corei7-64:~# echo '231 0' > /sys/devices/platform/INT34D2:00/simplecmd
root@intel-corei7-64:~# 
root@intel-corei7-64:~# echo '235 2' > /sys/devices/platform/INT34D2:00/simplecmd
root@intel-corei7-64:~# 
```

If you send an invalid command, it will give you an error.
```
root@intel-corei7-64:~# echo '235 9' > /sys/devices/platform/INT34D2:00/simplecmd
pmc-ipc-plat INT34D2:00: IPC failed: command not supported, STS=0x10092, CMD=0x90eb
pmc-ipc-plat INT34D2:00: command 235 error with -5
sh: write error: Input/output error
root@intel-corei7-64:~# 
```

andreyv1978, 2020-08-11:
Hi myguitar

I've merged the  https://github.com/slimbootloader/slimbootloader/pull/809 and it seems sorted IPC1 the boot have no messages related to IPC1
Unfortunately, I still have a problem with #809 see https://groups.io/g/slimbootloader/topic/apl_pcie_configurations_on/75909357?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,75909357 

Hope when PCI would be sorted the issue can be closed 

Andrey

andreyv1978, 2020-08-12:
Hi

The #809 makes the IPC1 error in Yocto boot disappeare, closing it now
thank you for your support

Andrey

closed 2020-08-12 by andreyv1978