slimbootloader/slimbootloader #834: HASH verification failure for x64 build on APL platform

shehamb, 2020-09-02:
Hello, 
I have built the SBL for x64 architecture using these commands:

python3 BootloaderCorePkg/Tools/GenerateKeys.py -k BootloaderCorePkg/Tools/SblKeys
python3 BuildLoader.py build -a x64 apl
and finally stitched it using:
python3 Platform/ApollolakeBoardPkg/Script/StitchLoader.py -i /mnt/hgfs/slimbootloader/brc3-fit-output-rp.bin -s Outputs/apl/Stitch_Components.zip -o /mnt/hgfs/slimbootloader/brc3-sbl.bin

While building it, i got these errors:

Fd File Name:STAGE1B (/home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/FV/STAGE1B.fd)

Generate Region at Offset 0x0
   Region Size = 0x14000
   Region Name = FV

Generating STAGE1B FV
###### ['GenFv', '-F', 'TRUE', '-a', '/home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/FV/Ffs/STAGE1B.inf', '-o', '/home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/FV/STAGE1B.Fv', '-i', '/home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/FV/STAGE1B.inf']
Return Value = 2
GenFv: ERROR 3000: Invalid
  the required fv image size 0x14bb8 exceeds the set fv image size 0x14000

i fixed it by changing the size in BoardConfig.py
-        self.STAGE1B_FD_SIZE      = 0x0006B000
+        self.STAGE1B_FD_SIZE      = 0x0008B000^M

and
-        self.STAGE2_SIZE          = 0x00032000
+        self.STAGE2_SIZE          = 0x00035000^M

With this, when the board boots up, HASH verification is failing. 
I am enclosing the log for reference. 
[sbl-hash-error-ia64.txt](https://github.com/slimbootloader/slimbootloader/files/5160895/sbl-hash-error-ia64.txt)

Regards, 
Mahesh.

mauricema, 2020-09-02:
I just tried with the latest SBL.  It booted fine to SBL shell in x64 mode on LeafHill CRB.
Please make sure you do a clean build and try.  Also please make sure you set your key directory properly.

lsubash, 2020-09-02:
Mahesh, Could you share build logs?
Keys are being generated to BootloaderCorePkg/Tools/SblKeys.  Are you setting env variable SBL_KEY_DIR?

shehamb, 2020-09-03:
I just repeated all the steps:
python3 BuildLoader.py clean
export
SBL_KEY_DIR=/home/mahesh/brc3/slimbootloader/BootloaderCorePkg/Tools/SblKeys
python3 BootloaderCorePkg/Tools/GenerateKeys.py -k
BootloaderCorePkg/Tools/SblKeys
python3 BuildLoader.py build -a x64 apl
python3 Platform/ApollolakeBoardPkg/Script/StitchLoader.py -i
brc3-fit-output-rp.bin -s Outputs/apl/Stitch_Components.zip -o
brc3-sbl-64.bin

But SBL is still failing with this error:
ASSERT_EFI_ERROR (Status = Invalid Parameter)
ASSERT [Stage1B]
/home/mahesh/brc3/slimbootloader/BootloaderCorePkg/Library/StageLib/StageLib.c(134):
!EFI_ERROR (Status)

Enclosing the build log and console log for your reference.

On another note, is it possible  to "disable" HASH verification? How can i
move back to older version of SBL which didn't have this verification?

Regards,
Mahesh.

BtGuard: VB : 0, MB : 0

============= Intel Slim Bootloader STAGE1A =============
SBID: SB_APLI
ISVN: 001
IVER: 000.005.001.000.20086
SVER: 011BC982B9B2B31D-dirty
FDBG: BLD(D X64) FSP(R)
FSPV: ID($APLFSP$) REV(01040301)
Loader global data @ 0xFEF71CC8
Run  STAGE1A @ 0xFEF00000
Load STAGE1B @ 0xFEF80000
No BtGuard verification !

============= Intel Slim Bootloader STAGE1B =============
CPU : APL-F1 [2C @ 1300MHz: premium SKU], uCode rev.8
Early GpioInit for EMMC
MMC global data init
Use SDMA instead of ADMA2
MMC Phase 1 init
SpiInstance = FEF73030
PchSpiBase at 0x00000D02
ScSpiBar0 at 0xD2000000
Registering container KEYH
HASH verification for usage (0x00000100) with Hash Alg (0x1): Security Violation
First 32 Bytes Input Data
  00000000: B7 CD A1 A1 6E 05 A1 99-0E 48 28 B2 1F 46 13 C2  *....n....H(..F..*
  00000010: DD E8 7E 56 33 14 71 4A-B8 BD 16 45 6E C8 2A 38  *...V3.qJ...En.*8*
Last 32 Bytes Input Data
  00000000: 2C 8B 10 B7 EC FA E3 CB-6D 02 D1 20 B0 0D A4 1F  *,.......m.. ....*
  00000010: 1F 2C 1E 5B E1 53 B0 2F-F8 73 19 BF 00 01 00 01  *.,.[.S./.s......*
Image Digest
  00000000: 6A 09 E6 67 BB 67 AE 85-3C 6E F3 72 A5 4F F5 3A  *j..g.g..<n.r.O.:*
  00000010: 51 0E 52 7F 9B 05 68 8C-1F 83 D9 AB 5B E0 CD 19  *Q.R...h.....[...*
HashStore Digest
Registering container failed due to Security Violation
Append public key hash into store: Unsupported
Load EXT CFG Data @ 0xFEF73B28:0x0D18 ... Success
HASH verification for usage (0x00000200) with Hash Alg (0x1): Security Violation
First 32 Bytes Input Data
  00000000: C0 6C 98 C5 68 A0 62 C7-31 D4 89 39 14 B9 BE F2  *.l..h.b.1..9....*
  00000010: 10 E1 A4 1D 79 66 D2 59-2B 27 66 0C 99 ED 22 CC  *....yf.Y+'f...".*
Last 32 Bytes Input Data
  00000000: FB 41 78 FB 2E 39 4E C1-14 AC DF F2 3A 24 ED 65  *.Ax..9N.....:$.e*
  00000010: F9 9E 19 54 FF 98 A3 E6-80 0C 22 FF 00 01 00 01  *...T......".....*
[... 442 lines cut ...]
"gcc" -g -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -ffunction-sections -fdata-sections -include AutoGen.h -fno-common -DSTRING_ARRAY_NAME=ElfLibStrings -m64 -fno-stack-protector "-DEFIAPI=__attribute__((ms_abi))" -maccumulate-outgoing-args -mno-red-zone -Wno-address -mcmodel=small -fpie -fno-asynchronous-unwind-tables -Wno-address -flto -DUSING_LTO -Os -DDISABLE_NEW_DEPRECATED_INTERFACES -c -o /home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/X64/BootloaderCommonPkg/Library/ElfLib/ElfLib/OUTPUT/./ElfLib.obj -I/home/mahesh/brc3/slimbootloader/BootloaderCommonPkg/Library/ElfLib -I/home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/X64/BootloaderCommonPkg/Library/ElfLib/ElfLib/DEBUG -I/home/mahesh/brc3/slimbootloader/MdePkg -I/home/mahesh/brc3/slimbootloader/MdePkg/Include -I/home/mahesh/brc3/slimbootloader/MdePkg/Include/X64 -I/home/mahesh/brc3/slimbootloader/BootloaderCommonPkg -I/home/mahesh/brc3/slimbootloader/BootloaderCommonPkg/Include /home/mahesh/brc3/slimbootloader/BootloaderCommonPkg/Library/ElfLib/ElfLib.c
rm -f /home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/X64/BootloaderCommonPkg/Library/ElfLib/ElfLib/OUTPUT/ElfLib.lib
"gcc-ar" cr /home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/X64/BootloaderCommonPkg/Library/ElfLib/ElfLib/OUTPUT/ElfLib.lib  @/home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/X64/BootloaderCommonPkg/Library/ElfLib/ElfLib/OUTPUT/object_files.lst
Building ... /home/mahesh/brc3/slimbootloader/BootloaderCommonPkg/Library/LiteFvLib/LiteFvLib.inf [X64]
"gcc" -g -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -ffunction-sections -fdata-sections -include AutoGen.h -fno-common -DSTRING_ARRAY_NAME=LiteFvLibStrings -m64 -fno-stack-protector "-DEFIAPI=__attribute__((ms_abi))" -maccumulate-outgoing-args -mno-red-zone -Wno-address -mcmodel=small -fpie -fno-asynchronous-unwind-tables -Wno-address -flto -DUSING_LTO -Os -DDISABLE_NEW_DEPRECATED_INTERFACES -c -o /home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/X64/BootloaderCommonPkg/Library/LiteFvLib/LiteFvLib/OUTPUT/./FvLib.obj -I/home/mahesh/brc3/slimbootloader/BootloaderCommonPkg/Library/LiteFvLib -I/home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/X64/BootloaderCommonPkg/Library/LiteFvLib/LiteFvLib/DEBUG -I/home/mahesh/brc3/slimbootloader/MdePkg -I/home/mahesh/brc3/slimbootloader/MdePkg/Include -I/home/mahesh/brc3/slimbootloader/MdePkg/Include/X64 -I/home/mahesh/brc3/slimbootloader/BootloaderCommonPkg -I/home/mahesh/brc3/slimbootloader/BootloaderCommonPkg/Include /home/mahesh/brc3/slimbootloader/BootloaderCommonPkg/Library/LiteFvLib/FvLib.c
[... cut ...] /Library/FspApiLib/X64/Thunk64To32.nasm > /home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/X64/BootloaderCorePkg/Library/FspApiLib/FspsApiLib/OUTPUT/X64/Thunk64To32.i
Trim --trim-long --source-code -o /home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/X64/BootloaderCorePkg/Library/FspApiLib/FspsApiLib/OUTPUT/X64/Thunk64To32.iii /home/mahesh/brc3/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/X64/BootloaderCorePkg/Library/FspApiLib/FspsApiLib/OUTPUT/X64/Thun

shehamb, 2020-09-03:
And just to confirm, i checked out a fresh copy of the SBL and followed all
the steps. I was able to boot 32-bit SBL successfully. But for 64-bit build
of SBL, i am facing the same problem:
Shell>  BtGuard: VB : 0, MB : 0

============= Intel Slim Bootloader STAGE1A =============
SBID: SB_APLI
ISVN: 001
IVER: 000.005.001.000.20087
SVER: 98B55AFFA19301CE-dirty
FDBG: BLD(D X64) FSP(R)
FSPV: ID($APLFSP$) REV(01040301)
Loader global data @ 0xFEF71CC8
Run  STAGE1A @ 0xFEF00000
Load STAGE1B @ 0xFEF80000
No BtGuard verification !

============= Intel Slim Bootloader STAGE1B =============
CPU : APL-F1 [2C @ 1300MHz: premium SKU], uCode rev.8
Early GpioInit for EMMC
MMC global data init
Use SDMA instead of ADMA2
MMC Phase 1 init
SpiInstance = FEF73030
PchSpiBase at 0x00000D02
ScSpiBar0 at 0xD2000000
Registering container KEYH
HASH verification for usage (0x00000100) with Hash Alg (0x1): Security
Violation
First 32 Bytes Input Data
  00000000: B7 CD A1 A1 6E 05 A1 99-0E 48 28 B2 1F 46 13 C2
 *....n....H(..F..*
  00000010: DD E8 7E 56 33 14 71 4A-B8 BD 16 45 6E C8 2A 38
 *...V3.qJ...En.*8*
Last 32 Bytes Input Data
  00000000: 2C 8B 10 B7 EC FA E3 CB-6D 02 D1 20 B0 0D A4 1F  *,.......m..
....*
  00000010: 1F 2C 1E 5B E1 53 B0 2F-F8 73 19 BF 00 01 00 01
 *.,.[.S./.s......*
Image Digest
  00000000: 6A 09 E6 67 BB 67 AE 85-3C 6E F3 72 A5 4F F5 3A
 *j..g.g..<n.r.O.:*
  00000010: 51 0E 52 7F 9B 05 68 8C-1F 83 D9 AB 5B E0 CD 19
 *Q.R...h.....[...*
HashStore Digest
Registering container failed due to Security Violation
Append public key hash into store: Unsupported
Load EXT CFG Data @ 0xFEF73B28:0x0CD8 ... Success
HASH verification for usage (0x00000200) with Hash Alg (0x1): Security
Violation
First 32 Bytes Input Data
  00000000: C0 6C 98 C5 68 A0 62 C7-31 D4 89 39 14 B9 BE F2
 *.l..h.b.1..9....*
  00000010: 10 E1 A4 1D 79 66 D2 59-2B 27 66 0C 99 ED 22 CC
 *....yf.Y+'f...".*
Last 32 Bytes Input Data
  00000000: FB 41 78 FB 2E 39 4E C1-14 AC DF F2 3A 24 ED 65
 *.Ax..9N.....:$.e*
  00000010: F9 9E 19 54 FF 98 A3 E6-80 0C 22 FF 00 01 00 01
 *...T......".....*
Image Digest
  00000000: 6A 09 E6 67 BB 67 AE 85-3C 6E F3 72 A5 4F F5 3A
 *j..g.g..<n.r.O.:*
  00000010: 51 0E 52 7F 9B 05 68 8C-1F 83 D9 AB 5B E0 CD 19
 *Q.R...h.....[...*
HashStore Digest
EXT CFG Data ignored ... Security Violation
GPIO based PID detection is disabled
Get Embedded BoardId
BOOT: BP0
MODE: 0
BoardID: 0x07
PlatformName: LeafHill
BootPolicy : 0x00000000
Memory Init
Looking for MRC saving data
MRC data NOT found
Load memory parameters from CfgData.
Load general parameter from CfgData.
Call FspMemoryInit ... Success
Loader global data @ 0x3ABBFD30
Load page table from memory @ 0x3ABB7000
Remapping Stage to 0x3A074000
Switch to memory stack @ 0x3ABFFF00
Stage1 stack: 0x2000 (0x1808 used)
Stage1 heap: 0xE000 (0x5240 used)
Call FspTempRamExit ... Success
CSE: MBP Data HOB not valid!
MRC: [v0.56.89.24]
Memory FSP @ 0x3AC00000
Memory TOP @ 0x3A074000
HASH verification for usage (0x00000002) with Hash Alg (0x1): Security
Violation
First 32 Bytes Input Data
  00000000: 4C 5A 34 20 2F 26 03 00-00 00 08 00 00 00 00 00  *LZ4
/&..........*
  00000010: 00 00 08 00 93 E0 94 01-01 70 00 00 01 00 01 00
 *.........p......*
Last 32 Bytes Input Data
  00000000: FF FF FF FF FF FF FF FF-FF FF FF FF FF FF FF FF
 *................*
  00000010: FF FF FF FF FF FF FF FF-FF DF 50 FF FF 4F E5 00
 *..........P..O..*
Image Digest
  00000000: 6A 09 E6 67 BB 67 AE 85-3C 6E F3 72 A5 4F F5 3A
 *j..g.g..<n.r.O.:*
  00000010: 51 0E 52 7F 9B 05 68 8C-1F 83 D9 AB 5B E0 CD 19
 *Q.R...h.....[...*
HashStore Digest
Loading Stage2 error - Security Violation !
Failed to load Stage2!
STAGE_1B: System halted!

mauricema, 2020-09-03:
ok, thanks for reporting.   I only tried on Windows.   If Linux has issue, we will look into it.
BTW, if you really want to disable security boot,  you can set "self.HAVE_VERIFIED_BOOT = 0" in the BoardConfig.py.

shehamb, 2020-09-03:
Sure Maurice. I will try disabling it and shall let you know.

shehamb, 2020-09-03:
Hi,
I have disabled the security boot in BoardConfig.py.
But i get this error:
mahesh@ubuntu:~/new-sbl/slimbootloader$ python3 BuildLoader.py build -a x64
apl
Using "/usr/local/bin/python3", Version 3.6.3
Using GCC5, Version 5.4.0
Using /usr/bin/openssl, Version OpenSSL 1.0.2g  1 Mar 2016
Using /usr/bin/nasm, Version NASM version 2.11.08
Using /usr/bin/git, Version git version 2.7.4
Build [apl] ...
SBL_KEY_DIR is set to
/home/mahesh/new-sbl/slimbootloader/BootloaderCorePkg/Tools/SblKeys !!
Create FSP component file
'/home/mahesh/new-sbl/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/FV/FSP_S.bin'
Create FSP component file
'/home/mahesh/new-sbl/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/FV/FSP_M.bin'
Create FSP component file
'/home/mahesh/new-sbl/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/FV/FSP_T.bin'
Warning: No 'SlimBootloader.bin' component in image list !
Traceback (most recent call last):
  File "BuildLoader.py", line 1466, in <module>
    main()
  File "BuildLoader.py", line 1463, in main
    args.func(args)
  File "BuildLoader.py", line 1375, in cmd_build
    Build(board).build()
  File "BuildLoader.py", line 1227, in build
    self.pre_build()
  File "BuildLoader.py", line 1159, in pre_build
    self._board._SIGNING_SCHEME, svn, self._key_dir,
os.path.join(self._fv_dir, 'KEYHASH.bin'))
  File
"/home/mahesh/new-sbl/slimbootloader/BootloaderCorePkg/Tools/BuildUtility.py",
line 382, in gen_pub_key_hash_store
    key_type = get_key_type(signing_key)
  File
"/home/mahesh/new-sbl/slimbootloader/BootloaderCorePkg/Tools/CommonUtility.py",
line 263, in get_key_type
    if not os.path.exists(in_key):
  File "/usr/local/lib/python3.6/genericpath.py", line 19, in exists
    os.stat(path)
TypeError: stat: path should be string, bytes, os.PathLike or integer, not
NoneType

I did :
python3 BuildLoader.py clean
export
SBL_KEY_DIR=/home/mahesh/new-sbl/slimbootloader/BootloaderCorePkg/Tools/SblKeys
python3 BootloaderCorePkg/Tools/GenerateKeys.py -k
 BootloaderCorePkg/Tools/SblKeys
python3 BuildLoader.py build -a x64 apl

shehamb, 2020-09-03:
> This is a bug in SBL build.
> I have submitted a PR to address it. Please try:
> #836 <https://github.com/slimbootloader/slimbootloader/pull/836>

Thank you for the quick response. This has resolved the build issue.

mauricema, 2020-09-04:
For the hash verification failure in GCC x64,  we root caused to be some calling convention issue. 
The following PR should fix it.  Please verify.
https://github.com/slimbootloader/slimbootloader/pull/839

commit 340b8708 (2020-09-04): Fix GCC x64 Crypto boot issue

This patch fixed GCC x64 Crypto boot issue. The issue was related
to the calling convention. "EFIAPI" was missing for several ASM
provided functions.
It also fixed #834.
closed 2020-09-04 by mauricema