slimbootloader/slimbootloader #841: convert dsc file to yaml

AntonioSanta, 2020-09-08:
Hello, 
I need to convert a .dsc file to yaml.
I read at https://slimbootloader.github.io/tools/index.html#gpio-tool that this should be possible by using GenGpioData.py tool.
But I cannot find this GenGpioData.py file in the repository.
I found Dsc2Yaml.py file instead. I tried with this but I get error:
_Cannot open file 'slimbootloader/Platform/CommonBoardPkg/CfgData/Template_BootOption.dsc'_
This file Template_BootOption.dsc doesn't exist on the repository but I found a "Template_BootOption.yaml" instead.
How can I then convert .dsc to yaml ? 
Should I convert Template_BootOption.yaml to .dsc first? and how ?

thank you

mauricema, 2020-09-08:
Do you need to convert the full CFGDATA from DSC to YAML ?  GenGpioData.py is designed to convert DSC to YAML.  We have sent a notification in the mailing list already on how to do the conversion.  Please follow that steps.  I copied it here for your convenience. 

If you still have legacy projects on your own, and want to move to use the new YAML format,  please follow below:
1.	Clone the latest SBL code 
2.	Run the following:
        python BootloaderCorePkg\Tools\Dsc2Yaml.py -i  <CfgDataDefPath >   -o  < YamlOutputDir>
        CfgDataDefPath :  Full path to CfgDataDef.dsc in your old SBL tree 
        YamlOutputDir:    Output DIR to store converted YAML files
3.	Review and copy the generated yamls files under YamlOutputDir.  These files can be used for CFGDATA in yaml format.
4.	To fully support build using YAML,  it is still required to rebase to the latest SBL source tree.

BTW the common DSC such as Template_BootOption.dsc, it is already converted in the new tree.  And it should be the same for all projects.

AntonioSanta, 2020-09-08:
Hello,
thank you for instructions.
"Do you need to convert the full CFGDATA from DSC to YAML ?"
yes, I do.
Thank you for the information about the mailing list.
I tried command : 
python BootloaderCorePkg\Tools\Dsc2Yaml.py -i <CfgDataDefPath> -o < YamlOutputDir>
but the script raises an errors as it is expecting Template_BootOption.dsc.
I found the already converted Template_BootOption.yaml in the tree, as you said, but Dsc2Yaml.py script requires  Template_BootOption.dsc  when running . Can I generate it or should I get it from older revisions of the tree?

AntonioSanta, 2020-09-08:
.. I got that Template_BootOption.dsc and other common .dsc files from "testing" branch and it looks Dsc2Yaml works fine now.
Thank you

mauricema, 2020-09-08:
Thanks for the updates!   Glad to know you addressed this issue.  :)

mauricema, 2020-09-10:
Close the issue since the issue was resolved.

closed 2020-09-10 by mauricema