slimbootloader/slimbootloader #844: Config Editor data in inconsistent format

JayTalbott, 2020-09-10:
Some of the fields of data shown on the various pages in the Config Editor changed format (hex or decimal) when the configuration data was changed to YAML. As an example, in the Config Editor when changing the settings for CFL-H, on the POWER page the first 4 ratios are in hex, while the next 4 are in decimal. These sort of inconsistencies are all over the place in the configuration data on various pages within the Config Editor. This really should be cleaned up.

mauricema, 2020-09-11:
Please verify with PR https://github.com/slimbootloader/slimbootloader/pull/846.

JayTalbott, 2020-09-11:
That appears to have fixed a lot of it, at least for CFL (I didn't check the other platforms). However, for each of the PCIe Clock Config pages, the values are displayed in decimal, but the help text is all hex values. Need to either change the format for these, or change the help text.

mauricema, 2020-09-11:
Updated the PR to address the additional issues you reported include #850.  Please check.

PR #846 (2020-09-15): Represent data in required format in ConfigEditor
Current ConfigEditor relies on the original input data format in YAML
to determine how to represent data in GUI. For example, if the data
value is HEX in YAML, then the data will be displayed in HEX format.
This patch switched to use the specified format type to reformat the
value string so that the display is always consistent with the required
format type.

It fixed #844.
commit 5f5cbaeb (2020-09-15): Represent data in required format in ConfigEditor

Current ConfigEditor relies on the original input data format in YAML
to determine how to represent data in GUI. For example, if the data
value is HEX in YAML, then the data will be displayed in HEX format.
This patch switched to use the specified format type to reformat the
value string so that the display is always consistent with the required
format type.

It fixed #844.
closed 2020-09-15 by mauricema