slimbootloader/slimbootloader #850: Errors in Config Editor

JayTalbott, 2020-09-11:
I just noticed when checking the latest changes for #844 that in the window where I launched Config Editor I got the following error messages:

$ python BootloaderCorePkg/Tools/ConfigEditor.py
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Python36\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "BootloaderCorePkg/Tools/ConfigEditor.py", line 547, in edit_num_finished
    widget.delete(0, END)
NameError: name 'END' is not defined
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Python36\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "BootloaderCorePkg/Tools/ConfigEditor.py", line 547, in edit_num_finished
    widget.delete(0, END)
NameError: name 'END' is not defined
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Python36\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "BootloaderCorePkg/Tools/ConfigEditor.py", line 547, in edit_num_finished
    widget.delete(0, END)
NameError: name 'END' is not defined

mauricema, 2020-09-11:
Thank you for reporting it.  Updated PR #846 should address this too.

JayTalbott, 2020-09-11:
Now I get this when I try and start up the Config Editor:

$ python BootloaderCorePkg/Tools/ConfigEditor.py
Traceback (most recent call last):
  File "BootloaderCorePkg/Tools/ConfigEditor.py", line 20, in <module>
    from GenCfgData import CGenCfgData, bytes_to_value, bytes_to_bracket_str, value_to_bytes, array_str_to_value
  File "C:\BMC\Test\slimbootloader\BootloaderCorePkg\Tools\GenCfgData.py", line 1137
    <<<<<<< HEAD
    ^
IndentationError: expected an indented block

mauricema, 2020-09-11:
Looks like you did not get a clean merge ?
Can you reset your branch to be in sync with master, and then apply the PR again ?
I tried it on my side and did not see this issue.

JayTalbott, 2020-09-11:
I re-cloned and now it's working. And the other issues seem to now be resolved.

mauricema, 2020-09-15:
Close this issue since PR #846 was merged.

closed 2020-09-15 by mauricema