slimbootloader/slimbootloader #851: Fast Boot not working for Coffee Lake

JayTalbott, 2020-09-11:
I just noticed that fast boot doesn't work on Coffee Lake.

I did a fresh clone of SBL and did an unmodified stock build for CFL to confirm this behavior on the CFL-H RVP. The SBL console log indicates that the MRC training data is being saved, but it's not being used on subsequent boots, as the FspMemoryInit time remains at 20+ seconds every boot.

The FSP has an MrcFastBoot FSP-M UPD setting, which SBL doesn't set anywhere. However, it's default is supposed to be enabled, so that shouldn't be the problem.

[nofastboot.log](https://github.com/slimbootloader/slimbootloader/files/5211868/nofastboot.log)

Praveenhp7, 2020-09-14:
Hi Jay,

From my side, I did a fresh clone of  SBL and tested on both CFL-H RVP& WHL CRB. Fastboot is working fine for both platforms
as expected.
on CFL-H RVP, FspMemoryInit  takes ~25 sec for 1st boot
                [2020-09-14 11:38:45.519] FSP-M variables for Intel(R) SGX were NOT updated.
                [2020-09-14 11:38:45.519] Call FspMemoryInit ... Success
                [2020-09-14 11:39:10.854] Loader global data @ 0x8CFCED70
                [2020-09-14 11:39:10.919] PostMemoryInit called
Subsequent boot takes around ~200 msec 
              [2020-09-14 11:39:17.830] FSP-M variables for Intel(R) SGX were NOT updated.
              [2020-09-14 11:39:17.830] Call FspMemoryInit ... Success
              [2020-09-14 11:39:18.041] Loader global data @ 0x8CFCED70
              [2020-09-14 11:39:18.041] PostMemoryInit called

- form your logs I don't see any error related memory save & store, hence not able to root cause issue.

please make sure you are using the latest & correct  CSE binaries from intel. 

Thanks
Praveen

JayTalbott, 2020-09-14:
HI Praveen,

I'm stitching SBL into the IFWI extracted from the CFL-H CRB. Why would the version of the CSE matter for fast boot to work?

- Jay

mauricema, 2020-09-14:
ME might have impacts on the MRC fast boot behavior.
Please refer to:
https://github.com/intel/FSP/issues/41
Not sure if it is your case here. 
You can check this using a debug FSP to see if FSP decided to do full training flow for some specific reason.

JayTalbott, 2020-09-15:
New ME solved the problem.

Thanks!

closed 2020-09-15 by JayTalbott