slimbootloader/slimbootloader #857: Build error for cml platform

wrongvoid, 2020-09-28:
Hello,
i just checked out the slimbootloader repository. And i did a build for cml platform using:
python BuildLoader.py build cml

Here is the toolchain version info:
Using "/usr/bin/python3", Version 3.6.9
Using GCC5, Version 7
Using /usr/bin/openssl, Version OpenSSL 1.1.1  11 Sep 2018
Using /usr/bin/nasm, Version NASM version 2.13.02
Using /usr/bin/git, Version git version 2.17.1

But it throws this error:

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  21722:     Processor(PR00,
Error    6074 -                                                                                                                    Name already exists in scope ^  (PR00)

    Original name creation/declaration below:
    /home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii     58:   External(\_SB.PR00, DeviceObj)

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  21841:     Name(CPC2, Package()
Warning  4089 -                                                                                                                   Object is not referenced ^

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  21869:     Name(CPOC, Package()
Warning  4089 -                                                                                                                   Object is not referenced ^

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  22832:     Name(PRES, 1)
Warning  4089 -                                                                                                                   Object is not referenced ^

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  22904:     Method (I2CM, 3, Serialized) {
Warning  4089 -                                                                                                                      Object is not referenced ^

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  23037:         Method(_CRS, 0x0, NotSerialized)
Remark   2120 -                                                                                                         Control Method should be made Serialized ^  (due to creation of named objects within)

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  23078:         Function(_DSM, {BuffObj, PkgObj}, {BuffObj, IntObj, IntObj, PkgObj})
Remark   2119 -                                                                                                                   Control Method marked Serialized ^  (Due to use of Switch operator)

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  23185:         Method(_CRS, 0x0, NotSerialized)
Remark   2120 -                                                                                                         Control Method should be made Serialized ^  (due to creation of named objects within)

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  23226:         Function(_DSM, {BuffObj, PkgObj}, {BuffObj, IntObj, IntObj, PkgObj})
Remark   2119 -                                                                                                                   Control Method marked Serialized ^  (Due to use of Switch operator)

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  23333:         Method(_CRS, 0x0, NotSerialized)
Remark   2120 -                                                                                                         Control Method should be made Serialized ^  (due to creation of named objects within)

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  23374:         Function(_DSM, {BuffObj, PkgObj}, {BuffObj, IntObj, IntObj, PkgObj})
Remark   2119 -                                                                                                                   Control Method marked Serialized ^  (Due to use of Switch operator)

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  23512:   External(\_SB.GGOV, MethodObj)
Error    6074 -                                                                                                                                  Name already exists in scope ^  (\_SB.GGOV)

    Original name creation/declaration below:
    /home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii   4626:     Method(GGOV, 0x1, Serialized)

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  23513:   External(\_SB.SGOV, MethodObj)
Error    6074 -                                                                                                                                  Name already exists in scope ^  (\_SB.SGOV)

    Original name creation/declaration below:
"gcc" -o /home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/BootloaderCorePkg/PcdData/PcdData/DEBUG/PcdData.dll -nostdlib -Wl,-n,-q,--gc-sections -z common-page-size=0x40 -Wl,--entry,_ModuleEntryPoint -u _ModuleEntryPoint -Wl,-Map,/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/BootloaderCorePkg/PcdData/PcdData/DEBUG/PcdData.map,--whole-archive -flto -Os -Wl,-m,elf_i386,--oformat=elf32-i386 -no-pie -Wl,--start-group,@/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/BootloaderCorePkg/PcdData/PcdData/OUTPUT/static_library_files.lst,--end-group -g -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -ffunction-sections -fdata-sections -include AutoGen.h -fno-common -DSTRING_ARRAY_NAME=PcdDataStrings -m32 -march=i586 -malign-double -fno-stack-protector -D EFI32 -fno-asynchronous-unwind-tables -Wno-address -fno-pic -fno-pie -flto -Os -fno-pic -DDISABLE_NEW_DEPRECATED_INTERFACES -Wl,--defsym=PECOFF_HEADER_SIZE=0x220 -Wl,--script=/home/uvm/slimbootloader/BaseTools/Scripts/GccBase.lds -Wno-error -no-pie
    /home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii   4650:     Method(SGOV, 0x2, Serialized)

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  23527:       Method(STA, 1, Serialized) {
Warning  4089 -                                                                                                                      Object is not referenced ^

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  23539:       Method(ON, 1, Serialized) {
Warning  4089 -                                                                                                                     Object is not referenced ^

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  23548:       Method(OFF, 1, Serialized) {
Warning  4089 -                                                                                                                      Object is not referenced ^
[... 21 lines cut ...]

/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  23899:     Method(WAK, 1) {
Remark   2146 -                                                                                                               Method Argument is never used ^  (Arg0)

[... cut ...]
        make tbuild [/home/uvm/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables]

build.py...
 : error F002: Failed to build module
        /home/uvm/slimbootloader/Platform/CometlakeBoardPkg/AcpiTables/AcpiTables.inf [IA32, GCC5, DEBUG]

- Failed -
Build end time: 08:11:06, Sep.28 2020
Build total time: 00:00:29

Error in running process:
  build --platform BootloaderCorePkg/BootloaderCorePkg.dsc -b DEBUG --arch IA32 --tagname GCC5 -n 4 -y Report.log -Y PCD -Y FLASH -Y LIBRARY

Is this a problem caused by the gcc version?

wrongvoid, 2020-09-28:
This will also cause problems
python BuildLoader.py build -a x64 cml

mauricema, 2020-09-28:
It seems to be IASL version issue.
Please refer:
https://slimbootloader.github.io/getting-started/build-host-setup.html
So you might need to use  IASL 20160422 to give a try.

wrongvoid, 2020-09-29:
Hi mauricema,

Thanks for your information.
Download link from official website www.acpica.org is 403
https://stg.syncevolution.org/sites/acpica/files/acpica-unix-20160422.tar.gz
I downloaded source code from:
https://acpica.org/sites/acpica/files/acpica-unix-20160422.tar.gz

Used following instruction:
tar xvf acpica-unix-20160422.tar.gz
cd acpica-unix-20160422
sudo apt install flex bison
//flex version: 2.6.4-6
//bison version: 2:3.0.4.dfsg-1build1
make
sudo make install
iasl -v

$
Intel ACPI Component Architecture
ASL+ Optimizing Compiler version 20160422-64
Copyright (c) 2000 - 2016 Intel Corporation

Maybe can add it to Note
Thanks for the pointers. I will close this issue now.
Best Regards,

closed 2020-09-29 by wrongvoid

mauricema, 2020-09-29:
You are welcome!