slimbootloader/slimbootloader #864: SBL-64 bit doesn't boot all the way to UEFI payload

shehamb, 2020-10-08:
Hello,
I have built a 64-bit SBL with a UEFI payload (uefi shell). But it seems like SBL is attempting to jump to payload, but nothing happens. I don't see anything after that on the console. However, 32-bit SBL with UEFI payload works fine. 
I have built SBL on the windows 10. And the target is custom hardware based on Leaf Hill CRB. 
Regards, 
Mahesh.
[sbl64-with-uefi-payload-log1.txt](https://github.com/slimbootloader/slimbootloader/files/5347530/sbl64-with-uefi-payload-log1.txt)

shehamb, 2020-10-08:
> Are you using a 64-bit built UEFI payload?

Yes. I am using 64 bit payload.

mauricema, 2020-10-09:
The current UEFI payload entry in open source is always using 32bit entry point.
For IA32 build,  it is 32bit PEI + 32bit DXE.
For X64 build, it is 32bit PEI + 64bit DXE.  
So in this case, it is always required to switch to 32bit mode to call the UEFI payload entry point.
We will take a look to see if there is any good way to handle it automatically in SBL.

mauricema, 2020-10-09:
I have submitted a PR:
https://github.com/slimbootloader/slimbootloader/pull/870
This patch should be able to address this issue for mixed payload and SBL mode support.

shehamb, 2020-10-12:
Hello Maurice,
Thanks for this fix. I am able to boot up the 64-bit SBL with the UEFI
payload now.
Regards,
Mahesh.

mauricema, 2020-10-12:
Since the PR was merged, close it.

closed 2020-10-12 by mauricema