slimbootloader/slimbootloader #875: APL "NOOPT" build fails due to compiler intrinsic functions

mmustafa88, 2020-10-14:
When build for APL with "NOOPT" flag, the build fails. Please find below errors:

NvmExpressLib.lib(NvmExpressPassthru.obj) : error LNK2001: unresolved external symbol __aullshr
TimeStampLib.lib(TimeStampLib.obj) : error LNK2001: unresolved external symbol __aullshr
IppCryptoLib.lib(pcpbnu32arith.obj) : error LNK2001: unresolved external symbol __aullshr
IppCryptoLib.lib(pcpsm3ca.obj) : error LNK2001: unresolved external symbol __allshl
IppCryptoLib.lib(pcpsha512ca.obj) : error LNK2001: unresolved external symbol __allshl
IppCryptoLib.lib(pcpsha256ca.obj) : error LNK2001: unresolved external symbol __allshl
IppCryptoLib.lib(pcpbnu32arith.obj) : error LNK2001: unresolved external symbol __allshl

mauricema, 2020-10-14:
Please check PR#876.   Please note, that patch will only address compiling issue caused by compiler intrinsic functions.

rprangar, 2020-10-15:
Please close the issue if the build issue is resolved.

Thanks.

mmustafa88, 2020-10-16:
Fixed.

closed 2020-10-16 by mmustafa88