slimbootloader/slimbootloader #881: HeciInitialize error on APL platform

shehamb, 2020-10-15:
Hello, 

We are experiencing significant manufacturing fallout (poor yield) on our custom design which is based on Apollo Lake (E3930). Some of the units boot-up quickly and reliably. Other units boot correctly immediately after power-up, but fail to boot on subsequent hardware or software resets.

We have adapted the Slim Bootloader BIOS to our product and built it with the Debug version of the FSP library.

When a unit fails to boot, it seems to pause at the following message in the boot log: 
 
HeciInitialize HECIMBAR 0XD3708000
 
Occasionally, after the unit pauses at this message, it automatically reboots, proceeds past this message on its second attempt, and successfully loads the operating system. In other situations, the self-reboot does not succeed and the unit remains stuck, displaying the second copy of the message indefinitely.

I had the following questions to Intel rep:

1.   What is the FSP library trying to accomplish when it displays this message "HeciInitialize HECIMBAR 0XD3708000"?
2.   What hardware or software failures would prevent this boot step from succeeding?
3.   Why does the Apollo Lake processor (or FSP library) reboot itself when this step fails the first time?
4.   After a self-reboot, why would a second attempt at executing the same code succeed? 
5.   Why would the FSP library stop self-reboot attempts? Does it count these attempts?
 
But Intel rep has come back to us saying that:
"According to Apollo Lake FSP Release Note, in chapter 8.0 "Hardware and Software Compatibility" stated that Coreboot is the supported bootloader. Besides that, based on IOTG Platform Boot Solution roadmap, Intel FSP with Slim Bootloader is not supporting Apollo Lake while Intel FSP with coreboot is supporting in Apollo Lake and you may refer to IOTG NDA 5Q Platform Roadmap for more details."

So, could someone please provide any leads on this issue?

Regards, 
Mahesh.

mauricema, 2020-10-15:
Hi, Mahesh,

Could you let us know which FSP release version you are using?  Could you give a try with APL FSP MR5 if you are using other version?  Also if you can share a debug log in a failure case, that will be very helpful.  
In general "HeciInitialize HECIMBAR" indicates FSP was trying to initialize HECI interface to talk to CSME.   
The reset could be triggered by the hardware itself or sometimes, issued by the FSP.

rprangar, 2020-10-15:
Hi Mahesh

Since SBL was introduced after APL was launched, the support for APL SBL will be through this open source channels (github issues and mailing list). 

For FSP issues, you can still get direct support from Intel through the Intel Premier Support (IPS) channel if required. For example, you may be able to a debug version of FSP to see what is happening.

Please try the suggestions provided by Maurice above to help debug this further.

Thanks.

rprangar, 2020-10-15:
BTW, it may also help to file an issue here - https://github.com/intel/FSP/issues

shehamb, 2020-10-20:
Hello Maurice and Ravi,

We're using the FSP MR6 version. I will try to get the failure log from our
hardware engineer since these failures happen on his set up.

If you're suggesting to try out using FSP MR5 version, then i'll have to
request FSP MR5 debug library from Intel rep first.

Regards,
Mahesh.

rprangar, 2020-10-23:
Hi Mahesh

As mentioned earlier, we can discuss any SBL related issues here. For FSP issues, you can follow up through the FSP channels.

Can we close this issue?

Thanks.

closed 2020-10-27 by rprangar