slimbootloader/slimbootloader #895: SBL update error!

tomwei666, 2020-11-06:
Desccribe the bug
There is an old sbl in my up2 board!
I compile the sbl and update throug the USB.

Compiling the sbl steps:

export SBL_KEY_DIR=/tmp/sbl_key_dir/
python3.6 BootloaderCorePkg/Tools/GenerateKeys.py -k $SBL_KEY_DIR
python3.6 BuildLoader.py build apl
python3.6 Platform/ApollolakeBoardPkg/Script/StitchLoader.py -b sbl.bios.bin -i UPA1AM52.bin -s Outputs/apl/Stitch_Components.zip -o up2_sbl.bin -p 0xAA00000E
ps: My board : up2-N3350, so I download the UPA1AM52.zip BIOS firmware.
Update the sbl firmware steps:

Copy up2_sbl.bin to the usb root dir, and rename up2_sbl.bin to FwuImage.bin
Enter SBL shell
input fwupdate command
The attach update-error.log: update sbl log.
The attach before-update.log: Before updating ,the sbl log.
The sbl in my board version is sbl_up2_ifwi.bin(2019-03-19 20:59) is from acrn daily server.

Platform
Product/Kit Name: UP2-N3350
Memory: 4G
Storage: 32G
[before-upgrade.txt](https://github.com/slimbootloader/slimbootloader/files/5499613/before-upgrade.txt)
[update-error.log](https://github.com/slimbootloader/slimbootloader/files/5499616/update-error.log)

raghavag, 2020-11-06:
Hi,

Slim Bootloader binary (up2_sbl.bin) cannot be used directly for firmware update. Slim Bootloader uses a specific capsule format. 

Steps to generate capsule image for Slim Bootloader are outlined here https://slimbootloader.github.io/security/firmware-update.html#generating-capsule

tomwei666, 2020-11-09:
Dear,

python ./BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p BIOS sbl.bios.bin -p CSME  csme.bin -p CSMD csme_fw_update_driver.bin -p TMAC:IPFW tsnmacaddr.bin -k $SBL_KEY_DIR/FirmwareUpdateTestKey_Priv_RSA2048.pem -o FwuImage.bin

Following your advice, I find the above command , my question is:

Where can I get the images of csme.bin,csme_fw_update_driver.bin and tsnmacaddr.bin?

Brs,
Tom

raghavag, 2020-11-09:
Hi Tom,

that command is an example command to integrate SBL, CSME, CSME update driver and TsnMacaddr container component into capsule image.

In you case, if you just need to update SBL, your command would be 

python ./BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p BIOS up2_sbl.bin -k $SBL_KEY_DIR/FirmwareUpdateTestKey_Priv_RSA2048.pem -o FwuImage.bin

tomwei666, 2020-11-10:
Hi,

   I want to generate IAS image for fastboot support , I followed the link: https://slimbootloader.github.io/how-tos/boot-acrn.html

  The steps I do:
1）mkdir -p Platform/ApollolakeBoardPkg/SpiIasBin
2）touch cmdline.txt
3）chmod 755 ./iasimage
4）sudo pip3 install cryptography==2.2.2 --target=/home/tom/work/slimbootloader
5）  ./iasimage create -o Platform/ApollolakeBoardPkg/SpiIasBin/iasimage1.bin -d ./BootloaderCorePkg/Tools/Keys/TestSigningPrivateKey.pem -i 0x40000 cmdline.txt fastboot.elf
  note:  In SBL master branch ,there is no dir: BootloaderCorePkg/Tools/Keys/. I copy this dir from Testing branch. I compile the code in master branch.

6) python3.6 BuildLoader.py build apl
7) python3.6 Platform/ApollolakeBoardPkg/Script/StitchLoader.py -i UPA1AM40.bin -s Outputs/apl/Stitch_Components.zip -o sbl_up2_ifwi.bin -p 0xAA00000E -b bios.bin
8) python3.6  BootloaderCorePkg/Tools/GenCapsuleFirmware.py -k $SBL_KEY_DIR/FirmwareUpdateTestKey_Priv_RSA2048.pem -o FwuImage.bin -p BIOS  bios.bin

When I update FwuImage using USB, I got these error log in attach file of 
[update-error-20201110.txt](https://github.com/slimbootloader/slimbootloader/files/5514510/update-error-20201110.txt)

tomwei666, 2020-11-10:
Hi I followed as link https://github.com/projectacrn/acrn-hypervisor/issues/5478
yet another error!

raghavag, 2020-11-12:
Hello Tom,

We have updated keys in our code recently and look like in your case the board has older version of SBL and is using older key, which does not match with the key from latest code, hence we are seeing security violation.

I will have to try and use the older code on my side and verify it does work properly. I will get back to you on this.

Do you have a dediprog that you can use to update SBL to latest on UP2 board?

Thanks,
Raghava

mauricema, 2020-11-14:
As Raghava mentioned,  lots of things have changed in SBL, especially the signing keys.  And your current image (Jan 2019) is too old to catch up with the latest code.  To make FWU work, lots of intermediate FW updates are required in order to get to the latest SBL.  In your case, the safest way is to use DediProg to update to latest SBL image directly.  Once you are on latest SBL image, you can try ACRN again.

mauricema, 2020-11-16:
If you still want to do the FWU from old image to new SBL image,  I can provide the detailed flow.   Again, please remember this process could be very risky.  If anything is wrong here, potentially it will break the board boot.   It is highly recommended that you have Dediprog as backup for recovery before you try these steps below.

1. Clone a refresh SBL into a workspace. ex: D:\Work\SblNew

2. Build latest SBL image, it needs to be in python3 environment
   d:
   cd \Work\SblNew
   python BuildLoader.py build apl -k
   python Platform\ApollolakeBoardPkg\Script\StitchLoader.py -i UPA1AM40.bin -s Outputs\apl\Stitch_Components.zip -b Outputs\apl\Bios.bin -o Outputs\apl\AplIfwi.bin
   python BootloaderCorePkg\Tools\GenCapsuleFirmware.py -p BIOS Outputs\apl\Bios.bin -k KEY_ID_FIRMWAREUPDATE_RSA2048 -o Outputs\apl\FwuImage.bin

3. Clone another refresh SBL into a workspace for old SBL stitching. ex: D:\Work\SblOld
   d:
   cd \Work\SblOld
   git checkout EDC112328CF3E414
   # NOTE: Use Bios.bin generated from the SblNew tree
   # It needs to use python2
   python2 BootloaderCorePkg\Tools\GenCapsuleFirmware.py -t b -b ..\SblNew\Outputs\apl\Bios.bin -k BootloaderCorePkg\Tools\Keys\TestSigningPrivateKey.pem -o FwuImage.bin

4. Copy D:\Work\SblOld\FwuImage.bin to USB key and then boot to SBL shell to start Firmware update flow.

5. The firmware update will finish and reboot to SBL Shell.

6. Copy D:\Work\SblNew\Outputs\apl\FwuImage.bin to USB key and then start Firmware update flow again.

7. It might get into a boot loop during firmware update flow.  If so, power the board off.

8. Power on board again.  The firmware update should continue now.

9. Wait till the full firmware update is done.

10. The board should have latest SBL now.

From there, you can try other things.

mauricema, 2020-11-24:
Any update ?  Can we close this ?

mauricema, 2020-12-01:
Since there is no update for this issue,  will close it for now.

closed 2020-12-01 by mauricema