slimbootloader/slimbootloader #901: [TGL] Error was found when switching Slim Bootloader for TGL

wt-kevin, 2020-11-12:
1. Download 614394_sbl-tgl-up3-pv-rel.zip fron Intel RDC.
2. Follow SBL_TGL_UP3_ReleaseNotes.pdf chapter5 to build a SBL bin successfully.
3. Refer to Chapter6 - Slim Bootloader Stitching to stitch SBL and use command as below.
4. python SblInternal\Platform\TigerlakeBoardPkg\Script\StitchIfwi.py -b vm -s Outputs\tgl\Stitch_Components.zip -c SblInternal\Platform\TigerlakeBoardPkg\Script\StitchIfwiConfig_tglu.py -w D:\Stitching -p tglu_b0
5. Error was found, please refer to the following txt

jgutbub, 2020-11-12:
Looks like the stitching versions you are using could be having some compatibility issues. Please refer to the versions given in the documentation included with the release package you downloaded from Intel RDC, sbl-tgl-up3-pr1-rel\SBL_TGL_UP3_ReleaseNotes.pdf page 16. You may need to contact your Intel representative to download any packages you are not able to access yourself.

When stitching with the versions in the documentation it is successful. And double check the filename Input/PchcSubPartition**Dat**.bin, it should be PchcSubPartition**Data**.bin.

Also please remove the dropbox contents you uploaded, some of the tools being shared are not publicly available.

wt-kevin, 2020-11-12:
Hi James,

Removed the dropbox contents, sorry for that,  we can download all of file in Table 5 Packages required for stitching of sbl-tgl-up3-pr1-rel\SBL_TGL_UP3_ReleaseNotes.pdf page 16, it should be right files, let me doulbe check it.
If you have any ideals, feel free to tell me, thanks for hlep.

jgutbub, 2020-11-12:
Hi Kevin,

I think if you can just try with the stitching version from the documentation your command will work, when I tried this it was able to work aside from the one filename issue. If you still have issue after setting up the new stitching environment please share the stitching log.

wt-kevin, 2020-11-13:
Hi James,

We followed Table5 and 6 to download all of files except for Diagnostic ACM, because we can't download B0\tgl-up3_diagnostic_acm_b0_200225.bin as doc mentioned, we used tgl-up3_diagnostic_acm_b0_200812.bin. 
The result is to stitch successfully, thanks for help.
BTW,  we can use the IFWI to boot on TGL CRB directly? don't need to modify ME settin via FIT?
we will try to boot next week, hope to get some suggestions, thank you.

Kevin,

jgutbub, 2020-11-13:
Hi Kevin,

Can you share where you read about 200225.bin? In the 614394 SBL_TGL_UP3_ReleaseNotes.pdf rev 1.5 I am seeing it as 200812.bin but would like to confirm in case others encounter this as well.

The IFWI being generated will work on the board mentioned in the RDC document 612924_TGL_UP3_RVP_UG_Rev1p1.pdf (you can use this document for further reference and board bring-up). Just want to confirm this is the board you are referring to, there is TGL UP3 DDR4 RVP and TGL UP3 LP4X RVP. The SBL package you are using can support both of these boards with the default build. The stitching command you are using will work for DDR4 RVP. If using LP4 RVP there is an optional argument 'lp4' mentioned in SBL_TGL_UP3_ReleaseNotes.pdf 6.4.1 that needs to be added to the stitching command (-o "lp4").

wt-kevin, 2020-11-15:
Hi James,
We use TGL UP3 LPDDR4 RVP, it seems like we don't need to modify ME setting by FIT(just use ME default setting) so that we can boot on it directly after stitching, right?
I can see 200225.bin in release note of 614394_sbl-tgl-up3-pv-rel.zip,
link on table 5 linked to 200812.bin but table 6 mentioned 200225.bin as picture as below,
[image]
[image]

jgutbub, 2020-11-16:
Correct, you won't need to adjust any ME or other FIT settings using the stitching command that you provided initially, e.g. python SblInternal\Platform\TigerlakeBoardPkg\Script\StitchIfwi.py -b vm -s Outputs\tgl\Stitch_Components.zip -c SblInternal\Platform\TigerlakeBoardPkg\Script\StitchIfwiConfig_tglu.py -w D:\Stitching -p tglu_b0

I see now where you are getting 200225.bin from, it will be best to refer to the release document that is coming from 614394_sbl-tgl-up3-**pr1**-rel.zip instead of 614394_sbl-tgl-up3-**pv**-rel.zip, PR1 is the latest release.

jgutbub, 2020-11-16:
Correct, you won't need to adjust any ME or other FIT settings manually but if you are using LPDDR4 board you should use the following stitching command, python SblInternal\Platform\TigerlakeBoardPkg\Script\StitchIfwi.py **-o "lp4"** -b vm -s Outputs\tgl\Stitch_Components.zip -c SblInternal\Platform\TigerlakeBoardPkg\Script\StitchIfwiConfig_tglu.py -w D:\Stitching -p tglu_b0

I see now where you are getting 200225.bin from, it will be best to refer to the release document that is coming from 614394_sbl-tgl-up3-pr1-rel.zip instead of 614394_sbl-tgl-up3-pv-rel.zip, PR1 is the latest release.

jgutbub, 2020-11-17:
Hi Kevin,

Since your stitching issue has been resolved I will close this issue. If you encounter any other issues or need further guidance you can open a new issue or for more general/non-issue SBL questions you can contact our mailing list (more details here https://groups.io/g/slimbootloader).

Thanks,
James

closed 2020-11-17 by jgutbub