slimbootloader/slimbootloader #925: Error: infinite reboot Loop in Update Firmware.

vudinh98, 2020-12-11:
I use Slim Bootloader to develop BIOS  for UP2 (Apollo Lake), but when I triggered to Firmware update, my board experienced **infinite reboot loop** without stopping. Only when I manually force shutdown (hold power button for 5 seconds) and turn on the board again, the **infinite loop disappear.** 
I also attach 2 files below, "FWU_process" describes process from start of FW update to infinite loop, "FWU_success" describes process after I force shutdown and turn on the board again.

Can anyone tell me how to solve this problem? Thanks in advance!

[FWU_Sucess.log](https://github.com/slimbootloader/slimbootloader/files/5676681/FWU_Sucess.log)
[FWU_Process.log](https://github.com/slimbootloader/slimbootloader/files/5676682/FWU_Process.log)

mauricema, 2020-12-11:
Thank you for reporting this issue.
We were able to reproduce this on our APL board.   It seems to be a regression issue.  It was working in early version.
To unblock you, please use the following commit for now for your SBL:
commit 467c804172b160a19fac436fec55431dbace1783
Meanwhile we will look into the issue and fix it.  

Regards
Maurice

vudinh98, 2020-12-14:
Thanks a lot! It's working.
Please notify if this problem solved in next commit.

mauricema, 2020-12-14:
Hi,
This issue has been fixed by Aiden.  And it has been merged into master.
Please refer to commit id  67cb685614b83864cef1db7b23e71a6444f37b54 for the fix.

Regards,
Maurice

mauricema, 2020-12-14:
The fix has been merged, close it.

closed 2020-12-14 by mauricema