slimbootloader/slimbootloader #938: Initialize Real Non-volatile storage in UefiPayload

vudinh98, 2021-01-04:
I have currently used Slim Bootloader to integrate Uefi Payload. In edk2, I already tried SetVariable() with VendorGUID and Attributes = EFI_VARIABLE_BOOTSERVICE_ACCESS | EFI_VARIABLE_NON_VOLATILE | EFI_VARIABLE_RUNTIME_ACCESS, but when reset the board, GetVariable() get EFI_NOT_FOUND.

I simulate create and get new EFI Variable pretty well on WinHost Emulator, but when I simulate on QEMU and UP Squared board, the created variable does not exist in next reboot.

I'd like to know if this is EDK2 bug or slim bootloader bug. Please let me know. Thank you!

mauricema, 2021-01-04:
Thank you for reporting this issue.

For UEFI payload variable, it has nothing to do with SBL.  It is the expected behavior since UEFI payload in EDK2 uses EmulatorVariable driver by default to make it generic.  It means it uses memory to emulate the behavior of the flash, and the content will be lost after reset or power cycles.  You can try to enable the real flash variable driver in UEFI payload, but that will need some extra work to enable SMM infrastructure in EDK2 and develop/port some platform specific SPI flash SMM drivers.  We do have internal UEFI payload solution which provides real UEFI variable support,  but that is not currently under open source yet.  If you need it, you can contact your Intel representative to get it.  Or you can leverage required drivers into UEFI payload from other open source EDK2 projects such as MinPlatform.

Thanks
Maurice

vudinh98, 2021-01-05:
Thanks a lot!
I will work on it.

mauricema, 2021-01-06:
Sure.   Close this issue for now.

closed 2021-01-06 by mauricema