slimbootloader/slimbootloader #943: [APL] SBL + UEFI paylod assering on PciBusDxe

andreyv1978, 2021-01-06:
It is been a long time since the last time I've used Slimbootloader.
I've received a new redesigned board (customised APL), which do not requires Spread Spectrum to be disabled
The last SBL  + UEFI revision build used in September 2020 (it was a last merge I've made for both SBL + UEFI)
I can confirm the build from September works as expected, verified DEBUG and RELEASE build

I've merged SBL to the latest code and UEFI payload to the today code. However the boot is not working anymore
It is asserted in the UEFI payload  
Found PCI VGA device
[GraphicsOutputDxe]: GraphicsDeviceInfo HOB:
  VendorId = 8086, DeviceId = 5A84,
  RevisionId = FF, BarIndex = 1,
  SubsystemVendorId = FFFF, SubsystemId = FFFF
ASSERT [PciBusDxe] d:\tfs\softwarernd\edk2\MdeModulePkg\Bus\Pci\PciBusDxe\PciIo.c(1826): ((BOOLEAN)(0==1))

I am ataching a full debug 
[sbl-assert.log](https://github.com/slimbootloader/slimbootloader/files/5776536/sbl-assert.log)

Do you have an idea why is that failing now ? Is it something I should ask from UEFI community ?

mauricema, 2021-01-06:
In your SBL resource log: 
  BarType-1: Base 0x0000000000001000 Length 0x0000000000003000
  BarType-3: Base 0x0000000080000000 Length 0x0000000002F00000
  BarType-4: Base 0x00000000A0000000 Length 0x0000000040000000
In your EDK2 resource log:
           Bus: 0 - 8 Translation=0
            Io: 1000 - EFFF Translation=0
           Mem: 80000000 - CFFFFFFF Translation=0
    MemAbove4G: FFFFFFFFFFFFFFFF - 0 Translation=0
          PMem: FFFFFFFFFFFFFFFF - 0 Translation=0
   PMemAbove4G: 400000000 - 7FFFFFFFF Translation=0
They are not aligned.    You MEM32 should be  from "80000000 to DFFFFFFF" instead of   "80000000 to CFFFFFFF"

I believe you used the hard coded resource in UEFI payload as we mentioned in UEFI Payload AdjustRootBridgeResource()
Since your resource is now bigger, you will need to adjust accordingly.
If you have:
   Mem ->Limit = 0xCFFFFFFF;
It needs to be adjusted to:
   Mem ->Limit = 0xDFFFFFFF;

Thanks
Maurice

andreyv1978, 2021-01-07:
Thank you for finding out 
It is something that shouldn't happen, the new design has no memory increases
Unfortunately, I do not have the old board to check out previous logs
I would need to investigate first what is wrong with a BAR settings on the new board

mauricema, 2021-01-07:
It might not be your BAR setting changed.  Since the new SBL also adjusted some of the PCI resource alignment calculation.  It might also cause some source changes.   The PCIE MMCFG base is at 0xE0000000.   The MEM32 limit 0xDFFFFFFF should be the right value in any case.

andreyv1978, 2021-01-08:
thank you for the update, it is working now

closed 2021-01-08 by andreyv1978