slimbootloader/slimbootloader #959: [EHL][SBL][Container] Where can I download .bmap & .wic.bz2 for WICing Yocto image to pendrive?

Jiayan5877, 2021-01-14:
Dear Sir,

I try to WIC Yocto image to pendrive by following chapter 7 create container boot image 7.2 WIC Yocto image & Copy container image in Sbl_Ehl_ReleaseNotes_Beta3_ER.pdf
According to step 1. Download .bmap & .wic.bz2 from EHL repo and copy to linux machine, I need to download image.wic.bmap and image.wic.bz2 but I cannot make sure how to get EHL repo.
Could you provide the path or web site of EHL repo then I can download .bmap & .wic.bz2?

Best Regards,

ywho79, 2021-01-15:
Hi Jiayan5877,

You can generate the images following the Yocto Project BKC guide. The SBL release notes will also be corrected in the upcoming release.

Jiayan5877, 2021-01-15:
Dear ywho79,

Thank for your reply.
May I know the ID number of Yocto Project BKC guide?

Best Regards,

ywho79, 2021-01-15:
Hi Jiayan5877,

I suppose you have an Intel representative contact for this program? He/She will be able to provide you the details and information you need as EHL is still not openly available to general public yet.

Thanks

Jiayan5877, 2021-01-18:
Dear ywho79,

Thank for your suggestion.

Best Regards,

closed 2021-01-18 by Jiayan5877