slimbootloader/slimbootloader #963: [APL] RAM shrinks after SBL merge to the new changeset

andreyv1978, 2021-01-15:
Hi

I've made a merge of the latest changes into my customized build for [APL] board based LeafHill 
The merge up to the changeset: 29ecb88d147d1c105251489ec0dec02f5e2a49e1 (Fix ModuleGuid print in FPDT record). 
The previous build made up to the commit 302bb701def63114156abfce0333df6526f19833 (Always set framebuffer cache attribute)
I've not changed any memory settings the only diff are the merge from the SBL git. Unfortunately, the RAM shrinks from 4Gb to 2Gb
I see the clear 2G in the log now, but only when Yocto loads (sbl-debug-ram-4G.-oldsbllog.log)
Effectively, the OS is the same on both the only change is SBL bios. Can you find what is wrong ?

Maybe it is a different problem but put it here for the moment:
the 2G log prints on the line 2666 
pci 0000:00:02.0: can't claim BAR 2 [mem 0xd0000000-0xdfffffff 64bit pref]: no compatible bridge window

The 4G log prints in line 2533:
pci 0000:00:02.0: BAR 2: assigned to efifb

Looks like the BAR2 failed to be set after merge

Can you please help to find the problem

[sbl-debug-ram-4G.-oldsbllog.log](https://github.com/slimbootloader/slimbootloader/files/5821128/sbl-debug-ram-4G.-oldsbllog.log)
[sbl-debug-ram-2G.log](https://github.com/slimbootloader/slimbootloader/files/5821130/sbl-debug-ram-2G.log)

Andrey

mauricema, 2021-01-20:
There are several different issue here.

1. Memory change from 4GB to 2GB.

If you check the boot log from SBL, SBL did report full 4GB memory in your log.
Graphics Info: 1920 x 1080 x 32 @ 0xA0000000
MEM: 0000000000000000 00000000000A0000 00 1
MEM: 00000000000A0000 0000000000060000 00 2
MEM: 0000000000100000 000000000FF00000 00 1
MEM: 0000000010000000 0000000002151000 00 2
MEM: 0000000012151000 00000000679B3000 00 1
MEM: 0000000079B04000 0000000000500000 01 2
MEM: 000000007A004000 0000000000068000 00 3
MEM: 000000007A06C000 0000000000008000 00 4
MEM: 000000007A074000 0000000000B8C000 00 2
MEM: 000000007AC00000 0000000000400000 00 2
MEM: 000000007B000000 0000000000800000 00 2
MEM: 000000007B800000 0000000004800000 00 2
MEM: 00000000FF800000 0000000000800000 00 2
MEM: 0000000100000000 0000000080000000 00 1
But OS only seems 2GB.    It think it still points to your PCI resource issue.

In you config,  it needs a pretty bit PCI MEM32 space, and the system cannot satisfy it.   In previous discussion, I mentioned to move PCI MEM32 limit from CFFFFFFF to DFFFFFFF,  that was not correct since 0xD0000000 is already allocated for chipset use and cannot be used by PCI device.   You will only have 0x80000000-0xCFFFFFFF MEM32 available to use.  But the actual PCI resource request is more than that.  See below:
  BarType-1: Base 0x0000000000001000 Length 0x0000000000003000
  BarType-3: Base 0x0000000080000000 Length 0x0000000001700000
  BarType-4: Base 0x00000000A0000000 Length 0x0000000040000000

If you check your old log, you will see some resource was moved to MEM64 instead, like below.  That makes it possible to satisfy the resource requirements.   Somehow, the new SBL log does not use PCI MEM64 at all.   Below is from your old log:
  BarType-1: Base 0x0000000000001000 Length 0x0000000000003000
  BarType-3: Base 0x0000000080000000 Length 0x0000000001700000
  BarType-4: Base 0x00000000A0000000 Length 0x0000000020100000
  BarType-6: Base 0x0000000400000000 Length 0x0000000010000000

My question is:   Did you enabled PCI MEM64 allocation at all in the new SBL code ?

mauricema, 2021-01-20:
Please check the value for self._PCI_ENUM_DOWNGRADE_PMEM64 in your BoardConfig.py.  
It needs to be set to 0 in order to use PCI MEM64.

andreyv1978, 2021-01-20:
Hi

I've merged everything to the latest 0d91f28053435354d6de2b4353d3ba7b121337ad just to be on the safe side
Also confirmed self._PCI_ENUM_DOWNGRADE_PMEM64 = 0
still the same result:

see the BoardConfig.py
[BoardConfig.py.txt](https://github.com/slimbootloader/slimbootloader/files/5842598/BoardConfig.py.txt)

Debug output:
[debug-20-01-2021.log](https://github.com/slimbootloader/slimbootloader/files/5842609/debug-20-01-2021.log)

My concern self._PCI_ENUM_DOWNGRADE_PMEM64 is not picked up with a new changes

Andrey

mauricema, 2021-01-20:
You will need to add  self._PCI_ENUM_DOWNGRADE_BUS0=0 as well to downgrade device for bus 0.

Currently, there is an known issue with paging table above 4GB.   As a workaround, you can just comment out the line CreateIdentifyMappingPageTables() in PayloadEntryLib.c file.

myguitar, 2021-01-21:
@andreyv1978 It looks you are using UefiPayload. Can you try a patch on top of your UefiPayload code base?
https://patch-diff.githubusercontent.com/raw/tianocore/edk2/pull/1375.patch (https://github.com/tianocore/edk2/pull/1375).

andreyv1978, 2021-01-21:
> (tianocore/edk2#1375 <https://github.com/tianocore/edk2/pull/1375>).

Hi

Should I use only UefiPayload patch or all together like:

1. self._PCI_ENUM_DOWNGRADE_BUS0=0 (SBL)
2. comment out the line CreateIdentifyMappingPageTables() in
PayloadEntryLib.c  (SBL)
3. UefiPayload patch 1375 (edk2)

or  only
1.  self._PCI_ENUM_DOWNGRADE_BUS0=0 (SBL)
2. UefiPayload patch 1375 (edk2)

Andrey

mauricema, 2021-01-21:
You can apply all of the 3 changes.   
BTW, the CreateIdentifyMappingPageTables() is for OsLoader payload only.

andreyv1978, 2021-01-22:
Hi

I've used self._PCI_ENUM_DOWNGRADE_BUS0=0 and UEFI patch to generate a buid
Looks all fixed Bar2 allocated and memory 4G again , closing it now

The only new messages from UEFI, should it be removed ?
!!! DEPRECATED INTERFACE !!! VariableLockRequestToLock() will go away soon!
!!! DEPRECATED INTERFACE !!! Please move to use Variable Policy!
!!! DEPRECATED INTERFACE !!! Variable: 8BE4DF61-93CA-11D2-AA0D-00E098032B8C PlatformLangCodes
!!! DEPRECATED INTERFACE !!! VariableLockRequestToLock() will go away soon!
!!! DEPRECATED INTERFACE !!! Please move to use Variable Policy!
!!! DEPRECATED INTERFACE !!! Variable: 8BE4DF61-93CA-11D2-AA0D-00E098032B8C LangCodes
!!! DEPRECATED INTERFACE !!! VariableLockRequestToLock() will go away soon!
!!! DEPRECATED INTERFACE !!! Please move to use Variable Policy!
!!! DEPRECATED INTERFACE !!! Variable: 8BE4DF61-93CA-11D2-AA0D-00E098032B8C BootOptionSupport
!!! DEPRECATED INTERFACE !!! VariableLockRequestToLock() will go away soon!
!!! DEPRECATED INTERFACE !!! Please move to use Variable Policy!
!!! DEPRECATED INTERFACE !!! Variable: 8BE4DF61-93CA-11D2-AA0D-00E098032B8C HwErrRecSupport
!!! DEPRECATED INTERFACE !!! VariableLockRequestToLock() will go away soon!
!!! DEPRECATED INTERFACE !!! Please move to use Variable Policy!
!!! DEPRECATED INTERFACE !!! Variable: 8BE4DF61-93CA-11D2-AA0D-00E098032B8C OsIndicationsSupported
Variable Driver Auto Update Lang, Lang:eng, PlatformLang:en Status: Success
!!! DEPRECATED INTERFACE !!! VariableLockRequestToLock() will go away soon!
!!! DEPRECATED INTERFACE !!! Please move to use Variable Policy!
!!! DEPRECATED INTERFACE !!! Variable: 8BE4DF61-93CA-11D2-AA0D-00E098032B8C PlatformRecovery0000

[SBL-debug-22012021.log](https://github.com/slimbootloader/slimbootloader/files/5856563/SBL-debug-22012021.log)

Andrey

closed 2021-01-22 by andreyv1978