slimbootloader/slimbootloader #988: [APL] i915 kernel crashes after merge, when monitor is not connected

andreyv1978, 2021-01-26:
Hi

I am still trying to get the merge working the same way as my old September branch on the  customized APL platform
It looks I have one more outstanding issue 
I have a number of i915 driver kernel crashes while booting SBL/Yocto without a monitor connected. I've made a few experiments and it looks like the hob assigned when a monitor is not connected is too big in SBL updated code 
see the logs:
Log when a monitor is not connected while booting
[Jan-branch-merge-no-monitor.log](https://github.com/slimbootloader/slimbootloader/files/5872970/Jan-branch-merge-no-monitor.log)

Log when a monitor is connected while booting
[Jan-branch-merge-monitor-connected.log](https://github.com/slimbootloader/slimbootloader/files/5872971/Jan-branch-merge-monitor-connected.log)

when the monitor is not connected I can see a following:
Graphics Info: 1440 x 2560 x 32 @ 0xA0000000:  instead of 1920 x 1080 x 32 @ 0xA0000000

also MRC data NOT found by comparing to what I see when the monitor is connected 
MRC data was found @ 0xFEF40400(Param) and 0xFEF40000(Var)

However, in the old September branch when the monitor is not connected I have much smaller resolution allocated
[Sep-branch-no-monitor.log](https://github.com/slimbootloader/slimbootloader/files/5872980/Sep-branch-no-monitor.log)

like: Graphics Info: 1024 x 768 x 32 @ 0xA0000000

Is anything I can adjust to make the SBL working without monitor been connected to the board ? The monitor is only optional for debugging, we can disable it completely in the BIOS ? But it would be nice to have an option when SBL in the Debug build we can connect a monitor and investigate
The current SBL September  code is allowing us to use a monitor with and without. 

Andrey

mauricema, 2021-01-26:
In the old log, you have "Select VBT ImageId 0x00000003"
In the new log,  you have "Select VBT ImageId 0x00000001"

So you are using a different VBT now.   Please ensure you select correct VBT in your build.  I assume you added a new VBT in the old tree, and it was not properly integrated into the new SBL.
Please check  BoardConfig.py to add you new VBT binary.
self._MULTI_VBT_FILE      = {1:'Vbt.dat', 2:'Vbt_Up2.dat', 3:'YourNewVbt.dat'}
Then please set the VbtImageId to 3 in your Board specific dlt file.
GEN_CFG_DATA.VbtImageId  | 3

andreyv1978, 2021-01-27:
thank you for spot on, the GEN_CFG_DATA.VbtImageId | 3 missed, thank you for your help

closed 2021-01-27 by andreyv1978