slimbootloader/slimbootloader #994: [EHL] [SBL] no fwupdate command in SBL shell

Jiayan5877, 2021-01-28:
Dear Sir,

I want to trigger firmware update in SBL shell but no fwupdate command in built-in shell.
May I know how to add or enable fwupdate command in built-in shell?

The source code of EHL's SBL is from sbl-ehl-beta3_er.zip
UEFI payload is the default payload.

Best Regards,

raghavag, 2021-01-28:
Hello,

Fwupdate command is available in SBL Shell, if you are using UEFI payload as default, then you are booting to UEFI shell by default. Fwupdate command is not available in UEFI shell.

Please make sure you are booting to SBL shell and try the fwupdate command again.

If you are planning to use UEFI payload, SBL support triggering firmware update from OS, please refer to https://slimbootloader.github.io/security/firmware-update.html#triggering-firmware-update

Thank you,
Raghava

Jiayan5877, 2021-01-29:
Dear Raghava,

Thank for your reply.
May I know how to boot to SBL shell?
According to following website, To enter shell interface, press any key after reboot and before OS is launched.
https://slimbootloader.github.io/developer-guides/shell.html?highlight=sbl%20shell
I tired OsLoader payload or UEFI payload, but I cannot enter SBL shell by this method.

Best Regards,

ywho79, 2021-01-29:
Hi Jiayan5877,

For EHL Customer Reference Board you can use the jumper J2E1 [pins 2-4] on the board to switch between UEFI payload and OsLoader payload. To get access to SBL shell you need to have a debug build of the EHL SBL.

Hope this helps.

Jiayan5877, 2021-01-29:
Dear ywho79,

Does to boot to SBL shell need to use OsLoader payload?
I set following setting but I still cannot enter SBL shell by pressing any key after reboot and before OS is launched.
1. Set GEN_CFG_DATA.PayloadId   | '' or GEN_CFG_DATA.PayloadId   | '0' in CfgData_xt_lotgCrb.dlt
2. Set self.ENABLE_PRE_OS_CHECKER   = 1 in BoardConfig.py
3. Build UefiPld.fd into SBL image:
python BuildLoader.py build qemu -p "OsLoader.efi:LLDR:Lz4;UefiPld.fd:UEFI:Lzma"
Are these settings right? 

BTW, I don't have  EHL Customer Reference Board.

Best Regards,

ywho79, 2021-01-29:
Hi Jiayan5877,

Yes you need OsLoader payload to launch SBL shell.
Looks like you're building for QEMU instead of EHL?

Can you try booting without any OS image first with a debug build? SBL should boot straight to the SBL shell if no OS image found.

Hope this helps

Jiayan5877, 2021-01-29:
Dear ywho79,

I am sorry that qemu is my typo.
Here is right command for EHL debug build.
py sblBuild.py build ehl -p "OsLoader.efi:LLDR:Lz4;UefiPld.fd:UEFI:Lzma"

About QEMU, I can see SBL shell in Command Prompt and SLIM Bootloader Logo in QEMU emulator.
About EHL, I only see SLIM Bootloader Logo on Monitor.

Best Regarts,

ywho79, 2021-01-29:
Hi Jiayan5877,

May I know how are you booting the EHL SBL that you generated as you're not using the EHL Customer Reference Board?

Jiayan5877, 2021-01-29:
Dear ywho79,

To modify MEMORY_CFG_DATA.SpdAddressTable and GEN_CFG_DATA.PayloadId | 'UEFI' can boot to Win10 on my EHL M/B.
For hardware configuration, I modify other setting by ConfigEditor.py.

Best Regards,

Jiayan5877, 2021-01-29:
Dear Mauricema,

My M/B doesn't have debug serial port 2 but I will try to rework it.
I will reply my test result.
Thank for your comment.

Best Regards,

mauricema, 2021-01-29:
That explained.  If you have serial port, you will see SBL Shell there.

If no serial port is there,  maybe you can try to enable local console (USB keyboard + GFX).  This is for experimental only. 
To do so, you will need to set below in your BoardConfig.py.
self.CONSOLE_IN_DEVICE_MASK  = 0x00000003
self.CONSOLE_OUT_DEVICE_MASK = 0x00000003

Regards
Maurice

Jiayan5877, 2021-01-29:
Dear Mauricema,

SBL shell is on debug serial port 2 and monitor after modifying below settings in BuildLoader.py and setting Payload to OSLoader.
self.CONSOLE_IN_DEVICE_MASK = 0x00000003
self.CONSOLE_OUT_DEVICE_MASK = 0x00000003
The fwupdate command is in SBL shell.
Thank for your help.

Best Regards,

closed 2021-01-29 by Jiayan5877