slimbootloader/slimbootloader #1030: Fail to boot from Yockto Image with container.bin and has error "Registering container failed due to Security Violation"

Jiayan5877, 2021-02-17:
Dear Sir,

I follow 7.0 Create Container Boot Image in Sbl_Ehl_ReleaseNotes_Beta3_ER.pdf to boot from Yockto Image with container.bin but has error "Registering container failed due to Security Violation".
EHL uses 3k signing format.

Here is my step.
1. Generate container.bin File
GenContainer.py create -cl CMDL:cmdline.txt KRNL:bzImage INRD:initrd -k ContainerTestKey_Priv_RSA3072.pem -t CLASSIC -o container.bin
Here bzImage is the same as vmlinuz
MasterTestKey_Priv_RSA3072.pem, ContainerTestKey_Priv_RSA3072.pem and ContainerCompTestKey_Priv_RSA3072 have the same the same error.
2. WIC Yocto Image & Copy container.bin File
3.  Change boot option 0 
           - !expand { BOOT_OPTION_TMPL : [ 0 ,   0    ,  16 ,    5   ,   0   ,    0  ,     1 ,     1 , 'container.bin' ] }
4. Set payPayload to OsLoader payload
5. Attach USB drive with Yocto image to the target board
6. Power on the board and boot Yocto but has error "Registering container failed due to Security Violation".

Here is a part of log
[ContainerTestKey_Priv_RSA3072_Log.txt](https://github.com/slimbootloader/slimbootloader/files/5993972/ContainerTestKey_Priv_RSA3072_Log.txt)

Is my setting wrong?
Could you help this case?

Best Regards,

mauricema, 2021-02-18:
Since  you are signing an OS boot image, you should use  OS1_TestKey_Pub_RSA3072_priv.pem key instead.
For other containers, you can use the container key.

lsubash, 2021-02-18:
Please use this command for creating OS container,
GenContainer.py create -cl CMDL:cmdline.txt KRNL:bzImage INRD:initrd -k ** OS1_TestKey_Priv_RSA3072.pem** -t CLASSIC -o container.bin -a RSA3072_PSS_SHA2_384.

While loading the OS container, SBL would check OS public hashes which is registered in hash store.

Jiayan5877, 2021-02-19:
Dear IsuBash & Mauricema,

Thank for your help.

The following commands can work.
GenContainer.py create -cl CMDL:cmdline.txt KRNL:bzImage INRD:initrd -k OS1_TestKey_Pub_RSA3072_priv.pem -t CLASSIC -o container.bin -a RSA3072_PSS_SHA2_384
or
GenContainer.py create -cl CMDL:cmdline.txt KRNL:bzImage INRD:initrd -k OS1_TestKey_Pub_RSA3072_priv.pem -t CLASSIC -o container.bin

May I know the difference between with and without "-a RSA3072_PSS_SHA2_384"?
Authentication algorithm has choices=['SHA2_256', 'SHA2_384', 'RSA2048_PKCS1_SHA2_256', 'RSA3072_PKCS1_SHA2_384', 'RSA2048_PSS_SHA2_256', 'RSA3072_PSS_SHA2_384', 'NONE'], default='NONE'.
Could you provide information about how to choose them?

Best Regards,

lsubash, 2021-02-19:
Hi Jiayan,

>>May I know the difference between with and without "-a RSA3072_PSS_SHA2_384"?
     As you RSA3K keys default auth type is set to RSA3072_PSS_SHA2_384. So, there should not be delta with/without RSA3072_PSS_SHA2_384.

There are various authentication mechanisms supported while container generation,
 SHA2_256/ SHA2_384: Does hash verification while loading the container. Private key need not be used while using these mechanisms.
RSA3072_PKCS1_SHA2_384/RSA3072_PSS_SHA2_384: These auth types are used when container is signed with RSA3K private key and for doing RSA verification. SHA384 alg is used for signing hashes. PKCS1, PSS are RSA padding schemes.
Similarly RSA2048_PKCS1_SHA2_256/RSA2048_PSS_SHA2_256 are for RSA2K keys.

As per crypto recommendation, its recommended use RSA3K/SHA384 & PSS. i.e RSA3072_PSS_SHA2_384.
Also, EHL uses RSA3K/SHA384 by default.

Jiayan5877, 2021-02-19:
Dear Isubash,

Thank for your reply.

May I reconfirm your explanation?
As the default auth type of XX_XXX_Priv_RSA3072.pem in "SblKeys" folder is set to RSA3072_PSS_SHA2_384.
When one of RSA3K keys(XX_XXX_Priv_RSA3072.pem)  in "SblKeys" folder is used by GenContainer.py, I don't need to execute "-a RSA3072_PSS_SHA2_384".
Is my understanding right?

Best Regards,

lsubash, 2021-02-19:
Hi Jiayan,

When RSA3K private key is used with container generation, default auth type is RSA3072_PSS_SHA2_384.
Similarly for RSA3K private key auth type would be defaulted to auth type is RSA2048_PSS_SHA2_256.

Yes, You need not execute "-a" option when you use RSA3K private key. You would need to use "-a" when a different auth type is needed. 
Note: You cannot use RSA3K private key & RSA2048_PSS_SHA2_256 combination.

Jiayan5877, 2021-02-22:
Dear Isubash,

Thank for your comment.
May I correct one typo?
Similarly for RSA**2**K private key auth type would be defaulted to auth type is RSA2048_PSS_SHA2_256.
Is the subject RSA2K private key in this sentence?

Best Regards,

lsubash, 2021-02-22:
> Dear Isubash,
> 
> Thank for your comment.
> May I correct one typo?
> Similarly for RSA**2**K private key auth type would be defaulted to auth type is RSA2048_PSS_SHA2_256.
> Is the subject RSA2K private key in this sentence?
> 
> Best Regards,

Yeah Jiyan. It’s typo. Thanks for correcting.

Jiayan5877, 2021-02-22:
Dear Isubash,

Thank for your help.
I don't have other question for this topic.

Best Regards,

closed 2021-02-22 by Jiayan5877