slimbootloader/slimbootloader #1033: Fail to update PseFw from shell and has error "Updating component IPFW failed with status = Device Error".

Jiayan5877, 2021-02-23:
Dear Sir,

I follow 3.3 Generating Container component binary and 3.5.2 Generate Capsule image to update Config Data and Container Comp in EHL_FirmwareUpdate.pdf to update PseFw from shell but has error "Updating component IPFW failed with status = Device Error".
EHL uses 3k signing format.

Here is my step.
1. Generating PseFw binary
python ./SblOpen/BootloaderCorePkg/Tools/GenContainer.py sign -f ./ifwi_stitch/IPFW/PseFw.bin -o PseFwu.bin -c lz4 -a RSA3072_PSS_SHA2_384 -k ./SblKeys/ContainerTestKey_Priv_RSA3072.pem -td ./SblOpen/BaseTools/Bin/Win32
2. Generate Capsule image with PseFw binary
python ./SblOpen/BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p PSEF:IPFW PseFwu.bin -k ./SblKeys/FirmwareUpdateTestKey_Priv_RSA3072.pem -o FwuImage.bin
3. Copy FwuImage.bin into root directory on FAT partition of a USB key
4. Boot and press any key to enter SBL shell
5. Type command fwupdate from shell

Here is error log
[fwupdate_PSEFW_Fail_02 with -c lz4.txt](https://github.com/slimbootloader/slimbootloader/files/6026406/fwupdate_PSEFW_Fail_02.with.-c.lz4.txt)

Is my setting wrong?
Could you help this case?

Best Regards,

jinjhuli, 2021-02-23:
Hi Jiayan5877,

For the 1st step to generate PseFw binary, can you please try to use this key instead?
-ContainerCompTestKey_Priv_RSA3072.pem

python ./SblOpen/BootloaderCorePkg/Tools/GenContainer.py sign -f ./ifwi_stitch/IPFW/PseFw.bin -o PseFwu.bin -c lz4 -a RSA3072_PSS_SHA2_384 -k ./SblKeys/ContainerCompTestKey_Priv_RSA3072.pem -td ./SblOpen/BaseTools/Bin/Win32

Regards,
Jin Jhu

Jiayan5877, 2021-02-23:
Dear Jin Jhu,

You are right.
PseFw can be updated after changing key to ContainerCompTestKey_Priv_RSA3072.pem.

Thank for your help.

Best Regards,

closed 2021-02-23 by Jiayan5877