slimbootloader/slimbootloader #1039: [APL] Source Level Debugging with Intel(R) UDK Debugger

vudinh98, 2021-02-25:
I have currently used Intel(R) UDK Debugger Tool to debug SBL that boot to OsLoader on UP2 Board and I also set **self.ENABLE_SOURCE_DEBUG=1** and  **self.SKIP_STAGE1A_SOURCE_DEBUG  = 1**. I used a USB UART to connect UP2 to PC. I start Intel UDK Debugger Tool and power up UP2 Board, but the Tool didn't response anything and show a message "Debuggee not connected".

Beside that, UEFIPayload also support Source Level Debug, so Intel UDK Debugger Tool work well when boot to UEFIPayload.

I followed your guideline in https://slimbootloader.github.io/developer-guides/debugging-with-udk.html
Please tell me know how to fix this problem. Thank you!

mauricema, 2021-02-25:
Please ensure your SBL has UART debug output before you try to start UDK debugger tool.
You should be able to see "Send INIT break packet and try to connect to HOST ..."  message from UART when ENABLE_SOURCE_DEBUG is on.
If you cannot see this output, it might be some issue on the SBL config or your UART host config.

vudinh98, 2021-02-26:
My UART Host worked well. The reason is I set ENABLE_SOURCE_DEBUG to 1 at wrong platform. 

But when I set ENABLE_SOURCE_DEBUG at my desired platform (APL) and build again, the compiler show me error: _.../BootloaderCommonPkg\Library\DebugAgentLib\DebugAgent.h(13): fatal error C1083: Cannot open include file: 'Library/LocalApic.h': No such file or directory_. I also checkout my SBL to commit **4eeb962977209b7979326e11762187c3708262ce**, but it still show the same error. 

I attach build result at build.log file below.
Please tell me know how to fix this problem. Thank you!
[build.log](https://github.com/slimbootloader/slimbootloader/files/6048303/build.log)

Praveenhp7, 2021-02-26:
Please can you try this patch to fix the build error  (PR: https://github.com/slimbootloader/slimbootloader/pull/1040 ) 
 https://patch-diff.githubusercontent.com/raw/slimbootloader/slimbootloader/pull/1040.patch

vudinh98, 2021-03-01:
I followed your solutions and fixed above error, but another error related to **CmdBoot.c** appeared.  

I attach build result at build.log file below.
Please tell me know how to fix this problem. Thank you!
[build1.log](https://github.com/slimbootloader/slimbootloader/files/6058826/build1.log)

Praveenhp7, 2021-03-01:
I do not see the build error from my side with the latest commit **de6bc23** and SOURCE_DEBUG is set to 1. 
- from your build logs, it seems you are using Visual Studio 2019. can you try compile using recommended version VS15 or VS17? https://slimbootloader.github.io/getting-started/build-host-setup.html#building-on-windows

vudinh98, 2021-03-01:
I found a problem in BuildUitily.py, line 224: **for vs_ver in ['2019', '2017']:** and I changed it to **for vs_ver in ['2017']:** because my system has VS2019 and SBL found it and chose VS2019 instead of VS2015x86.

I experienced in UDK Debug Tool with SBL successful, but I can only debug in Stage1B. So I wonder whether SBL supports Souce level Debug in Stage1A and Stage2?

Please let me know. Thanks you!

Praveenhp7, 2021-03-02:
currently, on the APL platform Stage1A source debug is skipped due to IBBL Size ( 32KB )  limitation.

vudinh98, 2021-03-02:
Does other Platform like TGL support debug in stage 1A?
How about source level debug in Stage 2? When I finished debugging stage 1B, SBL went to Shell directly.

mauricema, 2021-03-02:
Stage1A in other platforms runs from Flash directly, so there is no size limitation.   For APL, it runs starting from CSME RAM, and only 32KB is allocated.  Enabling full debug will not fit into 32KB.  
Stage2 debug should be fine.   After you return from PrepareStage2() in Stage1B,  I think you will need to setup a breakpoint for some function in Stage2 in order to break.

vudinh98, 2021-03-03:
Thanks you a lot!

closed 2021-03-03 by vudinh98