slimbootloader/slimbootloader #1050: How can I boot the Linux kernel into the SPI area using Qemu Emulator.

ionutnechita, 2021-03-17:
Hi Slimbootloader Team,

I tried to start with the idx set on SPI, but it can't find my iasimage1.bin image.

Reference Page for SPI / MEM:
https://slimbootloader.github.io/how-tos/add-fastboot.html?highlight=fastboot

mauricema, 2021-03-18:
Please be more specific on the issue you had.   
You might need to attach the build log if it is a build related issue, or boot log if it is a boot related issue.

rprangar, 2021-03-18:
Documentation about booting from SPI is here - https://slimbootloader.github.io/developer-guides/flash-boot.html

ionutnechita, 2021-03-19:
Hi @rprangar,

For create container or iasimage.bin.
What key should be used? 
I try with this: OS1_TestKey_Priv_RSA2048.pem, but not working.

I try with this:
python3 iasimage create -o iasimage.bin -d ./slimboot/BootloaderCorePkg/Tools/Keys/OS1_TestKey_Priv_RSA2048.pem  -i 0x40000 cmdline.txt kernel initrd
...
======== Try Booting with Boot Option 4 ========
BootMediumPciBase(0x1000000)
Getting boot image from MEM
Try to find boot partition
Part 00: 0xFFB60000--0xFFF60000, LBA count: 0x400000
SPI BIOS region: (1 logical partitions)
Find partition success
Init File system
Requested FsType 3. Assume RAW format partition.
BootSlot = 0x0
Load image from SwPart (0x0), LbaAddr(0x0)
LoadBootImage ImageType-0 Image
ParseBootImage ImageType-0
HASH verification for usage (0x00000800) with Hash Alg (0x2): Security Violation
First 48 Bytes Input Data
  00000000: A8 A9 C5 10 A8 CC F3 8A-B9 8F 72 03 80 70 7F 57  *..........r..p.W*
  00000010: 39 C7 B9 70 92 22 04 DD-97 34 36 9F EC 6C EF 3B  *9..p."...46..l.;*
  00000020: 92 C2 41 5C 23 BC 60 3E-F2 C8 4B 86 97 A2 4F 40  *..A\#.`>..K...O@*
Last 48 Bytes Input Data
  00000000: 95 1D 11 D4 38 4B A3 53-1F BC 27 3A 3F 02 3B 78  *....8K.S..':?.;x*
  00000010: 43 F1 2F CF 81 FD 46 D0-04 83 64 4C 79 C5 A8 39  *C./...F...dLy..9*
  00000020: 0E 93 CA 96 D2 CD 03 A9-2B BE D5 CB 00 01 00 01  *........+.......*
Image Digest
  00000000: EE C5 AA F8 43 9A D3 CA-A6 DC A2 72 9B ED B9 30  *....C......r...0*
  00000010: 0C 44 AA 4F 6E F9 DB D8-CB EC DA DD 45 B2 30 17  *.D.On.......E.0.*
  00000020: E4 AA C1 8B FE 34 A4 7C-6D A1 39 F1 F6 54 AA 0B  *.....4..m.9..T..*
HashStore Digest
IAS image verification failed!
Image given is not a valid IAS image
ParseLoadedImage: Status = Load Error
Failed to Parse Boot Image
Payload normal heap: 0x2000000 (0x10F000 used)
Payload reserved heap: 0x4000 (0x0 used)
Payload stack: 0x10000 (0xE00 used)

mauricema, 2021-03-19:
Are you using latest SBL code ?
If so, the keys should be located at  ./SblKeys/OS1_TestKey_Pub_RSA2048_priv.pem by default if you did not specify a different folder.   It seems you are using the key from old SBL tree BootloaderCorePkg/Tools/Keys.

ionutnechita, 2021-03-19:
I view this, with master branch:

python3 ./BuildLoader.py build qemu
make: Leaving directory '/dev/shm/slimboot/BaseTools'
!!! ERROR: SBL_KEY_DIR is set to directory /dev/shm/slimboot/BootloaderCorePkg/Tools/Keys/ does not exist!!!

!!! PRE-REQUISITE: Path to SBL_KEY_DIR has to be set with SBL KEYS DIRECTORY !!!
!!! Generate keys using GenerateKeys.py available in BootloaderCorePkg/Tools directory !!!
!!! Run $python BootloaderCorePkg/Tools/GenerateKeys.py -k $PATH_TO_SBL_KEY_DIR !!!
!!! Set SBL_KEY_DIR environ with path to SBL KEYS DIR !!!
!!! Windows $set    SBL_KEY_DIR=$PATH_TO_SBL_KEY_DIR !!!
!!! Linux   $export SBL_KEY_DIR=$PATH_TO_SBL_KEY_DIR !!!

mauricema, 2021-03-19:
It seems you have SBL_KEY_DIR set to slimboot/BootloaderCorePkg/Tools/Keys somehow.
Can you try to delete the env variable SBL_KEY_DIR from your current env ?
Then you can try:   python3 ./BuildLoader.py build qemu -k   
This will let SBL re-generate new keys at  SblKeys folder located at the same directory level as slimboot root.
Now you can use  SblKeys/OS1_TestKey_Priv_RSA2048.pem  to sign your image.

mauricema, 2021-03-19:
BTW, not sure which RSA you used for QEMU.  It is configured in SBL QEMU BoardConfig.py.
If you want to use IAS image (only supports RSA2048),  you will need to modify BoardConfig.py to use RSA2048 and SHA2_256 as below.  The default is RSA3072.
        # RSA2048 or RSA3072
        self._RSA_SIGN_TYPE          = 'RSA2048'
        # 'SHA2_256' or 'SHA2_384'
        self._SIGN_HASH              = 'SHA2_256'

Otherwise, you can also use Container if you want to use the default RAS3072 for signing. It supports both RSA2048 and RSA3072.  The command is like below:
python %SBL_ROOT%\BootloaderCorePkg\Tools\GenContainer.py create -cl CMDL:cmdline.txt KRNL:vmlinuz INRD:initrd -t CLASSIC -o IasImage.bin  -k \SlimBoot\SblKeys\OS1_TestKey_Priv_RSA3072.pem -a RSA3072_PSS_SHA2_384

ionutnechita, 2021-03-19:
With this parameter,  -a RSA3072_PSS_SHA2_384, working for create iasimage1.bin.

ionutnechita, 2021-03-19:
I try now for boot and working.

======== Try Booting with Boot Option 4 ========
BootMediumPciBase(0x1000000)
Getting boot image from MEM
Try to find boot partition
Part 00: 0xFFB60000--0xFFF60000, LBA count: 0x400000
SPI BIOS region: (1 logical partitions)
Find partition success
Init File system
Requested FsType 3. Assume RAW format partition.
BootSlot = 0x0
Load image from SwPart (0x0), LbaAddr(0x0)
LoadBootImage ImageType-0 Image
ParseBootImage ImageType-0
Registering container BOOT
HASH verification for usage (0x00000800) with Hash Alg (0x2): Success
SignType (0x2) SignSize (0x180)  SignHashAlg (0x2)
RSA verification for usage (0x00000800): Success
HASH verification for usage (0x00000000) with Hash Alg (0x2): Success
CONTAINER size = 0x1FD570, image type = 0xF3, # of components = 4
COMP:CMDL Success
COMP:KRNL Success
COMP:INRD Success
Unregister done - Success!
SetupBootImage ImageType-0
Assume BzImage...
Found bzimage Signature

Dump normal boot image info:

============ KERNEL SETUP ============
SetupSectorss: 0x1F
RootFlags: 0x1
SysSize: 0x1962F
RamSize: 0x0
VideoMode: 0xFFFF
RootDev: 0x0
Signature: 0xAA55
Jump: 0x6AEB
Header: 0x53726448
Version: 0x20F
SuSwitch: 0x0
SetupSeg: 0x0
StartSys: 0x1000
KernalVer: 0x3200
LoaderId: 0xFF
LoadFlags: 0x1
MoveSize: 0x8000
Code32Start: 0x100000
BootsectorKludge: 0x0
HeapEnd: 0x5410
ExtLoaderVer: 0x0
ExtLoaderType: 0x0
CmdlineSize: 0x800
CmdLinePtr: 0x1E9ED000
Cmd Args: init=/init root=/dev/ram0 rw 3 console=tty0 console=ttyS0,115200
RamDiskStart: 0x1E98A000
RamDisklen: 0x62560
RamDiskMax: 0x7FFFFFFF
KernelAlignment: 0x200000
RelocatableKernel: 0x0
MinAlignment: 0x15
XloadFlags: 0x21
HardwareSubarch: 0x0
HardwareSubarchData: 0x0
PayloadOffset: 0x2C8
PayloadLength: 0x190E74
SetupData: 0x0
PrefAddress: 0x1000000
InitSize: 0xF4C000
HandoverOffset: 0x0

Payload normal heap: 0x2000000 (0x311000 used)
Payload reserved heap: 0x4000 (0x0 used)
Payload stack: 0x10000 (0xEA0 used)

MP Init (Done)
Call FspNotifyPhase(40) ... Success
Call FspNotifyPhase(F0) ... Success
 Id   | Time (ms)  | Delta (ms) | Description
------+------------+------------+----------------------------------
 1000 |       6 ms |       6 ms | Reset vector
 1010 |      32 ms |      26 ms | Stage1A entry point
 1040 |      36 ms |       4 ms | Board PostTempRamInit hook
 1060 |      45 ms |       9 ms | Stage1A continuation
 1080 |      58 ms |      13 ms | Load Stage1B
 10A0 |     143 ms |      85 ms | Verify Stage1B
 10B0 |     147 ms |       4 ms | Decompress Stage1B
 2000 |     147 ms |       0 ms | Stage1B entry point
 2020 |     323 ms |     176 ms | Board PreMemoryInit hook
 2030 |     364 ms |      41 ms | FSP MemoryInit
 2040 |     382 ms |      18 ms | Board PostMemoryInit hook
 2050 |     384 ms |       2 ms | Board PreTempRamExit hook
 2060 |     390 ms |       6 ms | FSP TempRamExit
 2070 |     390 ms |       0 ms | Board PostTempRamExit hook
 2080 |     390 ms |       0 ms | Load Stage2
 2090 |     392 ms |       2 ms | Copy Stage2 to memory
 20A0 |     514 ms |     122 ms | Verify Stage2
 20B0 |     519 ms |       5 ms | Decompress Stage2
 20C0 |     519 ms |       0 ms | Extend Stage2 hash
 20D0 |     520 ms |       1 ms | Rebase Stage2
 3000 |     525 ms |       5 ms | Stage2 entry point
 3010 |     694 ms |     169 ms | Board PreSiliconInit hook
 3020 |     695 ms |       1 ms | Save NVS data
 3030 |     730 ms |      35 ms | FSP SiliconInit
 3040 |     731 ms |       1 ms | Board PostSiliconInit hook
 3050 |     753 ms |      22 ms | Display splash
 3060 |     759 ms |       6 ms | MP wake up
 3080 |     760 ms |       1 ms | MP init run
 3090 |     760 ms |       0 ms | Board PrePciEnumeration hook
 30A0 |     776 ms |      16 ms | PCI enumeration
 30B0 |     777 ms |       1 ms | Board PostPciEnumeration hook
 30C0 |     778 ms |       1 ms | FSP PostPciEnumeration notify
 30D0 |     789 ms |      11 ms | ACPI init
 30E0 |     791 ms |       2 ms | Board PrePayloadLoading hook
 3100 |     791 ms |       0 ms | Load payload
 3110 |     791 ms |       0 ms | Locate payload
 3120 |     797 ms |       6 ms | Copy payload to memory
 3130 |     946 ms |     149 ms | Verify payload
 3140 |    1084 ms |     138 ms | Decompress payload
 3150 |    1084 ms |       0 ms | Extend payload hash
 31A0 |    1084 ms |       0 ms | Board PostPayloadLoading hook
 31B0 |    1087 ms |       3 ms | Decode payload format
 31F0 |    1087 ms |       0 ms | End of stage2
 4000 |    1093 ms |       6 ms | Payload entry point
 4010 |    1109 ms |      16 ms | OS loader main entry
 4020 |    7865 ms |    6756 ms | Shell exit
 4040 |    7871 ms |       6 ms | Load image entry
 4040 |   13044 ms |    5173 ms | Load image entry
 4050 |   13319 ms |     275 ms | Boot device init
 4055 |   13319 ms |       0 ms | Boot device tuning
 4040 |   13323 ms |       4 ms | Load image entry
 4040 |   13329 ms |       6 ms | Load image entry
 4040 |   13333 ms |       4 ms | Load image entry
 4050 |   13333 ms |       0 ms | Boot device init
 4055 |   13333 ms |       0 ms | Boot device tuning
 4060 |   13336 ms |       3 ms | Parse partition info
 4070 |   13353 ms |      17 ms | Load boot images
 4080 |   13409 ms |      56 ms | Verify IAS image
 4080 |   15852 ms |    2443 ms | Verify IAS image
 40A0 |   15855 ms |       3 ms | Process IAS type
 40B0 |   15866 ms |      11 ms | Process ELF/MultiBoot
 40F0 |   15880 ms |      14 ms | FSP ReadyToBoot/EndOfFirmware notify
 4100 |   15880 ms |       0 ms | TPM IndicateReadyToBoot
------+------------+------------+----------------------------------

Starting Kernel ...

Switch to LongMode and jump to 64-bit kernel entrypoint ...

But not observe prompt this.

How to boot with prompt ?

mauricema, 2021-03-19:
Which kernel are you using ?    Maybe you can give a try on this kernel we used for SBL testing.  
https://github.com/slimbootloader/slimbootloader/files/4463548/QemuLinux.zip

ionutnechita, 2021-03-19:
this kernel: 
https://github.com/slimbootloader/slimbootloader/files/4463548/QemuLinux.zip

mauricema, 2021-03-19:
That might be the QEMU command line or version issue.  You can check our QEMU test cases.  
It has Linux boot test and will be checked for every PR on github.    
SlimBoot\SlimFork\Platform\QemuBoardPkg\Script.

ionutnechita, 2021-03-19:
I changed the initialization parameters to qemu and now it works.

> qemu-system-x86_64 -cpu max -machine q35 -m 256 --nographic -boot order=d -pflash Outputs/qemu/SlimBootloader.bin

Boot file:
[QemuBootSPI-MEM.txt](https://github.com/slimbootloader/slimbootloader/files/6172972/QemuBootSPI-MEM.txt)

It went perfectly.
Thank you with all my heart.

mauricema, 2021-03-19:
Glad to see it working for you.
You are welcome!  :-)

ionutnechita, 2021-03-20:
Please close this ticket.
Is resolved.

Thanks.

closed 2021-03-21 by mauricema