slimbootloader/slimbootloader #1052: [EHL] [SBL] /dev/ish-smhi doesn't exist

Jiayan5877, 2021-03-19:
Dear Sir,

I try to use HECI sample code to check PSE HECI function but I got the fail result while code runs to "heci_register()" and it returns "-16" not "0".
Yocto with SBL OS loader payload and SBL UEFI payload, we can't found the /dev/ish-smhi device node.
Currently, we only success on Yocto boot with AMI UEFI BIOS. The test boot devices(Yocto) are all the same with AMI UEFI BIOS and SBL UEFI payload.
**May I know how to enable the HECI between SOC and PSE by SBL?**
Here is successful display when executing HECI sample code
[image]
Here is unsuccessful display when executing HECI sample code
[image]

The source code of EHL's SBL is from sbl-ehl-beta3_er.zip

Best Regards,

ywho79, 2021-03-19:
Hi Jiayan5877,

Thanks for reporting this. The EHL SBL team will investigate and update you as soon as we can.

mauricema, 2021-03-20:
Meanwhile, can you also post your Yocto boot log for both BIOS and SBL ?
In addition please run "lspci" in Yocto and capture the output.

Jiayan5877, 2021-03-22:
Dear Mauricema,

May I reconfirm your comment?
You need my Yocto boot log for AMI UEFI BIOS and SBL OS loader payload, and run "lspci" in Yocto for AMI UEFI BIOS and SBL OS loader payload and capture the output.
Is my understanding right?

Best Regards,

jinjhuli, 2021-03-22:
Hi Jiayan5877,

We may missed something in EHL Beta 3, can you please try and verify this in our latest open source slimbootloader source code?

Get source code:
git clone https://github.com/slimbootloader/slimbootloader.git
Build command:
python BuildLoader.py build ehl -p "OsLoader.efi:LLDR:Lz4;UefiPld.fd:UEFI:Lzma"

Before you build, please set value below in file slimbootloader\Platform\ElkhartlakeBoardPkg\BoardConfig.py:
self.HAVE_FSP_BIN         = 1

Regards,
Jin Jhu

Jiayan5877, 2021-03-23:
Dear Jin Jhu,

I wanted to try the latest open source slimbootloader source code, but system hangs on Call FspSiliconInit ...
Here is log.
[SBL Boot Fail.txt](https://github.com/slimbootloader/slimbootloader/files/6187616/SBL.Boot.Fail.txt)
Could you provide me some idea for this problem?

Best Regards,

Jiayan5877, 2021-03-26:
Dear Jin Jhu,

May I know whether the latest open source slimbootloader source code can boot on EHL CRB?

Best Regards,

jinjhuli, 2021-03-26:
Hi Jiayan5877,

Our latest open source is able to boot on EHL CRB.
Are you building the bianry using Beta 3 FSP? I suspect that your current FSP is too old for latest open source thus you see the error.
We are preparing for the upstream of EHL FSP, please stay tuned.

Regards,
jin Jhu

Jiayan5877, 2021-03-29:
Dear Jin Jhu,

I build the binary using VbtBin, FspBin and Microcode folder from sbl-ehl-beta3_er.zip.
May I ask a couple of questions?
Does the latest open source on EHL CRB has /dev/ish-smhi?
May I know when new FSP is released?

Best Regards,

jinjhuli, 2021-04-16:
Hi Jiayan5877,

I've tried with the FSP that is going to upstream and I can see that /dev/ish-smhi is in Yocto.
Can we proceed to close this issue while pending for FSP upstream approval?

Regards,
Jin Jhu

closed 2021-04-16 by rprangar