slimbootloader/slimbootloader #1061: [QEMU] mtrr command in shell cause system halt

ionutnechita, 2021-03-23:
Hi Slimbootloader Team,

I noticed a problem typing a command in QEMU.

Shell> mtrr
MTRRs......: Enabled
Fixed MTRRs: Enabled
Default....: Uncacheable

Variable MTRRs
[0x000000000-0x00FFFFFFF] Write-Back
   IA32_MTRR_PHYSBASE_0 (0x200): 0x0000000000000006
   IA32_MTRR_PHYSMASK_0 (0x201): 0x000000FFF0000800
[0x00F000000-0x00FFFFFFF] Uncacheable
   IA32_MTRR_PHYSBASE_1 (0x202): 0x000000000F000000
   IA32_MTRR_PHYSMASK_1 (0x203): 0x000000FFFF000800
Exception #13 from 0x0010:0x0ECBBC39 !!!

STAGE_2: System halted!

Command for reproduce:
qemu-system-x86_64 -cpu max -smp 4 -machine q35 -m 256 --nographic -boot order=d -pflash Outputs/qemu/SlimBootloader.bin --enable-kvm

mauricema, 2021-03-23:
Please try:   https://github.com/slimbootloader/slimbootloader/pull/1062

mauricema, 2021-03-25:
This is fixed in 1062.  
Close it for now.  If you still see issue, please reopen.

closed 2021-03-25 by mauricema