slimbootloader/slimbootloader #1063: [QEMU] CPU SMP configuration is not kept at boot

ionutnechita, 2021-03-23:
Hi Slimbootloader Team,

I noticed that the number of cores is not kept in the QEMU configuration.

Step for reproduce:
qemu-system-x86_64 -machine q35 --nographic -boot order=d -pflash Outputs/qemu/SlimBootloader.bin -m 4096  --cpu host  --enable-kvm  -smp 4
...
Detected 4 CPU threads
 CPU  0 APIC ID: 0
 CPU  1 APIC ID: 2
 CPU  2 APIC ID: 1
 CPU  3 APIC ID: 3
...
dmesg | grep CPU
[    0.041305] smpboot: Boot CPU (id 0) not listed by BIOS
[    0.041309] smpboot: Allowing 1 CPUs, 0 hotplug CPUs
[    0.041341] setup_percpu: NR_CPUS:64 nr_cpumask_bits:64 nr_cpu_ids:1 nr_node_ids:1
[    0.064891] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.065286] rcu: \x09RCU restricting CPUs from NR_CPUS=64 to nr_cpu_ids=1.
[    0.167401] smp: Bringing up secondary CPUs ...
[    0.167820] smp: Brought up 1 node, 1 CPU

[cpu-smp-issue-boot-kernel.txt](https://github.com/slimbootloader/slimbootloader/files/6191476/cpu-smp-issue-boot-kernel.txt)

mauricema, 2021-03-23:
Did you check without "enable-kvm" ?   

Our primary goal for SBL QEMU is not to enable KVM.  Instead, trying to prove the firmware flow works on QEMU.   As a result, we have not checked/verified QEMU with KVM enabled.   To fully support QEMU KVM, it might have additional requirement in firmware.  Today it is not within SBL's scope.

mauricema, 2021-03-23:
BTW, in your log "APIC: ACPI MADT or MP tables are not detected" was seen.  It means the kernel could not find any ACPI table.
I guess the kernel might have ignored command line "acpi_rsdp=0x7EB04000" the parameter SBL passed ?

ionutnechita, 2021-03-23:
Hi @mauricema,

// Did you check without "enable-kvm" ?

I try with --enable-kvm and with out this option.
And this issue is same in both cause.

ionutnechita, 2021-03-23:
// I guess the kernel might have ignored command line "acpi_rsdp=0x7EB04000" the parameter SBL passed ?

Is passed in Kernel Cmdline by SBL bootloader.
But I don't know if this option works with kernel 5.1.0.

mauricema, 2021-03-23:
I tried it with our kernel (the one we used for testing), I did see all 4 processors.
So it definitely is related to your kernel build.

ionutnechita, 2021-03-23:
Could you give me a config file with kernel options?

mauricema, 2021-03-23:
You can extract the config from the following patch.  You don't need to change any other things.

https://github.com/universalpayload/linuxpayload/blob/master/patch/0001-Enable-universal-payload-x64-kernel.patch
file:  arch/x86/configs/upld_defconfig

mauricema, 2021-03-25:
Close it for now since it is a kernel specific config issue.

closed 2021-03-25 by mauricema