slimbootloader/slimbootloader #1064: [QEMU] can't find IRQ for PCI INT A - Network Controller - E1000

ionutnechita, 2021-03-23:
Hi Slimbootloader Team,

I noticed that the interrupts are not mapped or remapped correctly in the QEMU configuration.

Step for reproduce:
qemu-system-x86_64 -machine q35 --nographic -boot order=d -pflash Outputs/qemu/SlimBootloader.bin -m 4096 --cpu host --enable-kvm -smp 4

Kernel command line: console=ttyS0 acpi_rev_override=1 acpi_osi=! acpi_osi="Windows 2017" loglevel=7  initcall_debug tsc=reliable no_timer_check noreplace-smp kvm-intel.nested=1 intel_iommu=igfx_off cryptomgr.notests rcupdate.rcu_expedited=1 rcu_nocbs=0-64 rw nomodeset i915.modeset=0 debug ignore_loglevel log_buf_len=16M no_console_suspend systemd.log_target=null systemd.unit=rescue.target acpi_rsdp=0x7EB04000

dmesg | grep -E "irq|e1000|clocksource"
[    0.041322] clocksource: refined-jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645519600211568 ns
[    0.065311] NR_IRQS: 4352, nr_irqs: 24, preallocated irqs: 16
[    0.136994] clocksource: tsc-early: mask: 0xffffffffffffffff max_cycles: 0x26eb9640731, max_idle_ns: 440795302659 ns
[    0.156757] calling  spawn_ksoftirqd+0x0/0x39 @ 1
[    0.157193] initcall spawn_ksoftirqd+0x0/0x39 returned 0 after 0 usecs
[    0.179963] calling  init_jiffies_clocksource+0x0/0x19 @ 1
[    0.180494] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.181563] initcall init_jiffies_clocksource+0x0/0x19 returned 0 after 0 usecs
[    0.197191] calling  irq_sysfs_init+0x0/0x92 @ 1
[    0.197641] initcall irq_sysfs_init+0x0/0x92 returned 0 after 0 usecs
[    0.237260] calling  blk_softirq_init+0x0/0x71 @ 1
[    0.237700] initcall blk_softirq_init+0x0/0x71 returned 0 after 0 usecs
[    0.240304] calling  irq_poll_setup+0x0/0x71 @ 1
[    0.240730] initcall irq_poll_setup+0x0/0x71 returned 0 after 0 usecs
[    0.302159] calling  clocksource_done_booting+0x0/0x42 @ 1
[    0.302715] clocksource: Switched to clocksource tsc-early
[    0.303249] initcall clocksource_done_booting+0x0/0x42 returned 0 after 529 usecs
[    0.319390] calling  proc_softirqs_init+0x0/0x22 @ 1
[    0.319865] initcall proc_softirqs_init+0x0/0x22 returned 0 after 0 usecs
[    0.367085] calling  init_tsc_clocksource+0x0/0xb9 @ 1
[    0.367574] initcall init_tsc_clocksource+0x0/0xb9 returned 0 after 0 usecs
[    0.380322] calling  init_clocksource_sysfs+0x0/0x24 @ 1
[    0.380839] initcall init_clocksource_sysfs+0x0/0x24 returned 0 after 7 usecs
[    0.451816] serial8250: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[    0.461384] calling  e1000_init_module+0x0/0x7e @ 1
[    0.461834] e1000: Intel(R) PRO/1000 Network Driver - version 7.3.21-k8-NAPI
[    0.462554] e1000: Copyright (c) 1999-2006 Intel Corporation.
[    0.463142] e1000 0000:00:02.0: can't find IRQ for PCI INT A; please try using pci=biosirq
[    0.817608] e1000 0000:00:02.0 eth0: (PCI:33MHz:32-bit) 52:54:00:12:34:56
[    0.818327] e1000 0000:00:02.0 eth0: Intel(R) PRO/1000 Network Connection
[    0.819606] initcall e1000_init_module+0x0/0x7e returned 0 after 349385 usecs
**_[    0.872476] genirq: Flags mismatch irq 0. 00000080 (eth0) vs. 00015a00 (timer)_**
[    0.874129] e1000 0000:00:02.0 eth0: Unable to allocate interrupt Error: -16
[    1.429308] clocksource: tsc: mask: 0xffffffffffffffff max_cycles: 0x26eb9640731, max_idle_ns: 440795302659 ns
[    1.432407] clocksource: Switched to clocksource tsc

e1000 0000:00:02.0 eth0: Unable to allocate interrupt Error: -16
ifconfig: ioctl 8914: Resource busy
route: ioctl 0x890b failed: Network unreachable

Kernel:
lspci -nnkk
00:1f.2 Class 0106: 8086:2922
00:1f.0 Class 0601: 8086:2918
00:01.0 Class 0300: 1234:1111
00:1f.3 Class 0c05: 8086:2930
00:00.0 Class 0600: 8086:29c0
00:02.0 Class 0200: 8086:100e e1000

Slimbootloader:
Shell> pci
00:00.0 8086:29C0 Bridge Device
00:01.0 1234:1111 Display Controller
00:02.0 8086:100E Network Controller
00:1F.0 8086:2918 Bridge Device
00:1F.2 8086:2922 Mass Storage Controller
00:1F.3 8086:2930 Serial Bus Controllers

[config-linux-5.1.0.txt](https://github.com/slimbootloader/slimbootloader/files/6191579/config-linux-5.1.0.txt)

mauricema, 2021-03-24:
This is a similar issue with your kernel.  
Tested a standard yocto kernel listed in https://slimbootloader.github.io/supported-hardware/qemu.html,  we did not see this issue.   
Please see the log below.  e1000 device interrupt can be assigned properly.
[SblDebug.log](https://github.com/slimbootloader/slimbootloader/files/6194378/SblDebug.log)

mauricema, 2021-03-25:
Close it for now since it is kernel specific config issue.

closed 2021-03-25 by mauricema