slimbootloader/slimbootloader #1082: A possible mistake in source code

jon-xelex, 2021-03-30:
Greetings,

I've just found something seems a mistake in source code comment. In _BootloaderCommonPkg/Library/SecureBootLib/SecureBootHash.c_, function DoHashVerify()'s comment mention an argument name ComponentType which does not reference (starting from line 136)

`/**`
 ` Verify data block hash with the built-in one.
`
 ` @param[in]  Data           Data buffer pointer.`
  `@param[in]  Length         Data buffer size.`
  `@param[in]  Usage          Hash usage.`
  `@param[in]  ComponentType  Component type.`
  `@param[in,out]  Hash       On input,  expected hash value when ComponentType is not used.`
                           `On output, calculated hash value when verification succeeds.`
  `@retval RETURN_SUCCESS             Hash verification succeeded.`
  `@retval RETURN_INVALID_PARAMETER   Hash parameter is not valid.`
  `@retval RETURN_NOT_FOUND           Hash data for ComponentType is not found.`
  `@retval RETURN_UNSUPPORTED         Hash component type is not supported.`
  `@retval RETURN_SECURITY_VIOLATION  Hash verification failed.`

`**/`
`RETURN_STATUS`
`EFIAPI`
`DoHashVerify (`
 ` IN CONST UINT8           *Data,`
 ` IN       UINT32           Length,`
 ` IN       HASH_COMP_USAGE  Usage,`
  `IN       UINT8            HashAlg,`
  `IN OUT   UINT8           *HashData`
`  )`

The same problem also applied to header file in _BootloaderCommonPkg/Include/Library/SecureBootLib.h_

Can you please clarify whether ComponentType matter in DoHashVerify function and please fix if possible?

jon-xelex, 2021-03-30:
Forgot to mention, this "Component Type" mistake not just arise from _SecureBootHash.c_ but also other file like _SecureBootRsa.c_, for example the DoRsaVerify() function

lsubash, 2021-03-30:
Hi Jon,

These comments are fixed with https://github.com/slimbootloader/slimbootloader/pull/1083.

Hash component usage is used in DoRsaVerify & DoHashVerify API's. These API's can be used verify,
1. Standard slimboot components/public keys as defined in HASH_COMP_USAGE. Refer BootloaderCommonPkg\Include\Guid\KeyHashGuid.h. Hash value to compared for components would be available in Hash store. Hash usage is non-zero.
2. Non standard component where hash usage is 0. Slimboot hash store will not have hash value stored.  Caller has to provide hash value for comparison along with databuf.

Regards,
Subash

jon-xelex, 2021-03-31:
Thanks for the fix, I understand now! 

By the way, I always wonder where value of Hash store come from. Could you please tell me how Hash store is created, is it calculated then pass to RAM or built when build slimbootloader then flash to SPI Flash?

lsubash, 2021-03-31:
SBL Hash store is in build to flash image.
Refer to https://slimbootloader.github.io/security/verified-boot.html, picture "SBL Flash Image".

Hash store containing public keys hashes (components RSA signed) can be configured
1. at build time by configuring "GetKeyHashList" in BoardConfig.py.
2. For generating hash store & updating externally, use "GenExtKeyHashStore.py". Update generated component using capsule firmware update.

jon-xelex, 2021-03-31:
Thanks for your help, much appreciated. Guess I will close this issue.

closed 2021-03-31 by jon-xelex