slimbootloader/slimbootloader #1120: [question] EXT4 format partition is supported for boot SBL file?

ionutnechita, 2021-04-15:
Hi SBL Team,

I noticed in the documentation that the support for the EXT format is low.
Only EXT 2 and EXT 3 are supported.
Can support be added for EXT 4 as well? Or does this format already work?

Thanks.

jgutbub, 2021-04-15:
Hi @ionutnechita 

EXT4 is supported:

https://slimbootloader.github.io/developer-guides/boot-options.html

If there are some other parts in our https://slimbootloader.github.io/ documentation that needs to be updated it would help if you could point us to the page you saw about EXT 2 and 3 only or if you'd be up to it you could submit a patch in our documentation repo to help us update those pages.

Hope it helps!

ionutnechita, 2021-04-16:
Could you make a page with the supported file system types?

jgutbub, 2021-04-19:
We will take your request into consideration; if you would like to expand on this topic with the details you are looking for or want to follow up further please consider posting to our mailing list/forum:

https://groups.io/g/slimbootloader

Since this is not truly an issue I am closing it; requests are welcome in the mailing list. Thanks for the feedback!

closed 2021-04-19 by jgutbub