slimbootloader/slimbootloader #1121: PchCycleDecodingLib cannot build due to unresolved identifiers.

aaronpop, 2021-04-15:
Building the PchCycleDecodingLib generates unresolved identifier errors

R_PCH_LPC_GEN1_DEC': undeclared identifier
B_PCH_LPC_GENX_DEC_IOBAR': undeclared identifier
B_PCH_LPC_GENX_DEC_IODRA': undeclared identifier
B_PCH_LPC_GENX_DEC_EN': undeclared identifier
R_PCH_ESPI_PCBC': undeclared identifier
B_PCH_ESPI_PCBC_ESPI_EN': undeclared identifier
R_PCH_ESPI_CS1GIR1': undeclared identifier
B_PCH_LPC_GENX_DEC_IOBAR': undeclared identifier
B_PCH_LPC_GENX_DEC_IODRA': undeclared identifier
B_PCH_LPC_GENX_DEC_EN': undeclared identifier
R_PCH_LPC_IOD': undeclared identifier
R_PCH_LPC_IOD': undeclared identifier
R_PCH_LPC_IOE': undeclared identifier
R_PCH_LPC_IOE': undeclared identifier

To add the library into the build, I modified the GetDscLibrarys in the in BoardConfig.py to include the PchPcrLib and PchCycleDecodingLib:
            'PchPcrLib|Silicon/$(SILICON_PKG_NAME)/Library/PchPcrLib/PchPcrLib.inf',
            'PchCycleDecodingLib|Silicon/$(SILICON_PKG_NAME)/Library/PchCycleDecodingLib/PchCycleDecodingLib.inf',

In particular I am using the Elkhartlake, but I think this error will occur with any board that tries to build a PchCycleDecodingLib.

jinjhuli, 2021-04-16:
Hi aaronpop,

Do you mean you modified "GetPlatformDsc" in BoardConfig.py in Platform\ElkhartlakeBoardPkg?
I am able to build EHL by adding the two lines you mentioned:
'PchPcrLib|Silicon/$(SILICON_PKG_NAME)/Library/PchPcrLib/PchPcrLib.inf',
'PchCycleDecodingLib|Silicon/$(SILICON_PKG_NAME)/Library/PchCycleDecodingLib/PchCycleDecodingLib.inf',

I random picked some of the registers that showing "undeclared identifier" error but I couldn't find it in the source code.
Did you declared those registers in PchCycleDecodingLib?

aaronpop, 2021-04-16:
Hi junjhuli,

You are correct, it builds correctly with just modifying the BoardConfig.py. It seems that the EDK2 build process does not build libraries unless it is referenced by an INF file.

To get the build error, please modify Platform\ElkhartlakeBoardPkg\Library\Stage1BBoardInitLib\Stage1BBoardInitLib.inf's [LibraryClasses] to referenced PchCycleDecodingLib. Then the build error will occur.

jinjhuli, 2021-04-30:
Hi aaronpop,

The fix is already merged in the latest code base, please have a try and let us know if you still facing any issue.

aaronpop, 2021-04-30:
Hi junjhuli,

Yes, the latest version of the library is able to build without errors. Thank you.

closed 2021-04-30 by aaronpop