slimbootloader/slimbootloader #1136: Dockerfile installs incompatible git version

fitzged, 2021-04-27:
With the change added to verify correct tool versions are being used (bc5584d176878a154bb79cd8a94487dceb215670), the Dockerfile needs updating to pull the latest git version for Ubuntu 18.04 from the ppa:git-core/ppa repo, otherwise builds fail:
Checking Toolchain Versions...
- /usr/bin/python: Version 3.6.9 (>= 3.6.0) [PASS]
- /usr/bin/openssl: Version 1.1.1 (>= 1.1.0g) [PASS]
- /usr/bin/nasm: Version 2.13.02 (>= 2.12.02) [PASS]
- /usr/bin/iasl: Version 20160831 (>= 20160422) [PASS]
- /usr/bin/git: Version 2.17.1 (>= 2.20.0) [FAIL]
- /usr/bin/gcc: Version 7.5.0 (>= 7.3) [PASS]
...Failed! Please check toolchain versions!

myguitar, 2021-04-27:
Thanks for reporting 'git' version check issue. #1137 will print recommended 'git' version if not met and continue to build.

fitzged, 2021-04-27:
Great, thanks @myguitar that will work for me.

myguitar, 2021-04-27:
No problem. Let me close this.

closed 2021-04-27 by myguitar