slimbootloader/slimbootloader #1150: issue in building AbSupportLib

AntonioSanta, 2021-05-12:
Hello, 
I am finding an issue when building master branch for TigerLake, I am up to date with the master branch : 
I run:
python BuildLoader.py build tgl
Checking Toolchain Versions...
- /usr/bin/python: Version 3.8.5 (>= 3.6.0) [PASS]
- /usr/bin/openssl: Version 1.1.1f (>= 1.1.0g) [PASS]
- /usr/bin/nasm: Version 2.14.02 (>= 2.12.02) [PASS]
- /usr/bin/iasl: Version 20190509 (>= 20160422) [PASS]
- /usr/bin/git: Version 2.25.1 (>= 2.20.0) [PASS]
- /usr/bin/gcc: Version 9.3.0 (>= 7.3) [PASS]
...Done!

and after a while build process stops with error : 

Building ... /home/tiger/Ashare/tigerlake/slimbootloader/PayloadPkg/Library/AbSupportLib/AbSupportLib.inf [IA32]
"gcc" -MMD -MF /home/tiger/Ashare/tigerlake/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/PayloadPkg/Library/AbSupportLib/AbSupportLib/OUTPUT/AbSupportLib.obj.deps -g -Os -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -include AutoGen.h -fno-common -ffunction-sections -fdata-sections -DSTRING_ARRAY_NAME=AbSupportLibStrings -m32 -march=i586 -malign-double -fno-stack-protector -D EFI32 -fno-asynchronous-unwind-tables -Wno-address -fno-pic -fno-pie -flto -Os  -DDISABLE_NEW_DEPRECATED_INTERFACES -fno-pic -c -o /home/tiger/Ashare/tigerlake/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/PayloadPkg/Library/AbSupportLib/AbSupportLib/OUTPUT/./AbSupportLib.obj -I/home/tiger/Ashare/tigerlake/slimbootloader/PayloadPkg/Library/AbSupportLib -I/home/tiger/Ashare/tigerlake/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/PayloadPkg/Library/AbSupportLib/AbSupportLib/DEBUG -I/home/tiger/Ashare/tigerlake/slimbootloader/MdePkg -I/home/tiger/Ashare/tigerlake/slimbootloader/MdePkg/Include -I/home/tiger/Ashare/tigerlake/slimbootloader/MdePkg/Include/Ia32 -I/home/tiger/Ashare/tigerlake/slimbootloader/PayloadPkg -I/home/tiger/Ashare/tigerlake/slimbootloader/PayloadPkg/Include -I/home/tiger/Ashare/tigerlake/slimbootloader/BootloaderCommonPkg -I/home/tiger/Ashare/tigerlake/slimbootloader/BootloaderCommonPkg/Include -I/home/tiger/Ashare/tigerlake/slimbootloader/Silicon/TigerlakePkg/FspBin /home/tiger/Ashare/tigerlake/slimbootloader/PayloadPkg/Library/AbSupportLib/AbSupportLib.c
/home/tiger/Ashare/tigerlake/slimbootloader/PayloadPkg/Library/AbSupportLib/AbSupportLib.c: In function ‘ParseBootSlot’:
/home/tiger/Ashare/tigerlake/slimbootloader/PayloadPkg/Library/AbSupportLib/AbSupportLib.c:56:12: error: implicit declaration of function ‘CalculateCrc32WithType’; did you mean ‘CalculateCrc32’? [-Werror=implicit-function-declaration]
   56 |   Status = CalculateCrc32WithType ((UINT8 *)AbBootInfo, DataSize, Crc32TypeDefault, &CrcOut);
      |            ^~~~~~~~~~~~~~~~~~~~~~
      |            CalculateCrc32
/home/tiger/Ashare/tigerlake/slimbootloader/PayloadPkg/Library/AbSupportLib/AbSupportLib.c:56:67: error: ‘Crc32TypeDefault’ undeclared (first use in this function)
   56 |   Status = CalculateCrc32WithType ((UINT8 *)AbBootInfo, DataSize, Crc32TypeDefault, &CrcOut);
      |                                                                   ^~~~~~~~~~~~~~~~
/home/tiger/Ashare/tigerlake/slimbootloader/PayloadPkg/Library/AbSupportLib/AbSupportLib.c:56:67: note: each undeclared identifier is reported only once for each function it appears in
cc1: all warnings being treated as errors
make: *** [GNUmakefile:303: /home/tiger/Ashare/tigerlake/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/PayloadPkg/Library/AbSupportLib/AbSupportLib/OUTPUT/AbSupportLib.obj] Error 1

build.py...
 : error 7000: Failed to execute command
	make tbuild [/home/tiger/Ashare/tigerlake/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/PayloadPkg/Library/AbSupportLib/AbSupportLib]

build.py...
 : error F002: Failed to build module
	/home/tiger/Ashare/tigerlake/slimbootloader/PayloadPkg/Library/AbSupportLib/AbSupportLib.inf [IA32, GCC5, DEBUG]

- Failed -
Build end time: 09:10:49, May.12 2021
Build total time: 00:00:51

Error in running process:
  build --platform BootloaderCorePkg/BootloaderCorePkg.dsc -b DEBUG --arch IA32 --tagname GCC5 -n 1 -y Report.log -Y PCD -Y FLASH -Y LIBRARY

closed 2021-05-12 by AntonioSanta

akber1018, 2022-02-15:
Hello @AntonioSanta , have you resolved this issue?
I am currently getting the same issue for Tigerlake

akber1018, 2022-02-20:
Thank you for the response @mauricema 

The issue got resolved after:
1) Upgrading the IASL compiler from 2019 to 2021 latest version
2) Installing bison and flex dependencies