slimbootloader/slimbootloader #1157: Failed to boot to LinuxBoot in Stage2

Ziang95, 2021-05-18:
Hi, this is my first time opening an issue on Github, sorry in advance for any ambiguity.

Here's the environment of my machine:
OS: Ubuntu 20.04.2 LTS Focal
GCC: 9.3.0
Python3: 3.8.5

And I followed the instructions on the [Tutorial](https://slimbootloader.github.io/how-tos/boot-with-linux-payload.html).
The crucial steps that I did follow:

1. set `GEN_CFG_DATA.PayloadId  |  'LINX'` in QemuBoardPkg/CfgData/CfgDataExt_Brd1.dlt
2. set `ENABLE_LINUX_PAYLOAD = 1` in QemuBoardPkg/BoardConfig.py
3. set `HAVE_VERIFIED_BOOT = 0` in QemuBoardPkg/BoardConfig.py
4. used the [Linux.zip](https://github.com/slimbootloader/slimbootloader/files/4463548/QemuLinux.zip) provided in the Tutorial.
5. unzipped and put then in PayloadPkg/PayloadBin
6. build with command `python BuildLoader.py build qemu -p "config.cfg:OSLD;vmlinuz:LINX;config.cfg:CMDL;initrd:INRD"` and it passed the build.
7. run with command `qemu-system-x86_64 -cpu max -machine q35 -m 256 -serial stdio -boot order=d -pflash Outputs/qemu/SlimBootloader.bin`

And from here the issue comes up:
[image]

mauricema, 2021-05-18:
Thank you for reporting this issue.
This ASSERT indicates your memory heap is too small and SBL is running out of memory whiling load EPLD (Linux payload).
To address it, you can try to increase your HEAP in the BoardConfig.py.
For example, you can add:
self.LOADER_RSVD_MEM_SIZE         = 0x00800000
And then you can recompile and try.
Please let us know if you still have issue with it.

Thanks
Maurice

mauricema, 2021-05-18:
BTW, just tried on Ubuntu 18.04.  It worked fine.  

Please note, this kernel is compiled with "movbe" instruction enabled.  So QEMU will need to support this.   Latest QEMU for Windows has no issue with the command listed on github.   However, for Linux,
it seems a little bit different.  Here is the command line I tried on QEMU that worked in Ubuntu 18.04:
qemu-system-x86_64 -cpu qemu64,+movbe -machine q35 -m 256 -serial mon:stdio -boot order=d -pflash Outputs/qemu/SlimBootloader.bin -nographic

The "+movbe" is important.  Or the boot will hang at "Switch to LongMode and jump to 64-bit kernel entrypoint".  I hope it helps.

Thanks
Maurice

Ziang95, 2021-05-19:
Thank you Maurice, I modified the HEAP size and it works fine now on ubuntu 20.

Just a follow up:
Is there any support for [u-root](https://github.com/u-root/u-root)?

mauricema, 2021-05-19:
Hi, Ziang,

Yes, it can support u-root as well.   I have tried to compile a u-root as rootfs, and it can boot fine on QEMU with SBL. 

For you convenience, I have attached the my kernel and u-root image.   For SBL, you might have to adjust the flash layout a little bit since u-root is around 5M.

[Uroot.zip](https://github.com/slimbootloader/slimbootloader/files/6506139/Uroot.zip)
[Debug.log](https://github.com/slimbootloader/slimbootloader/files/6506326/Debug.log)

Ziang95, 2021-05-19:
Much appreciate for the images provided here! really helpful.

The combined firmware was still slightly over 817KB even I had shrunk the size of many other image sections, so I turned off the redundant layout, and now it works fine on Qemu (windows ver.). 

Actually I tried using an Lzma-compressed u-root image before (the image generated by the 'u-root' go program, then compressed by lzma), and passed "u-root:INRD:Lzma" to BuilLoader.py, but the system hang at:
[image]
Did I omit any step configuring the lzma decompressor?

Ziang95, 2021-05-19:
> Much appreciate for the images provided here! really helpful.
>
> The combined firmware was still slightly over 817KB even I had shrunk the size of many other image sections, so I turned off the [...]

Oh I see... The "Lzma" flag is to let BuildLoader compress it while building firmware, was using it in a wrong way.

mauricema, 2021-05-19:
Hi, Ziang

You are welcome!  :)

Yes, the LZMA you specified in SBL build command line is for SBL to compress the component. It will be decompressed by SBL Stage2.  Since u-root image is already compressed before feeding into SBL,  we should not compress it again.  
To allow kernel to support LZMA or XZ, there is a kernel config you can enable in kernel.  In my kernel image, I just enabled GZIP, so it won't support LZMA or XZ.  You can try to compile your own kernel.

Ziang95, 2021-05-20:
Thanks Maurice! Will try it later~

mauricema, 2021-05-20:
Since the issue was addressed, close it.

closed 2021-05-20 by mauricema