slimbootloader/slimbootloader #1166: slimboot kontron board bring up

erik-kline, 2021-05-25:
I am attempting to install slimboot onto a Kontron board(COME-CTL6) this card is mounted in a COMe Eval Carrier2 T6. I am not able to get serial output from the board. . I am compiling the head from slimboot

Kontron has connected a PLD to the LPC/espi bus. It appears that this is connected to io port 0x80. This allows me to place data on this bus via the outb(data,0x80); 

 

I am unable to get serial output from the Kontron board. But from adding outb’s to the code. I can see that the serial port appears to be initialized. If it is the legacy 0x3f8 serial port.  Boardinit.py provides for the ability to change this. But if I do this the code stops at the line below in stage1aBoardinitlib.c 

if (DebugPort < PCH_MAX_SERIALIO_UART_CONTROLLERS) {
      GpioConfigurePads (2, (GPIO_INIT_CONFIG *)mUartGpioTable + DebugPort * 2);
    }

The configuration of the board ID is confusing because of the so many different ways in which it can be set. I have configured the board id to be 1.  I have sprinkled outb’s with various values throughout the code.  the code appears to execute all the way to the shell. I do NOT see the splash screen. Or any output also the USB bus has no power.

mauricema, 2021-05-25:
Hi,

First of all, please provide a little bit more info on your board.  We are not familiar with this board. I assume it is TigerLake based platform here to start with.

In order to make serial port work, I think you need to know the detailed board info.
For example, if your board has EC (Embedded Controller), then the UART could be connected to EC legacy UART (0x3F8).  If your board has no EC, then mostly your UART is routed to one of the PCH SOC UARTs.  If so, you need to further understand which SOC UART it is connected to.  Then you will need to setup the default UART in BoardConfig.py accordingly.  For EC UART,  you can set self.DEBUG_PORT_NUMBER to 0xFF.  For SOC UART, you can set it to 0, 1, 2, etc.

For BoardId, in your case, you can use stitching platform data to specify a fixed ID and fixed UART debug port number. In this way SBL will not override it during boot. To do so, you can use "-p" argument for StitchLoader.py.  You can take a reference from https://slimbootloader.github.io/supported-hardware/whl-crb.html#stitching.  It is for WHL, but should be similar for TGL.

For GFX output, it might need additional configuration in VBT binary.  For USB bus power, I guess you might have to configure the GPIO correctly for your board.  It might be power-gated by GPIO on your board.

Regards
Maurice

mauricema, 2021-05-26:
For UART,  I think you should be able to figure it out from UEFI Shell if you have the original BIOS.  
If you can boot to UEFI Shell, you can try to write to I/O port 0x3F8 to see if you can see output from serial port.  EX:  "mm -io 3f8", then input "2a",  if your uart is on 0x3F8, you should see a "*" on your serial port.  Otherwise, it might indicate your uart is not at 0x3f8.  You can also try 0x2F8 and MMIO from your PCI UART device BAR 0.   Linux kernel log "dmsg" should also provide some info on it.  If  you can post your kernel log, maybe we can find some useful info.

For BoardId/PlatformId,  if the fixed ID is configured using "-p", SBL is supposed to skip all the board detection code.  If it is not the case, please submit bug report and we will fix it.  There is no standard way to detect board ID. It is design specific.  In your case, hard coded ID makes it much easier.

For VBT binary configuration, there is a tool called "Display Configuration Tool" (DisCon Tool) to customize the VBT binary for TGL platform.  You might want to contact your Intel representative to find how to get this tool and required support.

-Maurice

gdong1, 2021-05-27:
For the PlatformId override issue on TGL, please try this patch 
https://github.com/slimbootloader/slimbootloader/pull/1169

mauricema, 2021-05-27:
Patch #1169 was merged.

mauricema, 2021-05-27:
BTW, we recently submitted a PR:
https://github.com/slimbootloader/slimbootloader/pull/1171
This PR should help address your UART issue where SBL hung at the line below when debug UART is set to 0-2.
GpioConfigurePads (2, (GPIO_INIT_CONFIG *)mUartGpioTable + DebugPort * 2);
Please give a try and let us know.

rprangar, 2021-05-28:
Hi

Just wanted to provide some extra details on the Platform ID.

SBL maintains a variable called PlatformId as part of its global data structure. There are a couple of library functions to set/get this variable - SetPlatformId/GetPlatformId.

There are two places where the platform ID is set

1. Stage 1A - SBL will check if some platform data is passed through the stitch process and will use this to set the platform ID. This is useful if you support only board and there is no board specific mechanism available for runtime detection.

https://github.com/slimbootloader/slimbootloader/blob/9480958ee877def3d86211078f35b0296afbbad2/Platform/TigerlakeBoardPkg/Library/Stage1ABoardInitLib/Stage1ABoardInitLib.c#L93

2. Stage 1B - The SBL platform code detects the platform (implementation specific - read from EC, read from GPIO, etc.) and sets the Platform ID. This approach is useful when more than 1 board needs to be supported and the hardware provides a way to detect the boards. 
 https://github.com/slimbootloader/slimbootloader/blob/9480958ee877def3d86211078f35b0296afbbad2/Platform/TigerlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c#L763

Thanks.

erik-kline, 2021-05-28:
I tested the serial port fix and it is effective.
the code appears to be crashing somewhere in stage 2 and I am attempting to debug the problem.
how do I completely disable the graphic display?
this could be a source of the issue. I can just rely on the serial port for output

erik-kline, 2021-05-28:
currently the code stops in stage2 with
============= Intel Slim Bootloader STAGE2 =============                        
PCH gen (0x5), series (0x2), stepping (0x10)                                    
LPC DeviceId (0xA082)                                                           
GPIO PAD: 0x09000000   DATA: 0x0350A383 0x00002001                              
GPIO PAD: 0x09000001   DATA: 0x0350A383 0x00002001                              
GPIO PAD: 0x09000002   DATA: 0x0534AD81 0x00002601                              
GPIO PAD: 0x09000003   DATA: 0x0350A381 0x00002001                              
GPIO PAD: 0x09000004   DATA: 0x0550E281 0x00002001                              
GPIO PAD: 0x09000005   DATA: 0x0350A383 0x00002001                              
GPIO PAD: 0x09000006   DATA: 0x0350A383 0x00002001                              
GPIO PAD: 0x09000007   DATA: 0x0350A383 0x00002001                              
GPIO PAD: 0x09000008   DATA: 0x0350A383 0x00002001                              
GPIO PAD: 0x09000009   DATA: 0x0350A383 0x00002001                              
GPIO PAD: 0x0900000A   DATA: 0x0350A383 0x00002001                              
GPIO PAD: 0x0900000B   DATA: 0x0150A383 0x00002001

mauricema, 2021-05-28:
Please post your full boot log.
This seems to be in Stage2 BoardInit() PreSiliconInit hook.  What is your platform id you used?
For now you can try to skip the GPIO init by commenting out the TSN ConfigureGpio call listed below to see if it can continue.
https://github.com/slimbootloader/slimbootloader/blob/36ff6c2b6e8c7d424a4ac59f2733a28b5e48f952/Platform/TigerlakeBoardPkg/Library/Stage2BoardInitLib/Stage2BoardInitLib.c#L904-L916

-Maurice

mauricema, 2021-05-28:
You can also try to see if it makes a difference.
https://github.com/slimbootloader/slimbootloader/pull/1174

mauricema, 2021-06-03:
Did you get any chance to try it out ?   Is the UART issue resolved ?

erik-kline, 2021-06-03:
the serial ports are now working. the boot process appears to be handing off to the linux kernel. I still do not see the slimboot boot graphics.
I have created a buildroot bzImage that combines the root file systeme into the bzImage. i tried booting centos with slimboot. I have found in debugging these kinds of problems in the past that combining the rootfs and kernel together with buildroot simplifies the debug.  I have tested this image with slimboot ,qemu and on the kontron board. I have placed this image onto the target and using Kontrons bios. The device boots and I see serial output with the kontron bios.

in slimboot it appears that the control is passed to this image but I see no serial output and no image on the display.

erik-kline, 2021-06-03:
below is the serial output from slimboot

============= Intel Slim Bootloader STAGE1A =============
SBID: SBL_TGL 
ISVN: 001
IVER: 000.005.000.008.00009
SVER: 2F336E5447515C78-dirty
FDBG: BLD(D IA32) FSP(R)
FSPV: ID(TGLI-FSP) REV(0A002F32)
Loader global data @ 0xFEF2FD44
Run  STAGE1A @ 0xFFFEA000
Load STAGE1B @ 0xFFD80000

============= Intel Slim Bootloader STAGE1B =============
[Boot Guard] AcmStatus : 0x00000000
[Boot Guard] BootStatus: 0x00000000
[Boot Guard] Boot Guard Failed or is Disabled!
SpiInstance = FEF3D3E0
PchSpiBase at 0x00001F05
ScSpiBar0 at 0xF0000000
Loading Component KEYH:_HS_
Registering container KEYH
Append public key hash into store: Success
Load EXT CFG Data @ 0xFEF3DB54:0x04C4 ... Success
Processor does not support Boot Guard.
Boot Guard Support status: 0
GetPlatformDataPtr is copied 0xFEF3DB34 
BOOT: BP0 
MODE: 0
BoardID: 0x01
PlatformName: TGLU_DDR
GpioInit(0xFFD8A280:6) Done
Memory Init
Load Security Cfg Data
FSPM CfgData assignment
Load Memory Cfg Data
SkipMbpHob = 0x0
Load Graphics Cfg Data
board Id = 1 .....
Call FspMemoryInit ... Success
Loader global data @ 0x78C0FD70
Switch to memory stack @ 0x78C4FF00
Stage1 stack: 0x30000 (0x2CC90 used)
Stage1 heap: 0x10000 (0x3354 used)
Call FspTempRamExit ... Success
Memory FSP @ 0x78C50000
Memory TOP @ 0x78750000
Loading Component FLMP:SG02
Loaded STAGE2 @ 0x78A04000

============= Intel Slim Bootloader STAGE2 =============
PCH gen (0x5), series (0x2), stepping (0x10)
LPC DeviceId (0xA082)
GPIO PAD: 0x09000000   DATA: 0x0350A383 0x00002001
GPIO PAD: 0x09000001   DATA: 0x0350A383 0x00002001
GPIO PAD: 0x09000002   DATA: 0x0534AD81 0x00002601
GPIO PAD: 0x09000003   DATA: 0x0350A381 0x00002001
GPIO PAD: 0x09000004   DATA: 0x0550E281 0x00002001
GPIO PAD: 0x09000005   DATA: 0x0350A383 0x00002001
GPIO PAD: 0x09000006   DATA: 0x0350A383 0x00002001
GPIO PAD: 0x09000007   DATA: 0x0350A383 0x00002001
GPIO PAD: 0x09000008   DATA: 0x0350A383 0x00002001
GPIO PAD: 0x09000009   DATA: 0x0350A383 0x00002001
GPIO PAD: 0x0900000A   DATA: 0x0350A383 0x00002001
GPIO PAD: 0x0900000B   DATA: 0x0150A383 0x00002001
GPIO PAD: 0x0900000C   DATA: 0x0350A383 0x00002001
GPIO PAD: 0x0900000D   DATA: 0x0350A383 0x00002001
GPIO PAD: 0x0900000E   DATA: 0x0550E285 0x00002001
GPIO PAD: 0x09000010   DATA: 0x0550E281 0x00002001
GPIO PAD: 0x09000011   DATA: 0x0150A281 0x00002001
[... 550 lines cut ...]
 4010 |    5656 ms |      13 ms | OS loader main entry
 4020 |    7670 ms |    2014 ms | Shell exit
 4040 |    7725 ms |      55 ms | Load image entry
 4040 |    8359 ms |     634 ms | Load image entry
 4050 |    8397 ms |      38 ms | Boot device init
 4055 |    8397 ms |       0 ms | Boot device tuning
 4060 |    8539 ms |     142 ms | Parse partition info
 4040 |    8585 ms |      46 ms | Load image entry
 4040 |    8630 ms |      45 ms | Load image entry
 4040 |    8675 ms |      45 ms | Load image entry
 4050 |    8686 ms |      11 ms | Boot device init
 4055 |    8686 ms |       0 ms | Boot device tuning
 4040 |    8721 ms |      35 ms | Load image entry
 4050 |    8732 ms |      11 ms | Boot device init
 4055 |    8732 ms |       0 ms | Boot device tuning
 4060 |    8766 ms |      34 ms | Parse partition info
 4070 |    9189 ms |     423 ms | Load boot images
 40B0 |    9208 ms |      19 ms | Process ELF/MultiBoot
 40F0 |    9318 ms |     110 ms | FSP ReadyToBoot/EndOfFirmware notify
 4100 |    9318 ms |       0 ms | TPM IndicateReadyToBoot
------+------------+------------+----------------------------------

Starting Kernel ...

Switch to LongMode and jump to 64-bit kernel entrypoint ...

mauricema, 2021-06-03:
That is good progress.    

It looks like your UART console is not enabled by kernel somehow.
You cmdline uses "console=ttyS4,115200n8", but do you know if ttyS4 is the correct PCH debug UART ?
Maybe you can try to append "console=ttyS0,115200n8 console=ttyS1,115200n8 console=ttyS2,115200n8 console=ttyS3,115200n8" as well in case the UART device is not ttyS4.

To make it easier, you can also try to boot a mini Linux to see if it will work for you. 
'https://github.com/slimbootloader/slimbootloader/files/4463548/QemuLinux.zip',
You need to unzip all files into the root of the SATA disk UEFI partition (FAT partition).  
Then you can modify config.cfg to append more serial consoles as mentioned above.
Once done, you can try the boot.

-Maurice

erik-kline, 2021-06-03:
this booted. where is the linux .config file. so that I can look at the kernel configuration

mauricema, 2021-06-03:
you can look into https://github.com/mauricema/linux-mini
The config file is at https://github.com/mauricema/linux-mini/blob/master/misc/mini_config
It is based on Linux kernel v5.11.

mauricema, 2021-06-03:
As I mentioned before,  you might need to config your GPIO and VBT binary properly. 
You will need to get your board schematics to understand how the display interfaces are connected to your SOC DDI port.  This part is very board specific and needs porting.

mauricema, 2021-06-09:
Since the original UART enabling was resolved,  can we close this issue ?
GFX is a different problem, it will need VBT customization and board porting.

erik-kline, 2021-06-10:
it takes 30 seconds from the time i see
Switch to LongMode and jump to 64-bit kernel entrypoint ... 
before the linux image begins booting. it appears that it only boots about 1/3 of the time.
the error is completely silent. 
I have been unable to get any other linux image to boot on the card. But I see what you are doing and I see a path to debuging. I need to get handle on the boot problem first.

erik-kline, 2021-06-10:
the dmesg output is below. I note that the serial port is enabled at 26 seconds. so it appears that if the kernel crashs at a time before this it could be a silent error. My working assumption at this point is that the problem is related to some of the acpi errors that are in the log.

[    0.000000] Command line: init=/init root=/dev/ram0 rw 3 console=ttyS0,115200 console=ttyS1,115200 console=ttyS2,115200 console=ttyS0
[    0.000000] x86/split lock detection: warning about user-space split_locks                                                           
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'                                                  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'                                                                 
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'                                                                 
[    0.000000] x86/fpu: Supporting XSAVE feature 0x020: 'AVX-512 opmask'                                                                
[    0.000000] x86/fpu: Supporting XSAVE feature 0x040: 'AVX-512 Hi256'                                                                 
[    0.000000] x86/fpu: Supporting XSAVE feature 0x080: 'AVX-512 ZMM_Hi256'                                                             
[    0.000000] x86/fpu: Supporting XSAVE feature 0x200: 'Protection Keys User registers'                                                
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256                                                                   
[    0.000000] x86/fpu: xstate_offset[5]:  832, xstate_sizes[5]:   64                                                                   
[    0.000000] x86/fpu: xstate_offset[6]:  896, xstate_sizes[6]:  512                                                                   
[    0.000000] x86/fpu: xstate_offset[7]: 1408, xstate_sizes[7]: 1024                                                                   
[    0.000000] x86/fpu: xstate_offset[9]: 2432, xstate_sizes[9]:    8                                                                   
[    0.000000] x86/fpu: Enabled xstate features 0x2e7, context size is 2440 bytes, using 'compacted' format.                            
[    0.000000] BIOS-provided physical RAM map:                                                                                          
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable                                                            
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved                                                          
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000781b7fff] usable                                                            
[    0.000000] BIOS-e820: [mem 0x00000000781b8000-0x00000000786b7fff] reserved                                                          
[    0.000000] BIOS-e820: [mem 0x00000000786b8000-0x0000000078747fff] ACPI data                                                         
[    0.000000] BIOS-e820: [mem 0x0000000078748000-0x000000007874ffff] ACPI NVS                                                          
[    0.000000] BIOS-e820: [mem 0x0000000078750000-0x000000007fbfffff] reserved                                                          
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved                                                          
[    0.000000] BIOS-e820: [mem 0x00000000ff56d000-0x00000000ffffffff] reserved                                                          
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000002803fffff] usable                                                            
[    0.000000] NX (Execute Disable) protection: active                                                                                  
[    0.000000] SMBIOS 2.5 present.                                                                                                      
[    0.000000] DMI: Intel Corporation TigerLake Client Platform/TigerLake U DDR4 SODIMM RVP, BIOS SB_TGL.001.000.005.000.008.00007.D-2A2
[    0.000000] tsc: Detected 2800.000 MHz processor                                                                                     
[    0.000000] tsc: Detected 2803.200 MHz TSC                                                                                           
[    0.000449] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved                                                             
[    0.000451] e820: remove [mem 0x000a0000-0x000fffff] usable                                                                          
[    0.000453] last_pfn = 0x280400 max_arch_pfn = 0x400000000                                                                           
[    0.000455] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC                                                             
[    0.000455] last_pfn = 0x781b8 max_arch_pfn = 0x400000000                                                                            
[    0.000460] Using GB pages for direct mapping                                                                                        
[    0.000582] RAMDISK: [mem 0x77dc5000-0x77e2efff]                                                                                     
[... cut ...]
[   27.592374] 3                                                                                                                        
[   27.592375] with environment:                                                                                                        
[   27.592375] HOME=/                                                                                                                   
[   27.592376] TERM=linux                                                                                                               
[   27.592376] acpi_rsdp=0x786B8000

mauricema, 2021-06-10:
You have a long delay for EC timeout.  Do you actually have a EC (Embedded Controller) on your board at all.  If not,  you should consider removing EC from ACPI.

mauricema, 2021-06-10:
BTW, maybe it might be better to discuss each issue separately on mailing list.  A generic board enabling discussion could be hard to track in this case since many pieces might be involved, such as GPIO, UART, GFX, EC, etc.

erik-kline, 2021-06-10:
yes this thread is getting long a rambling.
I had difficulty removing the EC acpi code.
but I modfied the read and write methods in it to just return without error.
the code boots up every time now.

mauricema, 2021-06-10:
Yes,  we can also consider to add a EC disable in CFGDATA for TGL SBL to make it easier for user to handle this case.
Glad to know the progress you made!  Let me close this issue for now.  
If you have new issue to discuss , let us start new thread in mailing list instead.

Regards,
Maurice

closed 2021-06-10 by mauricema

mauricema, 2021-06-10:
For EC less board on TGL, we have created a PR to use CFGDATA to control it.
https://github.com/slimbootloader/slimbootloader/pull/1191
You can just put following line into your board DLT file to disable EC.
SILICON_CFG_DATA.EcEnable | 0

PennZheng, 2021-07-16:
> For EC less board on TGL, we have created a PR to use CFGDATA to control it.
> #1191
> You can just put following line into your board DLT file to disable EC.
> SILICON_CFG_DATA.EcEnable | 0

Disabling the EcEnable in the CfgData_Int_Tglu_Ddr4.dlt file is no use since the judgement result of ((SiCfgData == NULL) || (SiCfgData->EcEnable == 0)) in Stage1BBoardInitLib.c is always FALSE before PlatformId/BoardId is specified as 0x01 (BoardIdTglUDdr4).
https://github.com/slimbootloader/slimbootloader/blob/ec20bf39e7ce76bc90fa3d9191dff3c16c26c71e/Platform/TigerlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c#L779

It seems disabling the EcEnable in the CfgData_Silicon.yaml file is the only effective way. Is that right?

gdong1, 2021-07-16:
yes, in current logic, the platformId (==0) is not set yet so the delta file for DDR4 will not be apply.
you could locally update CfgData_Silicon.yaml to disable EC. we need figure out a solution if we need support both EC and EC-less board with same IFWI late.