slimbootloader/slimbootloader #1196: [EHL] Hardcoded GPIO pin was used for payload ID selection

mauricema, 2021-06-18:
Current SBL EHL platform used hard coded GPIO pin for payload ID selection.  
The preferred way is to use CFGDATA to allow platform determine which GPIO pin to use for payload ID section.  
Please look into CFL implementation for reference.

koktong-ong, 2021-06-21:
In SBL EHL Stage2BoardInitLib.c function UpdatePayloadId(VOID) :
```
  GenCfgData = (GEN_CFG_DATA *)FindConfigDataByTag (CDATA_GEN_TAG);
  if (GenCfgData == NULL) {
    ASSERT (FALSE);
    return;
  }
  SetPayloadId (GenCfgData->PayloadId);

  if (GetPayloadId () != AUTO_PAYLOAD_ID_SIGNATURE) {
    return;
  }
```
Payload ID will only check GPIO setting if 'AUTO' was detected in CFGDATA 
Else the Payload ID will get from CFGDATA

LeanSheng, 2021-06-22:
> In SBL EHL Stage2BoardInitLib.c function UpdatePayloadId(VOID) :
>
> ```
>   GenCfgData = (GEN_CFG_DATA *)FindConfigDataByTag (CDATA_GEN_TAG);
>   if (GenCfgData == NULL) {
>     ASSERT (FALSE); [...]

I thnak Maurice is referring to this line:
PayloadSelGpioPad = GPIO_VER3_GPP_D13;

We just need to let user configure via SiliconCfgData like CFL.

mauricema, 2021-06-22:
Yes,  hardcoded pin GPIO_VER3_GPP_D13 needs to be configurable.

koktong-ong, 2021-06-22:
Thanks @LeanSheng for helping on this!

LeanSheng, 2021-06-24:
#1204 here is the fix

PR #1297 (2021-09-22): [EHL] Add GPIO payload selection configuration
This patch added payload selection GPIO configuration
hardcoded GPIO pin for payload selection.

It also fixed #1196.
closed 2021-09-22 by mauricema