slimbootloader/slimbootloader #1197: [EHL] PlatformID cannot be overridden by platform data provided in stitching

mauricema, 2021-06-18:
In SBL, the PlatformData provided to stitching is supposed to allow hardcoded platform ID to be used.
However,  EHL platform will ignore this platform ID provided by stitching and always use the detected platform ID.  It makes the stitching data useless for SBL.   It needs to be fixed.

koktong-ong, 2021-06-21:
In Stage1B.c :
```
  // Overwrite platform ID if CFGDATA contains it
  PidCfgData = (PLATFORMID_CFG_DATA *)FindConfigDataByTag (CDATA_PLATFORMID_TAG);
  if ((PidCfgData != NULL) && (PidCfgData->PlatformId < 32)) {
    DEBUG ((DEBUG_INFO, "Override Board ID to %04X\n", PidCfgData->PlatformId));
    SetPlatformId (PidCfgData->PlatformId);
  }
```
The PidCfgData is returned NULL even the PlatformID is available in DLT file.
SBL EHL set the detected platform id in the PreConfigInit which before the code above was executed.
Should we standardize the PlatformID to be configure in which InitPhase in platform stage1B?

mauricema, 2021-06-21:
It is as expected.  PlatformId CFG is special in that it will not be enabled by default.  If enabled, the hard coded platform ID will be used.
The request here is to check  GetPlatformId() first.  If it is 0, then we can SetPlatformId(). Otherwise, we should not.

koktong-ong, 2021-06-22:
Submitted PR for the fix
https://github.com/slimbootloader/slimbootloader/pull/1202

mauricema, 2021-06-24:
Close it since it is fixed in #1202

closed 2021-06-24 by mauricema