slimbootloader/slimbootloader #1201: issue when building for Tigerlake: ‘FSP_M_CONFIG’ {aka ‘struct <anonymous>’} has no member named ‘SoftwareSramEnPreMem’

AntonioSanta, 2021-06-21:
Hi I merged Today with current master branch.
I get these errors when building for Tigerlake : 

/home/tiger/Ashare/tigerlake/slimbootloader/Platform/TigerlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c: In function ‘TccModePreMemConfig’:
/home/tiger/Ashare/tigerlake/slimbootloader/Platform/TigerlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c:97:22: error: ‘FSP_M_CONFIG’ {aka ‘struct <anonymous>’} has no member named ‘SoftwareSramEnPreMem’
   97 |   FspmUpd->FspmConfig.SoftwareSramEnPreMem   = TccCfgData->TccSoftSram;
      |                      ^
/home/tiger/Ashare/tigerlake/slimbootloader/Platform/TigerlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c:98:22: error: ‘FSP_M_CONFIG’ {aka ‘struct <anonymous>’} has no member named ‘DsoTuningEnPreMem’
   98 |   FspmUpd->FspmConfig.DsoTuningEnPreMem      = TccCfgData->TccTuning;
      |                      ^
/home/tiger/Ashare/tigerlake/slimbootloader/Platform/TigerlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c:99:22: error: ‘FSP_M_CONFIG’ {aka ‘struct <anonymous>’} has no member named ‘TccErrorLogEnPreMem’
   99 |   FspmUpd->FspmConfig.TccErrorLogEnPreMem    = TccCfgData->TccErrorLog;
      |                      ^
/home/tiger/Ashare/tigerlake/slimbootloader/Platform/TigerlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c:109:25: error: ‘FSP_M_CONFIG’ {aka ‘struct <anonymous>’} has no member named ‘TccStreamCfgBasePreMem’; did you mean ‘TccStreamCfgBase’?
  109 |     FspmUpd->FspmConfig.TccStreamCfgBasePreMem = (UINT32)(UINTN)TccStreamBase;
      |                         ^~~~~~~~~~~~~~~~~~~~~~
      |                         TccStreamCfgBase
/home/tiger/Ashare/tigerlake/slimbootloader/Platform/TigerlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c:110:25: error: ‘FSP_M_CONFIG’ {aka ‘struct <anonymous>’} has no member named ‘TccStreamCfgSizePreMem’; did you mean ‘TccStreamCfgSize’?
  110 |     FspmUpd->FspmConfig.TccStreamCfgSizePreMem = TccStreamSize;
      |                         ^~~~~~~~~~~~~~~~~~~~~~
      |                         TccStreamCfgSize
/home/tiger/Ashare/tigerlake/slimbootloader/Platform/TigerlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c:134:24: error: ‘FSP_M_CONFIG’ {aka ‘struct <anonymous>’} has no member named ‘TccCacheCfgBasePreMem’
  134 |     FspmUpd->FspmConfig.TccCacheCfgBasePreMem = (UINT32)(UINTN)TccCacheBase;
      |                        ^
/home/tiger/Ashare/tigerlake/slimbootloader/Platform/TigerlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c:135:24: error: ‘FSP_M_CONFIG’ {aka ‘struct <anonymous>’} has no member named ‘TccCacheCfgSizePreMem’
  135 |     FspmUpd->FspmConfig.TccCacheCfgSizePreMem = TccCacheSize;
      |                        ^
make: *** [GNUmakefile:321: /home/tiger/Ashare/tigerlake/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/TigerlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib/OUTPUT/Stage1BBoardInitLib.obj] Error 1

I double checked looking in the sources for these fields , and can't find declarations or references to them.

japelner-intc, 2021-06-21:
Hi Antonio,

This pull request [#1186](https://github.com/slimbootloader/slimbootloader/pull/1186) updated TGL to use a new microcode and a new version of FSP. It is in this version of the FSP (commit ID 4a31a4d972eaeb9aa41fef3b121df97f8f2a4e61) that these UPDs are defined. If you are intentionally using a different version of the FSP, then you should comment out those lines with the unresolved references.

Jim

mauricema, 2021-06-29:
Is this resolved ?   Can we close it ?

closed 2021-06-29 by AntonioSanta