slimbootloader/slimbootloader #1209: Move debug consent component to common code

LeanSheng, 2021-06-28:
many platforms are using debug consent flags (fsp params) to output debug tool logs/ traces. It would be great if can move them all into common yaml/ common codes since the implementation is pretty generic.

stalamudupula, 2021-07-01:
Do you have a list of what params you think can be common config items?

mauricema, 2021-09-11:
@LeanSheng   Do we know if all FSPs debug consent flags are common ?

LeanSheng, 2021-09-13:
the main debug consent flags are common for EHL, TGL and following projects:
      DEBUG ((DEBUG_INFO, "PlatformDebugConsent = %x\n",   Fspmcfg->PlatformDebugConsent));
      DEBUG ((DEBUG_INFO, "DciEn = %x\n",                  Fspmcfg->DciEn));
      DEBUG ((DEBUG_INFO, "DciModphyPg = %x\n",            Fspmcfg->DciModphyPg));
      DEBUG ((DEBUG_INFO, "DciUsb3TypecUfpDbg = %x\n",     Fspmcfg->DciUsb3TypecUfpDbg));
      DEBUG ((DEBUG_INFO, "DciDbcMode = %x\n",             Fspmcfg->DciDbcMode));
      DEBUG ((DEBUG_INFO, "DebugInterfaceEnable = %x\n",   Fspmcfg->DebugInterfaceEnable));
      DEBUG ((DEBUG_INFO, "CpuTraceHubMode = %x\n",        Fspmcfg->CpuTraceHubMode));
      DEBUG ((DEBUG_INFO, "CpuTraceHubMemReg0Size = %x\n", Fspmcfg->CpuTraceHubMemReg0Size));
      DEBUG ((DEBUG_INFO, "CpuTraceHubMemReg1Size = %x\n", Fspmcfg->CpuTraceHubMemReg1Size));
      DEBUG ((DEBUG_INFO, "PchTraceHubMode = %x\n",        Fspmcfg->PchTraceHubMode));
      DEBUG ((DEBUG_INFO, "PchTraceHubMemReg0Size = %x\n", Fspmcfg->PchTraceHubMemReg0Size));
      DEBUG ((DEBUG_INFO, "PchTraceHubMemReg1Size = %x\n", Fspmcfg->PchTraceHubMemReg1Size));
      
I am planning to move them into commonCfgdata

closed 2021-09-16 by LeanSheng