slimbootloader/slimbootloader #1221: [TGL] Slim Bootloader and UEFI Payload S3 resume fail

PennZheng, 2021-07-08:
Hi,

I'm working on a new designed Tiger Lake-U board.
The Slim Bootloader and UEFI Payload I used was download on April 2021.

During S3 resume phase, the SBL hangs on Stage2.

SBL log:
==============================
MP Init (Wakeup)
SMRR Base: 0x7B000006  Mask: 0xFF800000
SmmRamBase = 0x7B000000, SmmRamSize = 0x800000

MP Init (Run)
Detected 8 CPU threads
 CPU  0 APIC ID: 0
 CPU  1 APIC ID: 4
 CPU  2 APIC ID: 2
 CPU  3 APIC ID: 6
 CPU  4 APIC ID: 1
 CPU  5 APIC ID: 3
 CPU  6 APIC ID: 5
 CPU  7 APIC ID: 7

CPU SMM rebase failed!

STAGE_2: System halted!
==============================

Is this an known issue ?

Thanks

rprangar, 2021-07-08:
Hi

We have not seen such issue so far. So it may need further debugging.

Would it be possible for you to print the mMpDataStruct.SmmRebaseDoneCounter just before the "CPU SMM rebase failed!" debug message? Also, if possible, please attach the full debug logs during boot and s3 resume.

Thanks.

rprangar, 2021-07-09:
Also, I assume you are using the UEFI payload you obtained from Intel. 

Please try to print the SMMBASE_INFO structure here after it is located in the S3 path - 
https://github.com/slimbootloader/slimbootloader/blob/2832972ad6e8f550f9628eeca7027b0aa65eb10b/BootloaderCorePkg/Library/MpInitLib/MpInitLib.c#L419

PennZheng, 2021-07-09:
Hi,

The full debug logs during boot and S3 resume are attached here:

[From_G3_to_S0.log](https://github.com/slimbootloader/slimbootloader/files/6788252/From_G3_to_S0.log)
[From_S3_to_S0.log](https://github.com/slimbootloader/slimbootloader/files/6788253/From_S3_to_S0.log)

Thanks.

rprangar, 2021-07-09:
Hi 

Thanks for the log. From the log, it seems you are building the UEFI payload from open source EDK2.

SBL depends on the UEFI payload to fill in the ApicId/SmmBase fields in the SmmBaseInfo structure. Since the open source UEFI payload did not provide this information, it caused this issue.

mSmmBaseInfo->SmmBaseHdr.Signature = 0x4F435042
mSmmBaseInfo->SmmBaseHdr.Id        = 0x01
mSmmBaseInfo->SmmBaseHdr.Count     = 0x08
mSmmBaseInfo->SmmBaseHdr.TotalSize = 0x0048
_mSmmBaseInfo->SmmBase[0].ApicId_    = **0x00000000**
_mSmmBaseInfo->SmmBase[0].SmmBase_   = **0x00000000**
...

We will try to provide a fix next week. Meanwhile, you can either try the UEFI payload released by Intel or maybe comment the following line and give it a try. https://github.com/slimbootloader/slimbootloader/blob/2832972ad6e8f550f9628eeca7027b0aa65eb10b/Platform/TigerlakeBoardPkg/Library/Stage2BoardInitLib/Stage2BoardInitLib.c#L993

Thanks
Ravi

gdong1, 2021-07-14:
Could you check if this patch will fix this issue?
https://github.com/slimbootloader/slimbootloader/pull/1224

PennZheng, 2021-07-15:
> Could you check if this patch will fix this issue?
> #1224

I have checked this patch on my Tiger Lake board and it does fix this issue.

Thanks.

rprangar, 2021-07-15:
Thanks for reporting the issue and verifying the fix.

closed 2021-07-15 by rprangar