slimbootloader/slimbootloader #1226: [TGL] Slim Bootloader causes Windows 10 BSOD

PennZheng, 2021-07-15:
Hi,

I use the Slim Bootloader and UEFI Payload on an EC-less Tiger Lake UP3 board. However, a blue screen error occurs during Windows 10 startup. After removing the EC.ASL file content from the ACPI DSDT table, the Blue Screen of Death issue disappears. So it seems EC.ASL have something to do with this issue.
[image]

Besides, I encounter some build error during building SBL ACPI code.
[SBL_build_errors.txt](https://github.com/slimbootloader/slimbootloader/files/6821751/SBL_build_errors.txt)

Do you encounter the same issues?

The Slim Bootloader and UEFI Payload source code is the latest version:

Slim Bootloader Project
- Git Repository: https://github.com/slimbootloader/slimbootloader.git
- Download Date: 2021/07/09
- Git Commit ID: 2832972ad6e8f550f9628eeca7027b0aa65eb10b

EDK II Project (UEFI Payload)
- Git Repository: https://github.com/tianocore/edk2.git
- Download Date: 2021/07/09
- Git Commit ID: b560e9d9b67450d3baea56d0e211066722643c57

Thanks.

rprangar, 2021-07-15:
Hi

If your board does not have an EC, you can disable that using the config option as mentioned in this PR - https://github.com/slimbootloader/slimbootloader/pull/1191

What is the IASL version you are using?

Thanks
Ravi

japelner-intc, 2021-07-15:
Hi,

I have pulled the exact commit of Slim Bootloader and have verified no ACPI errors that you reported in SBL_build_errors.txt, previously. The version of IASL that we use is 20160422. Please use the **exact** versions of the tools on the [Windows Getting Started](https://slimbootloader.github.io/getting-started/build-host-setup.html#building-on-windows) or [Linux Getting Started](https://slimbootloader.github.io/getting-started/build-host-setup.html#building-on-linux) pages.

I also agree with Ravi that using the config option to disable EC support is a better way to do it.

Thanks,
Jim

PennZheng, 2021-07-16:
Hi,

I have disabled the EcEnable in the CfgData_Silicon.yaml file.
The IASL version I use is 20200214. I try to downgrade IASL to 20160422 version, after doing this the SBL build error and Windows 10 BSOD issues disappear.

BTW, disabling the EcEnable in the CfgData_Int_Tglu_Ddr4.dlt file is no use since the judgement result of ((SiCfgData == NULL) || (SiCfgData->EcEnable == 0)) in Stage1BBoardInitLib.c is always FALSE.
https://github.com/slimbootloader/slimbootloader/blob/ec20bf39e7ce76bc90fa3d9191dff3c16c26c71e/Platform/TigerlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c#L779

Thanks.

japelner-intc, 2021-07-16:
Hi,

I'm glad to hear that your BSOD and build error issues have been resolved.

WRT the .dlt file, I'm surprised with the behavior you see. To be honest, I typically modify the .yaml files directly. We will take a look into that issue.

Thanks,
Jim

PennZheng, 2021-07-17:
Thanks for your help.

closed 2021-07-17 by PennZheng