slimbootloader/slimbootloader #1228: [TGL] DRAM frequency lower than expected

PennZheng, 2021-07-17:
Hi,

I find that the DRAM frequency reported by CPU-Z in Windows 10 is lower than expected. The RAM is DDR4-3200 (1600MHz) SODIMM, but the "DRAM Frequency" shown in CPU-Z is 1066MHz. 
[image]

The following is the Slim Bootloader and UEFI Payload source code version :
Slim Bootloader Project
- Git Repository: https://github.com/slimbootloader/slimbootloader.git
- Download Date: 2021/07/16
- Git Commit ID: ec20bf39e7ce76bc90fa3d9191dff3c16c26c71e
EDK II Project (UEFI Payload)
- Git Repository: https://github.com/tianocore/edk2.git
- Download Date: 2021/07/09
- Git Commit ID: b560e9d9b67450d3baea56d0e211066722643c57

**I have done the following verification :**
1. After replacing the SBL firmware with the AMI BIOS on the same TGL UP3 board, the "DRAM Frequency" shown in CPU-Z is 1600MHz.
2. After changing the DdrFreqLimit field in the FSP-M UPD structure from 0 (Auto) to 2666/2933/3200 (1333/1466/1600 MHz), the "DRAM Frequency" shown in CPU-Z is 1333MHz.
[image]

In the Silicon\TigerlakePkg\FspBin\FspmUpd.h file, it says the maximal option of DdrFreqLimit is 2933 (1466MHz). **Does it mean the FSP do not support 3200 (1600MHz) DDR4 DRAM ?**
> /** Offset 0x018E - DDR Frequency Limit
>   Maximum Memory Frequency Selections in Mhz. Options are 1067, 1333, 1600, 1867,
>   2133, 2400, 2667, 2933 and 0 for Auto.
>   1067:1067, 1333:1333, 1600:1600, 1867:1867, 2133:2133, 2400:2400, 2667:2667, 2933:2933, 0:Auto
> **/
>   UINT16                      DdrFreqLimit;

Thanks

gdong1, 2021-07-18:
It looks not all the supported DdrFreqLimit options are listed in the FSP header file.
The same value supported from the BIOS should be also supported by FSP used by SBL.
So 3200 is supported in this case.

rprangar, 2021-07-20:
Hi

We will follow up with the FSP UPD options. Meanwhile, can you please try the 3200 setting and let us know if it resolves your issue.

Thanks

PennZheng, 2021-07-21:
Hi,

As I said, I had try the 3200 setting of DdrFreqLimit and the "DRAM Frequency" shown in CPU-Z was 1333MHz but not 1600MHz.

> I have done the following verification :
> 
> After replacing the SBL firmware with the AMI BIOS on the same TGL UP3 board, the "DRAM Frequency" shown in CPU-Z is 1600MHz.
> After changing the DdrFreqLimit field in the FSP-M UPD structure from 0 (Auto) to 2666/2933/**3200** (1333/1466/**1600** MHz), the "DRAM Frequency" shown in CPU-Z is 1333MHz.

Thanks

rprangar, 2021-07-21:
Yes, I missed it. Will provide an update soon.

japelner-intc, 2021-07-22:
Hi,

We have looked into this a bit and do see the restriction to 1064MHz when the DdrFreqLimit is set to AUTO. While we continue to look at this, could you please make the following modification to the option below and let us know what you see:

TigerlakeBoardPkg/CfgData/CfgData_Memory.yaml
```
  - MrcSafeConfig :
      name         : MRC Safe Config
      option       : $EN_DIS
      help         : >
                     Enables/Disable MRC Safe Config
      length       : 0x01
      value        : 0x01
```
Please add a delta override of this option setting it to **0 (zero)**.

We will provide more info soon.

Thanks,
Jim

PennZheng, 2021-07-26:
Hi,

I have done the following verfication, but it causes system hang.
1. Setting MrcSafeConfig to 0 (Disable MRC Safe Config) and setting DdrFreqLimit 0 (Auto)
2. Setting MrcSafeConfig to 0 (Disable MRC Safe Config) and setting DdrFreqLimit to 3200 (1600 MHz)

The following is the SBL log:
> Memory Init
> Load Security Cfg Data
> FSPM CfgData assignment
> Load Memory Cfg Data
> SkipMbpHob = 0x0
> Load Graphics Cfg Data
> board Id = 1 .....
> Call FspMemoryInit ...

Thanks

japelner-intc, 2021-07-26:
Hi,

Thanks for checking on this. This is good information. We will get back to you after we look into this further.

Thanks,
Jim

stanleyintel, 2021-07-27:
Hi,

   May you check if this patch fixes this issue: 
https://github.com/slimbootloader/slimbootloader/pull/1236

japelner-intc, 2021-07-27:
Just a quick update. I do see a consistent hang during memory initialization in three specific training procedures. Without those procedures being run, memory initialization completes and appears to be functional. Just wanted to let you know we're still actively looking into this.

PennZheng, 2021-08-10:
Hi,

Is there any update?

Thanks

stanleyintel, 2021-08-10:
Hi, 

   Could you add the following parameters in UpdateFspConfig()

  Fspmcfg->MrcSafeConfig = 0;
  Fspmcfg->DIMMODTT = 0;
  Fspmcfg->RDAPT = 0;
  Fspmcfg->RDODTT = 0;

PennZheng, 2021-08-11:
> Hi,
> 
> Could you add the following parameters in UpdateFspConfig()
> 
> Fspmcfg->MrcSafeConfig = 0;
> Fspmcfg->DIMMODTT = 0;
> Fspmcfg->RDAPT = 0;
> Fspmcfg->RDODTT = 0;

Hi,

After changing the above FSP UPD, the "DRAM Frequency" shown in CPU-Z is **1333MHz** but not 1600MHz.

Thanks

stanleyintel, 2021-08-11:
> > Hi,
> > Could you add the following parameters in UpdateFspConfig()
> > Fspmcfg->MrcSafeConfig = 0;
> > Fspmcfg->DIMMODTT = 0;
> > Fspmcfg->RDAPT = 0;
> > Fspmcfg->RDODTT = 0;
> 
> Hi,
> 
> After changing the above FSP UPD, the "DRAM Frequency" shown in CPU-Z is **1333MHz** but not 1600MHz.
> 
> Thanks

Good to know it does not hang anymore. Thanks. 
Have you try them together with Fspmcfg->DdrFreqLimit = 3200 ?
Also, it is good if you can add these settings before/around line 399.

stanleyintel, 2021-08-11:
> Yes, I have tried to set the DdrFreqLimit to 3200.
Hi, 
    Please try https://github.com/slimbootloader/slimbootloader/pull/1259/files .
    And keep MrcSafeConfig = 0, but no need to set DIMMODTT, RDAPT and RDODTT.

PennZheng, 2021-08-11:
Hi,

> Hi,
> Please try https://github.com/slimbootloader/slimbootloader/pull/1259/files .
> And keep MrcSafeConfig = 0, but no need to set (i.e., disable) DIMMODTT, RDAPT and RDODTT.

Now the SBL does not hang anymore, but the DRAM frequency is still 1333MHz with DdrFreqLimit set to 0 or 3200.

stanleyintel, 2021-08-11:
> Hi,
> 
> > Hi,
> > Please try https://github.com/slimbootloader/slimbootloader/pull/1259/files .
> > And keep MrcSafeConfig = 0, but no need to set (i.e., disable) DIMMODTT, RDAPT and RDODTT.
> 
> Now the SBL does not hang anymore, but the DRAM frequency is still 1333MHz with DdrFreqLimit set to 0 or 3200.

Thanks first for reporting this issue. It is a critical issue reproducible in Intel RVP (Reference Validation Platform). After fixing the hang problem, DRAM freq can reach 1600 Mhz in RVP.

Since AMI BIOS can get 1600 Mhz, it indicates some FSP UPDs are not aligned between AMI BIOS and your SBL configuration, or SPD is not correctly downloaded from Smbus. Please work with your BIOS vendor to align this settings. You may also contact your Intel representative to try with Intel released UEFI BIOS and/or SBL in TGL-U RVP board with your memory part.

PennZheng, 2021-08-12:
Hi,

I recheck the modification and find that I made a mistake yesterday. The UPD SaGv was set to 4 in the verfication yesterday. After restoring the SaGv to 0, the DRAM frequency is 1600 MHz on our board with SBL firmware.

Thanks for your support.

stanleyintel, 2021-08-12:
> Hi,
> 
> I recheck the modification and find that I made a mistake yesterday. The UPD SaGv was set to 4 in the verfication yesterday. After restoring the SaGv to 0, the DRAM frequency is 1600 MHz on our board with SBL firmware.
> 
> Thanks for your support.

Glad to hear that. Thanks for update. :)

closed 2021-08-12 by PennZheng