slimbootloader/slimbootloader #1233: [TGL] CPU base speed error

PennZheng, 2021-07-26:
Hi,

I encounter an issue that the CPU base speed shown in Windows 10 Task Manager is not mathed with the CPU nominal frequency.

CPU base speed shown in Windows 10 Task Manager with **Slim Bootloader and UEFI Payload firmware** :
[image]

CPU base speed shown in Windows 10 Task Manager with **AMI BIOS** on the same TGL UP3 board :
[image]

It seems there is an error on ACPI object **_PSS (Performance Supported States)**.

The following is the Slim Bootloader and UEFI Payload source code version :

Slim Bootloader Project

> Git Repository: https://github.com/slimbootloader/slimbootloader.git
> Download Date: 2021/07/16
> Git Commit ID: ec20bf3

EDK II Project (UEFI Payload)

> Git Repository: https://github.com/tianocore/edk2.git
> Download Date: 2021/07/09
> Git Commit ID: b560e9d9b67450d3baea56d0e211066722643c57

Thanks

japelner-intc, 2021-07-26:
Hi,

Thanks for reporting this. We will take a look at it.

Thanks,
Jim

stanleyintel, 2021-07-27:
Hi, 

  May you check if this patch fixes the issue? 
https://github.com/slimbootloader/slimbootloader/pull/1236

PennZheng, 2021-07-28:
Hi,

I have verify the patch on our Tiger Lake board. It does fix the CPU base speed error issue.
How does the CPOC duplication cause this issue?

Thanks

stanleyintel, 2021-07-28:
CP0C table is for _CPC. Duplicated tables may result in wrongly access.

closed 2021-07-28 by PennZheng