slimbootloader/slimbootloader #1234: [TGL] DDI TCP HDMI port not work

PennZheng, 2021-07-26:
Hi,

I encounter an issue on an EC-less Tiger Lake UP3 board that the DDI TCP1 HDMI port cannot work during SBL firmware execution **with the FSP provided GOP driver**. However, the DDI TCP1 HDMI port can function properly in Windows 10 **with the Intel Graphics Windows Driver**.

The monitor does not show the Slim Bootloader logo during SBL firmware execution.
It says in the SBL debug log : "Failed to get GFX HOB from FSP".

> Call FspSiliconInit ... 
> Success
> FspGfxHob is not available
> Failed to get GFX HOB from FSP

The DDI B HDMI port functions properly during SBL firmware execution.
Here is the partial SBL debug log :

> Call FspSiliconInit ... 
> Success
> FspGfxHob->FrameBufferBase = 0xB0000000
> Graphics Info: 1920 x 1080 x 32 @ 0xB0000000

Do you have any idea about this issue?

The following is the Slim Bootloader and UEFI Payload source code version :

Slim Bootloader Project

> Git Repository: https://github.com/slimbootloader/slimbootloader.git
> Download Date: 2021/07/16
> Git Commit ID: ec20bf3

EDK II Project (UEFI Payload)

> Git Repository: https://github.com/tianocore/edk2.git
> Download Date: 2021/07/09
> Git Commit ID: b560e9d9b67450d3baea56d0e211066722643c57

Thanks

japelner-intc, 2021-07-26:
Hi,

Thanks for reporting this issue. We'll take a look at it.

Thanks,
Jim

randylintw, 2021-07-27:
Reference validation platform is using DDI B HDMI as default configuration and sync to your observation. 

1. May I know which chip used in **DDI TCP1 HDMI port** ?
2. How about BIOS behavior? Do you see the Tianocore icon while booting ?

PennZheng, 2021-07-27:
1. There is no chip used in the DDI TCP1 HDMI port.
2. I cannot see any icon while booting.

randylintw, 2021-07-27:
On reference platform, I can see the SBL icon on Type C HDMI display.
Can you see UEFI menu on TCP1 port ?

PennZheng, 2021-07-27:
So you use a Type C to HDMI cable to connect the reference validation platform to a HDMI monitor ?
Our EC-less Tiger Lake UP3 board has no Type C port. I just use a HDMI cable to connect our Tiger Lake UP3 board with a HDMI monitor.
I cannot see UEFI menu on DDI TCP1 HDMI port.

randylintw, 2021-07-28:
How about the behavior on UEFI BIOS ?
For my testing, eDP, HDMI on DDI-B, Type C HDMI on TCP-0, all can show SBL icon.

PennZheng, 2021-07-28:
I cannot check the behavior on UEFI BIOS since we do not have any third-party UEFI BIOS which configures the DDI TCP1 as a HDMI port in the VBT.

randylintw, 2021-07-28:
VBT.bin is used for GFX driver in FSP, do you have the tool to modify it ?

PennZheng, 2021-07-28:
I do not have the source code of the third-party UEFI BIOS.
I have tried the following method :

1. I use a binary file editing software to extract the VBT.bin from the UEFI BIOS IFWI file.
2. After editing the VBT.bin file by means of the Intel DisCon tool, I merge the modified VBT.bin file into the UEFI BIOS IFWI file.

However, the modified UEFI BIOS halts with "X64 Exception Type - 0D(#GP - General Protection)" error on our EC-less TGL UP3 board. **The original UEFI BIOS does not cause the system halt.**

randylintw, 2021-07-28:
The hang cause by secure feature enabled in the UEFI BIOS bin file.
Here is the code snap that may affect the DDI port setting and you can try to modify to fit your board. 
https://github.com/slimbootloader/slimbootloader/blob/6a5511883e40657837e3b6925aa4224ff00c5d92/Platform/TigerlakeBoardPkg/Library/Stage2BoardInitLib/Stage2BoardInitLib.c#L801

PennZheng, 2021-07-29:
I tried to commented the connect of the **TglUDdr4GopVbtSpecificUpdate** function before reporting this issue, but it made no difference to this issue.

randylintw, 2021-07-30:
Have you modified the vbt.bin by Discon in 
Platform\TigerlakeBoardPkg\VbtBin\Vbt.dat ?
You can contact your Intel Graphics representative for the new DisCon tool to understand its capability.

PennZheng, 2021-08-02:
Yes, I have modified the Tiger Lake VBT.bin by DisCon.
I have checked the difference between the original VBT.bin of Tiger Lake RVP board and the modified VBT.bin of our EC-less Tiger Lake UP3 board.

- The TGL UP3 RVP board VBT.bin configures the DDI TCP1 port as DisplayPort and enables the "DP Alt mode over Type C" setting.
- Our TGL UP3 board VBT.bin configures the DDI TCP1 port as HDMI and disables the "DP Alt mode over Type C" setting.

Maybe the GOP driver FSP provided has issues with the DDI TCP 0/1/2/3 port.

randylintw, 2021-08-02:
Can you provide sjf [vbt change diff from origin VBT] ?

PennZheng, 2021-08-02:
The sjf file is attached here:
[DDI_TCP1_Configuration.zip](https://github.com/slimbootloader/slimbootloader/files/6914794/DDI_TCP1_Configuration.zip)

The HDMI topology we use is like this:
[image]

randylintw, 2021-08-02:
Hi reporter, even set the setting as DDI_TCP1_Configuration.zip but I can' find the error log as you mention. 
Do you stitch the SBL with 3rd party IFWI binary?

PennZheng, 2021-08-02:
Sorry. I realize that I made a mistake.
The old version FSP (commit ID: f27fb2424f6d2701737242d073d6935cfe542700) has this issue.
The new version FSP (commit ID: 4a31a4d972eaeb9aa41fef3b121df97f8f2a4e61) has resolved this issue.

> [TGL] Use MR2 FSP and Microcode for TGL-U (#1186)
> MR2 FSP is available so update SBL to use MR2 FSP.
> and also use new Microcode required by new FSP
> Update platform code on FSP UPDs, especially enable TCC feature.
> Update TGL platform version to 1.2 since MR2 released.

Thanks for your help.

closed 2021-08-02 by PennZheng