slimbootloader/slimbootloader #1235: [TGL] DDI A eDP port not work in Windows 10

PennZheng, 2021-07-26:
Hi,

I encounter an issue on an EC-less Tiger Lake UP3 board about the DDI A eDP port.
The internal panel on the DDI A eDP port can function properly in Windows 10 without the Intel Graphics Windows Driver. **After installing the Intel Graphics Windows Driver, the internal panel on the DDI A eDP port is turned off due to unknown reason.**

It seems that the Intel Graphics Windows Driver turns off the internal panel on the DDI A eDP port due to some reasons.
I have checked the relevant VBT settings and ACPI table, but I cannot find anything wrong.

Do you have any idea about this issue?

The following is the Slim Bootloader and UEFI Payload source code version :

Slim Bootloader Project

> Git Repository: https://github.com/slimbootloader/slimbootloader.git
> Download Date: 2021/07/16
> Git Commit ID: ec20bf3

EDK II Project (UEFI Payload)

> Git Repository: https://github.com/tianocore/edk2.git
> Download Date: 2021/07/09
> Git Commit ID: b560e9d9b67450d3baea56d0e211066722643c57

Thanks

japelner-intc, 2021-07-26:
Thanks for reporting this. We will look into it.

randylintw, 2021-07-27:
I tried to use SblOpen@d627d92c94bca39e and open edk2@b560e9d9b6745, 
win10 with intel gfx driver : igfx_win_100.9684.exe. 
I cannot see this issue on our reference validation platform [DDI-A connect to 4k eDP pannel].
Do the  output go to HDMI ?

PennZheng, 2021-07-27:
Is there an EC chip on your reference validation platform ?
What do you mean by "Do the output go to HDMI ?" ? Our EC-less Tiger Lake UP3 board only use the eDP signals of DDI A, but not the DDI signals of DDI A.

randylintw, 2021-07-27:
Our reference platform has EC chip.
I mean if there is HDMI cable connect to that system, what's the behavior ?

PennZheng, 2021-07-27:
When I connect a HDMI monitor to the DDI B HDMI port, the DDI B HDMI port functions properly during SBL and UEFI Payload execution and also functions properly in Windows 10 with Intel Graphics Driver. Meanwhile the DDI A eDP Port functions properly during SBL and UEFI Payload execution but cannot works in Windows 10 with Intel Graphics Driver (the eDP panel connected to the DDI A eDP port is turned off due to unknown reason).

randylintw, 2021-07-28:
Can you check the Lid State ? You can using flashlight to check the LCD panel to see if back light have been turn off.

PennZheng, 2021-07-28:
We have confirmed that the eDP backlight control signals (eDP_VDDEN and eDP_BKLTEN) disappear after Windows 10 starting up.
By the way, there is no such issue on third-party UEFI BIOS.

randylintw, 2021-07-28:
Do you turn off the EcEnable in CfgData/CfgData_Silicon.yaml ? 
Can you check the EC ACPI table exposed by SBL in win10?
Maybe Lid NVS State cause this issue.

PennZheng, 2021-07-28:
I have already turned off the EcEnable.

The **Lid State** field in the SA NVS area has a value of 0 (lid is closed). Is this the root cause?
[image]

randylintw, 2021-07-28:
Can you have a try to add else statement to have 
SaNvs->LidState               = 1;
@ https://github.com/slimbootloader/slimbootloader/blob/6a5511883e40657837e3b6925aa4224ff00c5d92/Platform/TigerlakeBoardPkg/Library/Stage2BoardInitLib/Stage2BoardInitLib.c#L2891

PennZheng, 2021-07-29:
After adding the else statement, the Lid State field in the SA NVS area has a value of 1 (lid is opened). However, the DDI A eDP port still does not work in Windows 10.

randylintw, 2021-07-30:
Can you try to force CLID to be 3 like this?
+++ b/Platform/TigerlakeBoardPkg/AcpiTables/SaSsdt/Igfx.asl
@@ -98,6 +98,7 @@ Method(_DOD,0)
   //
   // Two LFP devices are supporte by default
   //
+  Store(3,\_SB.PC00.GFX0.CLID)^M
Don't need this change SaNvs->LidState = 1;

PennZheng, 2021-07-30:
Hi,

"Store(3,_SB.PC00.GFX0.CLID)" causes build error "Object not found or not accessible from scope ^  (_SB.PC00.GFX0.CLID)".
I use "Store(3,CLID)" instead of "Store(3,_SB.PC00.GFX0.CLID)".
I have verified that adding "Store(3,CLID)" into Method(_DOD,0) can resolve this issue.

Thanks

randylintw, 2021-08-02:
Hi reporter, since it's' customizing platform issue and it has been identified to ACPI table, can you close this ticket? Thanks.

PennZheng, 2021-08-02:
Thanks for your help.

closed 2021-08-02 by PennZheng