slimbootloader/slimbootloader #1237: [TGL] 8254 Counter 2 failed to provide the speaker tone

PennZheng, 2021-07-28:
Hi,

I encounter an issue on an EC-less Tiger Lake UP3 board that the 8254 Counter 2 fails to provide the speaker tone.

I confirm that the GPP_B14 / SPKR / TIME_SYNC1 / GSPI0_CS1# pin is configured as native function 1 (SPKR) and the timer programming procedure is correct.

1. Write 0xB6 to the "Timer Control Word Register (TCW)—Offset 43h" IO register.
2. Write the least and most significant bytes of the 16 bit count value to the "Counter 2 - Counter Access Ports Register (C2_CAPR)—Offset 42h" IO register.
3. Set the SPKR_DAT_EN and TIM_CNT2_EN bit of the "NMI Status and Control (NMI_STS_CNT)—Offset 61h" IO register.

**There is no such issue on third-party UEFI BIOS (on the same TGL UP3 board).**

The following is the Slim Bootloader and UEFI Payload source code version :

Slim Bootloader Project

> Git Repository: https://github.com/slimbootloader/slimbootloader.git
> Download Date: 2021/07/16
> Git Commit ID: ec20bf3

EDK II Project (UEFI Payload)

> Git Repository: https://github.com/tianocore/edk2.git
> Download Date: 2021/07/09
> Git Commit ID: b560e9d9b67450d3baea56d0e211066722643c57

Thanks

gdong1, 2021-07-29:
Legacy 8254 timer support was replaced by HPET timer in the EDKII repo so 8254 is used in UEFI payload.
You could change the timerlib back to 8254 module to UEFI payload have a try.

PennZheng, 2021-08-09:
Thanks for your support.

closed 2021-08-09 by PennZheng