slimbootloader/slimbootloader #1238: [TGL] Add new config to CfgData cause error "Buffer Size too small" in CfgDataTool.py

qkiet, 2021-07-29:
Hi all,

I'd like to add new FSP config in CfgData_Silicon.yaml, which to be included in CfgData. This is how I added in CfgData_Silicon.yaml:
```yaml
  - EcEnable     :
      name         : Enable EC Device
      type         : Combo
      option       : $EN_DIS
      help         : >
                     Enable/disable EC
      length       : 0x01
      value        : 0x01
### Custom config below###
  - CnviMode :
      name         : CNVI Mode
      type         : Combo
      option       : $EN_DIS
      help         : >
                     Configure Cnvi mode, 0- Disable, 1- Auto
      length       : 0x01
      value        : 0x01
```
However, when I build slimbootloader, I got error "Buffer Size too small" in CfgDataTool.py

```prompt
Traceback (most recent call last):
  File "BootloaderCorePkg/Tools/CfgDataTool.py", line 954, in <module>
    sys.exit(Main())
  File "BootloaderCorePkg/Tools/CfgDataTool.py", line 951, in Main
    return Args.func(Args)
  File "BootloaderCorePkg/Tools/CfgDataTool.py", line 716, in CmdMerge
    CfgData.Parse(CfgBinFile)
  File "BootloaderCorePkg/Tools/CfgDataTool.py", line 341, in Parse
    DataCond = CCfgData.CDATA_COND.from_buffer(CondBin)
ValueError: Buffer size too small (0 instead of at least 4 bytes)
```

Can you please tell me how to get over with this problem? Thank you

stalamudupula, 2021-07-29:
Please adjust the length of the 'Dummy' variable (following your 'CnviMode') to be 0.

Basic idea is to have the entire config item's length to be 4-byte aligned. I just tested and it worked. Please update after your trial.

qkiet, 2021-07-30:
Many thanks, I tried and it actually works. Turn out these yaml file need 4-byte aligned

closed 2021-07-30 by qkiet