slimbootloader/slimbootloader #1240: [TGL] Boot to Windows but cannot use Intel Xe Graphic driver

qkiet, 2021-08-01:
Hi there,

I just build BIOS with slimbootloader and it successfully boots into Windows. But I cannot use Intel integrated graphics (Intel Xe graphics) even I've installed Intel Xe graphics driver.

I build froom slimbootloader commit d627d92c94bca39e892431878829e2869f9d616e. The payload is UEFI and used Edk2 06dc822d045c2bb42e497487935485302486e151 to build

I wonder if I misconfigure anything. Can you please give me some advice? Thank you!

randylintw, 2021-08-02:
What to mean for "use Intel integrated graphics" ?
I install the latest driver and can see device manager show the Intel Iris Xe Graphics.

qkiet, 2021-08-02:
> What to mean for "use Intel integrated graphics" ?
> I install the latest driver and can see device manager show the Intel Iris Xe Graphics.

my platform does not use any kind of GPU, only rely on Intel Xe integrated Graphics. My problem is when I installed the exact Intel Xe graphics driver then restarted, Device Manager did not recognize Intel Xe Graphics driver (sometimes it did, most of the time it didn't). I don't know if I need to configure ACPI in "Platform/<Platform_name/AcpiTables" or something else to make it work.

randylintw, 2021-08-02:
From https://slimbootloader.github.io/supported-hardware/tgl-rvp.html
The list of platform has been verified with default configuration.
Maybe you need to customize your ACPI table for-fit your HW configuration.

qkiet, 2021-08-03:
> From https://slimbootloader.github.io/supported-hardware/tgl-rvp.html
> The list of platform has been verified with default configuration.
> Maybe you need to customize your ACPI table for-fit your HW configuration.

Can you give me a hint of how to customize ACPI for proper Intel Xe Graphics function?

randylintw, 2021-08-03:
Hi Reporter, 
1. What kind of CPU do you use ? 
2. Can you to test on ubuntu with UEFI payload ?
3. Do you find any error message on win10 eventlog or device manager ? Is driver can't be loaded ?

qkiet, 2021-08-03:
> Hi Reporter,
> 
> 1. What kind of CPU do you use ?
> 2. Can you to test on ubuntu with UEFI payload ?
> 3. Do you find any error message on win10 eventlog or device manager ? Is driver can't be loaded ?

1. I use CPU Intel Tiger Lake 1180G7
2. We've already boot with ubuntu and it did not work
3. In Device Manager Windows 10, it report **error 43** for Intel Irix Xe Graphics

randylintw, 2021-08-04:
Hi Reporter, 
  1180G7 is UP4-class and it's not in the SBL officially supported hardware.

qkiet, 2021-08-05:
> Hi Reporter,
> 1180G7 is UP4-class and it's not in the SBL officially supported hardware.

What a pity. Thanks for your advices, anyway.

closed 2021-08-05 by qkiet