slimbootloader/slimbootloader #1244: [TGL] Virtualization is disabled in Windows 10

PennZheng, 2021-08-03:
Hi,

I encounter an issue on an EC-less Tiger Lake UP3 board that virtualization is disabled in Windows 10.
[image]
[image]

I have confirmed that the FSP UPD setting **VtdDisable** is configured as 0 (Intel VT-d is enabled).

**There is no such issue on the same board with third-party UEFI BIOS.**

The following is the Slim Bootloader and UEFI Payload source code version :

Slim Bootloader Project

- Git Repository: https://github.com/slimbootloader/slimbootloader.git
- Download Date: 2021/07/16
- Git Commit ID: ec20bf3

EDK II Project (UEFI Payload)

- Git Repository: https://github.com/tianocore/edk2.git
- Download Date: 2021/07/09
- Git Commit ID: b560e9d9b67450d3baea56d0e211066722643c57

Thanks

PennZheng, 2021-08-04:
After setting the VtdDisable as 0 and setting the VmxEnable as 1, the virtualization is enabled in Windows 10.
What is the difference between the **VtdDisable** and **VmxEnable**?

gdong1, 2021-08-05:
The CPU flag for Intel HW Virtualization is VMX is the CPU flag to enable Intel HW virtualization (VT-x). It is required by  hypervisor. VT-d allows hypervisor direct passthrough of devices. Often it is nice to have for hypervisor to enable virtualization.

PennZheng, 2021-08-06:
Thanks for your reply.

closed 2021-08-06 by PennZheng